thesofproject/sof #6448: [BUG][BDW] Playback test found IPC timeout TRIG_START 0x60040000

fredoh9, 2022-10-19:
**Describe the bug**
SOF CI found BDW_WSB_RT286 with just playback test, check-playback-50rounds.sh
Intenal test: 16338?model=BDW_WSB_RT286&testcase=check-playback-50rounds

console:
```
2022-10-17 06:04:31 UTC [REMOTE_INFO] ===== Testing: (Round: 22/50) (PCM: Low Latency [hw:0,0]) (Loop: 1/1) =====
2022-10-17 06:04:31 UTC [REMOTE_COMMAND] aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 1 /dev/zero -v -q
Hardware PCM card 0 'sof-bdw rt286' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16384
  period_size  : 4096
  period_time  : 85333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4096
  period_event : 0
  start_threshold  : 16384
  stop_threshold   : 16384
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
error: fetch_entry() failed with: -1, aborting
aplay: pcm_write:2127: write error: Input/output error
2022-10-17 06:04:33 UTC [REMOTE_ERROR] aplay on PCM hw:0,0 failed at 1/1.
2022-10-17 06:04:33 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2022-10-17 06:04:33 UTC [REMOTE_ERROR]  die()  @  /home/ubuntu/sof-test/test-case/../case-lib/lib.sh
2022-10-17 06:04:33 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-playback.sh:98
2022-10-17 06:04:33 UTC [REMOTE_INFO] Starting /usr/local/bin/sof-logger -l /etc/sof/sof-bdw.ldc
2022-10-17 06:04:37 UTC [REMOTE_INFO] pkill -TERM sof-logger
2022-10-17 06:04:37 UTC [REMOTE_ERROR] sof-logger was already dead
2022-10-17 06:04:38 UTC [REMOTE_INFO] nlines=7 /home/ubuntu/sof-test/logs/check-playback/2022-10-17-06:03:50-9852/etrace.txt
2022-10-17 06:04:38 UTC [REMOTE_INFO] ktime=939 sof-test PID=5464: ending
2022-10-17 06:04:39 UTC [REMOTE_INFO] Test Result: FAIL!
```

dmesg:
```
[  931.678373] kernel: snd_sof:ipc3_log_header: sof-audio-acpi-intel-bdw INT3438:00: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  932.185168] kernel: sof-audio-acpi-intel-bdw INT3438:00: ipc tx timed out for 0x60040000 (msg/reply size: 12/12)
[  932.185234] kernel: sof-audio-acpi-intel-bdw INT3438:00: preventing DSP entering D3 state to preserve context
[  932.185242] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump start ]------------
[  932.185284] kernel: sof-audio-acpi-intel-bdw INT3438:00: IPC timeout
[  932.185316] kernel: sof-audio-acpi-intel-bdw INT3438:00: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  932.185432] kernel: sof-audio-acpi-intel-bdw INT3438:00: unexpected fault 0x00000000 trace 0x80000000
[  932.185478] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc host -> DSP: pending yes complete no raw 0x80000000
[  932.185529] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask host: pending no complete no raw 0x7fff0000
[  932.185577] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: ipc DSP -> host: pending no complete no raw 0x00000000
[  932.185626] kernel: sof-audio-acpi-intel-bdw INT3438:00: error: mask DSP: pending no complete no raw 0x7f9e0000
[  932.185673] kernel: sof-audio-acpi-intel-bdw INT3438:00: ------------[ DSP dump end ]------------
[  932.185724] kernel: sof-audio-acpi-intel-bdw INT3438:00: ASoC: error at soc_component_trigger on INT3438:00: -110
[  932.185776] kernel:  Low Latency: ASoC: trigger FE cmd: 1 failed: -110
[  932.185787] kernel: sof-audio-acpi-intel-bdw INT3438:00: trace IO error
```

etrace:
```
[   258737018.677048] (   258737024.000000) c0 dma-trace  src/trace/dma-trace.c:654  ERROR dtrace_add_event(): number of dropped logs = 150
[   267173575.425143] (     8436557.000000) c0 dma-trace  src/trace/dma-trace.c:654  ERROR dtrace_add_event(): number of dropped logs = 4676
[   275577091.341217] (     8403516.000000) c0 dma-trace  src/trace/dma-trace.c:654  ERROR dtrace_add_event(): number of dropped logs = 4254
[   427082281.414692] (   151505184.000000) c0 dma-trace  src/trace/dma-trace.c:654  ERROR dtrace_add_event(): number of dropped logs = 561
```

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg MODEL=BDW_WSB_RT286 ~/sof-test/test-case/check-playback.sh -d 1 -l 1 -r 50

**Reproduction Rate**
1/50

**Environment**
- Kernel Branch: topic/sof-dev
- Kernel Commit: ea1ca7d0610c (https://github.com/thesofproject/linux/commit/ea1ca7d0610c)
- SOF Branch: stable-v2.2
- SOF Commit: afe466da0238
- TPLG: sof-bdw-rt286.tplg

**Attachment**
[slogger-bdw.txt](https://github.com/thesofproject/sof/files/9823184/slogger-bdw.txt)
[dmesg-bdw.txt](https://github.com/thesofproject/sof/files/9823185/dm
[etrace-bdw.txt](https://github.com/thesofproject/sof/files/9823187/etrace-bdw.txt)
esg-bdw.txt)

fredoh9, 2022-12-02:
Found one more case,
internal test link: 18236?model=BDW_WSB_RT286&testcase=check-capture-50rounds

plbossart, 2022-12-12:
we've stopped testing Broadwell, at this point there's no plan to fix this problem

closed 2022-12-12 by plbossart