thesofproject/sof #6465: [BUG] ipc timed out for 0x12030000|GLB_DELETE_PIPELINE when testing multiple-pipeline-capture on TGLU_RVP_SDW_IPC4ZPH

keqiaozhang, 2022-10-24:
**Describe the bug**
We have a lot of IPC4 related issue have been closed recently, the frequently occurring issues have disappeared in today's CI tests.
But we found one new issue on TGLU_RVP_SDW_IPC4ZPH when testing multiple-pipeline-capture. There's an IPC times out error for  0x12030000|GLB_DELETE_PIPELINE. The reproduction rate are high, the reproduction rate is almost 100% with `multiple-pipeline-capture-50.sh`. But it hard to say whether it's a regression, It may be hidden by previous issues before and exposed now. Need further debugging.

**To Reproduce**
`~/sof-test/test-case/multiple-pipeline.sh -f c -l 50`

**Reproduction Rate**
Almost 100% with` ~/sof-test/test-case/multiple-pipeline.sh -f c -l 50`

**dmesg**
```
[ 1713.606569] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x46010006|0x30004: MOD_UNBIND
[ 1713.606669] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x66000000|0x30004: MOD_UNBIND
[ 1713.606698] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x46010006|0x30004: MOD_UNBIND
[ 1713.606699] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget gain.2.1 freed
[ 1713.606701] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12030000|0x0: GLB_DELETE_PIPELINE
[ 1714.108812] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12030000|0x0
[ 1714.108829] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 1714.108833] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1714.108853] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[ 1714.108860] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1714.108881] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x92030000|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 1714.108889] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1714.108894] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1714.108899] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 1714.108903] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 1714.108919] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 1714.108952] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 1714.108957] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1714.108963] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.2
[ 1714.108975] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[ 1714.108989] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
```
**mtrace**
```
[   14.059000] <inf> ipc: buffer new size 0x300 id 4.0 flags 0x0
[   14.059000] <inf> pipe: comp:4 0x60002 connect buffer 0 as sink
[   14.059000] <inf> pipe: comp:4 0x20001 connect buffer 0 as source
[   14.059000] <inf> ipc: rx	: 0xc5010002|0x1000003
[   14.059000] <i[   14.062000] <err> copier: comp:4 0x40004 failed to find dai comp or sink pipeline not running.
[   14.062000] <err> copier: comp:4 0x40004 failed to find dai comp or sink pipeline not running.
[   14.062000] <inf> ipc: rx	: 0x93000003|0x1
[   14.062000] <inf> pipe: pipe:3 0x0 pipe trigger cmd 2
[   14.062000] <wrn> ipc: ipc_pipeline_complete(): no scheduling component specified, use comp 131072
[   14.062000] <inf> pipe: pipe:2 0x0 pipeline complete, clock freq 400000000Hz
[   14.063000] <inf> dw_dma: dw_dma_pause(): dma 1 channel 0 pause
[   14.063000] <inf> alh_dai: alh_trigger() cmd 2
[   14.063000] <inf> ll_schedule: task complete 0xbe0949c0 0x20240U
[   14.063000] <inf> ll_schedule: num_tasks 4 total_num_tasks 4
[   14.063000] <inf> ipc: rx	: 0x93000004|0x1
[   14.063000] <inf> pipe: pipe:3 0x0 pipe trigger cmd 8
[   14.063000] <inf> ll_schedule: task add 0xbe0949c0 0x20240U priority 0 flags 0x0
[   14.063000] <inf> pipe: pipe:2 0x0 pipe prepare
[   14.063000] <inf> pipe: pipe:2 0x0 pipe trigger cmd 7
[   14.063000] <inf> ll_schedule: task add 0xbe09a080 0x20240U priority 0 flags 0x0
[   14.064000] <inf> alh_dai: alh_trigger() cmd 8
[   14.064000] <inf> dw_dma: dw_dma_release(): dma 1 channel 0 release
[   14.064000] <inf> dw_dma: dw_dma_stop(): dma 1 channel 0 stop
[   14.064000] <inf> alh_dai: alh_trigger() cmd 3
[   14.064000] <err> copier: comp:2 0x40002 failed to find dai comp or sink pipeline not running.
[   14.064000] <err> copier: comp:2 0x40002 failed to find dai comp or sink pipeline not running.
[   14.064000] <inf> host: comp:2 0x40002 no bytes to copy, 384 free in buffer, 0 available in DMA
[   15.044000] <inf> ll_schedule: ll timer avg 5596, max 10501, overruns 0
[   16.068000] <inf> ll_schedule: ll timer avg 5755, max 6336, overruns 0
[   17.092000] <inf> ll_schedule: ll timer avg 5755, max 6346, overruns 0
[   18.116000] <inf> ll_schedule: ll timer avg 5755, max 6346, overruns 0
[   19.140000] <inf> ll_schedule: ll timer avg 5755, max 6342, overruns 0
[   19.688000] <inf> ipc: rx	: 0x93000003|0x1
[   19.688000] <inf> pipe: pipe:1 0x0 pipe trigger cmd 2
[... 42 lines cut ...]
[   19.696000] <inf> ipc: rx	: 0x93000003|0x1
[   19.696000] <inf> pipe: pipe:3 0x0 pipe trigger cmd 2
[   19.696000] <inf> component: comp:3 0x30000 comp_set_state(), state already set to 4
[   19.696000] <inf> pipe: pipe:4 0x0 pipe trigger cmd 2
[   19.697000] <err> copier: comp:4 0x40004 failed to find dai comp or sink pipeline not running.
[   19.697000] <inf> ll_schedule: task complete 0xbe099a80 0x20240U
[   19.697000] <inf> ll_schedule: num_tasks 1 total_num_tasks 1
[   19.697000] <inf> ll_schedule: zephyr_domain_unregister domain->type 1 domain->clk 4
[   19.697000] <inf> ipc: rx	: 0x93000003|0x1
[   19.697000] <inf> pipe: pipe:3 0x0 pipe trigger cmd Terminated
```
**tplg**
[image]

**Environment**
Kernel Branch: topic/sof-dev [7d2d29c3fc54](https://github.com/thesofproject/linux/commit/7d2d29c3fc54)

SOF Branch: main  [adce54445012](https://github.com/thesofproject/sof/commit/adce54445012)

Zephyr Commit: [zephyr-v3.2.0-441-ged661a6c6909](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.2.0-441-ged661a6c6909)
Platform: TGLU_RVP_SDW_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/9848080/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/9848082/mtrace.txt)

abonislawski, 2022-10-26:
Patches used in MTL case:

Kernel:
2a1be12c4d77 ASoC: SOF: Intel: add trigger callback into sdw_callback
6d1c1a73e112 soundwire: Intel: add trigger callback

SOF (Zephyr):
https://github.com/zephyrproject-rtos/zephyr/pull/51678

all previous patches for timeouts should be already on main

keqiaozhang, 2022-10-31:
Closing this bug as it has been fixed.

closed 2022-10-31 by keqiaozhang