thesofproject/sof #6466: [BUG][IPC4] various IPC errors on TGLU_UP_HDA_IPC4ZPH when testing multiple-pipeline-all

keqiaozhang, 2022-10-24:
**Describe the bug**
We Observed some different IPC times out on TGLU_UP_HDA_IPC4ZPH when testing `multiple-pipeline-all`.
if you run `multiple-pipeline-all` several times, you will get the following different IPC errors:
1. ipc timed out for 0x12070000|0x0 GLB_DELETE_PIPELINE
2. ipc error for msg 0x13050003|0x0,FW reported error: 6 - Unknown error while processing the request
3. FW reported error: 104 - Other failure of module instance initialization request,ipc error for msg 0x40070004|0x70015
4. ipc timed out for 0x13070002|0x0 GLB_SET_PIPELINE_STATE

**To Reproduce**
~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 25

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev [7d2d29c3fc54](https://github.com/thesofproject/linux/commit/7d2d29c3fc54)

SOF Branch: main [adce54445012](https://github.com/thesofproject/sof/commit/adce54445012)

Zephyr Commit: [zephyr-v3.2.0-441-ged661a6c6909](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.2.0-441-ged661a6c6909)
Platform: TGLU_UP_HDA_IPC4ZPH

**TPLG**
[image]

lgirdwood, 2022-10-24:
@keqiaozhang does this test work if we remove mixout pipelines ? i.e. we just keep the passthrough pipelines

lyakh, 2022-10-25:
@keqiaozhang sorry, I don't understand this report. It lists 5 different errors, of which we've just close several as "fixed," you say "run multiple-pipeline-all several times" and the reproduction rate is 100%. Could you clarify?

lgirdwood, 2022-10-25:
> @keqiaozhang sorry, I don't understand this report. It lists 5 different errors, of which we've just close several as "fixed," you say "run multiple-pipeline-all several times" and the reproduction rate is 100%. Could you clarify?

@keqiaozhang Can we split this to a GH BUG per open issue above ?

keqiaozhang, 2022-10-31:
@lgirdwood and @lyakh 
One week ago, there're various 'IPC errors ' on HDA platform when testing with multiple-pipeline-all, the error types are different each time. I didn't want to cause unnecessary panic by submitting so many bugs for one case. So I merged all errors in one bug.

But now almost all errors have been fixed, so I did a stress test this morning for this case and only observed one type of IPC times error, I have submitted a new bug https://github.com/thesofproject/sof/issues/6503 to track that issue. The bug can be closed now.

closed 2022-10-31 by keqiaozhang