thesofproject/sof #6471: [BUG] Initial FW ABI banner not found from cavstool output on TGLH_SKU0A70_HDA_ZEPHYR

miRoox, 2022-10-24:
**Describe the bug**
CI observed the initial FW ABI banner sometimes lost from cavstool (etrace) output on TGLH_SKU0A70_HDA_ZEPHYR.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg MODEL=TGLH_SKU0A70_HDA_ZEPHYR ~/sof-test/test-case/check-sof-logger.sh
```

**Reproduction Rate**
About 1/5

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/20a4b0774e58
    * SOF: https://github.com/thesofproject/sof/commit/73a413a38fc2
2) Name of the topology file
    * Topology: sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLH_SKU0A70_HDA

**Screenshots or console output**

**console log**
```
2022-10-20 21:31:30 UTC [REMOTE_COMMAND] sudo /usr/local/bin/sof-logger  -t -f 3 -l  /etc/sof/sof-tgl-h.ldc   >  /home/ubuntu/sof-test/logs/check-sof-logger/2022-10-20-21:31:24-17693/logger.data.txt  2>  /home/ubuntu/sof-test/logs/check-sof-logger/2022-10-20-21:31:24-17693/logger.error.txt  &
+ grep -i ERROR /home/ubuntu/sof-test/logs/check-sof-logger/2022-10-20-21:31:24-17693/logger.etrace.txt
2022-10-20 21:31:33 UTC [REMOTE_ERROR] Initial FW ABI banner not found in /home/ubuntu/sof-test/logs/check-sof-logger/2022-10-20-21:31:24-17693/logger.etrace.txt

2022-10-20 21:31:33 UTC [REMOTE_INFO] Log file logger.data.txt BEG::
 TIMESTAMP      (us)           DELTA  C# COMPONENT          LOCATION                      CONTENT	ktime=103.193s  @  2022-10-20 21:31:30 +0000 UTC
[   200240894.647] (           0.000) c0 dma-trace             src/trace/dma-trace.c:352  INFO DMA: FW ABI 0x3018000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-1542-g73a413a38fc2 zephyr:zephyr-v3.2.0-441-ged661a6c6909 src hash 0xb4cae834 (ldc hash 0xb4cae834)
[         131.875] (         131.875) c0 zll-schedule       src/schedule/zephyr_ll.c:268  INFO task add 0x9e07ea30 dma-trace-task <2b972272-c5b1-4b7e-926f-0fc5cb4c4690> priority 4 flags 0x0
[         212.031] (          80.156) c0 zll-schedule       ......../zephyr_domain.c:176  INFO zephyr_domain_register domain->type 1 domain->clk 2 domain->ticks_per_ms 38400 period 1000
[     2046306.429] (     2046094.375) c0 zll-schedule       ......../zephyr_domain.c:89   INFO ll timer avg 80, max 1838, overruns 0
2022-10-20 21:31:33 UTC [REMOTE_INFO] ::END log file logger.data.txt

2022-10-20 21:31:33 UTC [REMOTE_INFO] Log file logger.etrace.txt BEG::
/usr/local/bin/cavstool.py:706: DeprecationWarning: There is no current event loop
  asyncio.get_event_loop().run_until_complete(main())
INFO:cavs-fw:Existing driver "snd_sof_pci_intel_tgl" found
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 15 (GCAP = 0xffffffff)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Waiting forever for firmware handoff, FW_STATUS = 0xffffffff
2022-10-20 21:31:33 UTC [REMOTE_INFO] ::END log file logger.etrace.txt

2022-10-20 21:31:33 UTC [REMOTE_INFO] Log file logger.error.txt BEG::
GOOD: logger.error.txt was empty, no (std)err(or) output from that logger
2022-10-20 21:31:33 UTC [REMOTE_INFO] ::END log file logger.error.txt
```

marc-hb, 2022-10-24:
Very strange, the pseudo-"banner" `cavs-fw` _is_ there, it can be seen in the bug description!

It's not on the first line but that should not be a problem since sof-test fix https://github.com/thesofproject/sof-test/commit/736e79d526b7f

Some delay and race condition?

cc: @kv2019i

marc-hb, 2022-10-25:
> Some delay and race condition?

@enjiamai, @teburd, @andyross any idea?

teburd, 2022-10-25:
The INFO:cavs-fw logging comes from cavstool itself as its doing its pci memory mapping work.

The key line here is 

```
INFO:cavs-fw:Waiting forever for firmware handoff, FW_STATUS = 0xffffffff
```

cavstool never sees the firmware status change and never starts the winstream polling loop assuming that's how its setup.

Is cavstool actually needed still for TGL? mtrace is only used for newer IPC4 platforms I take it?

marc-hb, 2022-10-25:
> The key line here is FW_STATUS = 0xffffffff

Thanks @teburd for opening my eyes, I was looking everywhere except in the correct and most obvious place. I'm updating the title of this bug.

So the FW_STATUS is wrong. The firmware obviously booted because the DMA logs are found and look good.

I checked all the logs for this test across all devices and I copied below how they look when this test is passing (=most of the time).

So far both failures happen in the same configuration: TGLH_SKU0A70_HDA_ZEPHYR. Coincidence?

Both days it was the only test failing on that system, however other tests don't invoke cavstool.py so that does not say much.

> Is cavstool actually needed still for TGL? mtrace is only used for newer IPC4 platforms I take it?

Correct, cavstool will never be used in any officially supported configuration so this bug seems low priority. However we're about to release an "opt-in" Zephyr "demo" for TGL with IPC3 and cavstool is the only way to read Zephyr logs there: https://github.com/thesofproject/sof-bin/pull/106

```
INFO:cavs-fw:Existing driver "snd_sof_pci_intel_tgl" found
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Waiting forever for firmware handoff, FW_STATUS = 0x5
INFO:cavs-fw:FW alive, FW_STATUS = 0x5
[00:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7c000 initialized
[00:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7d000 initialized
*** Booting Zephyr OS build zephyr-v3.2.0-441-ged661a6c6909  ***
[00:00:00.000,000] <inf> main: SOF on intel_adsp_cavs25
[00:00:00.000,000] <inf> main: SOF initialized
4481113: SHM: FW ABI 0x3018000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-1555-gadce54445012 zephyr:zephyr-v3.2.0-441-ged661a6c6909 src hash 0x02960a2c (ldc hash 0x02960a2c)
```

keqiaozhang, 2023-05-30:
This is an IPC3 Zephyr platform, marked it as won't fix. Closing this bug.

closed 2023-05-30 by keqiaozhang