thesofproject/sof #6475: ADL - N : Resume failures with IPC causing suspend failures in stress tests

sathyap-chrome, 2022-10-24:


lgirdwood, 2022-10-24:
@sathyap-chrome do you have the FW log ? I would say we are broken at the resume time if we cant hookup a route to DMIC.

lgirdwood, 2022-10-24:
@mwasko fyi - pls assign.

sathyap-chrome, 2022-10-25:
@lgirdwood We dont have a clear reproduction scenario. FW trace wont help with suspend resume right ? ( since FW trace starts after DSP starts) - but will try how we can get it.

lgirdwood, 2022-10-25:
> @lgirdwood We dont have a clear reproduction scenario. FW trace wont help with suspend resume right ? ( since FW trace starts after DSP starts) - but will try how we can get it.

Sorry, I mean we need the log from part 1 - the resume that fails. 
```
0000:00:1f.3: ipc tx error for 0x30030000 (msg/reply size: 16/12): -22
0000:00:1f.3: sof_ipc3_route_setup: route DMIC1.IN -> BUF11.0 failed
0000:00:1f.3: sof_ipc3_set_up_all_pipelines: route set up failed
0000:00:1f.3: Failed to restore pipeline after resume -22
0000:00:1f.3: PM: pci_pm_resume+0x0/0xef returned -22 after 114680 usecs
0000:00:1f.3: PM: failed to resume async: error -22
```
This may shed some light on the previous suspend (where we may not have shutdown cleanly).

kv2019i, 2022-10-26:
@sathyap-chrome Would be good to have more samples of failing case logs. It would be curious to see if the problem is always:

1) TRIG_STOP fail on DMIC before suspend (ADSP hw potentially in undefined state)
2) suspend  (ADSP powered off but not cleanly as we have CTX_SAVE error, hw potentially in undefined state)
3) resume
4) error in setting up path for dmic

The HW power-cycling should of course help, but we've had cases in the past where e.g. the DMA hw has had issues that cannot be recovered by the above flow. The rootcause would then be the TRIG_STOP on DMIC.

kv2019i, 2022-10-26:
I was now able to reproduce once locally on a "Nereid" device. FYI @abonislawski as you have the same hw. 

No FW logs yet, but below are kernel logs showing the IPC sequence of a working and non-working case. Reproduction is once per day for me now, so will spend some time looking at the logs and then planning what to try next.  But FYI with others who have same hw, it is possible to trigger this bug with this setup. 

The sequence of IPC messages kernel sends seems valid. Here the failure is not with DMIC, but rather the first IPC sent to tear down the DSP pipelines for suspend (COMP_FREE for SSP1.IN). E.g. this line pair (not the timestamp showing 2sec, first line is last message to set up the DSP, second is start of suspend):
```
2022-10-26T12:01:59.399869Z DEBUG kernel: [ 1811.337622] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40020000
2022-10-26T12:02:01.380028Z DEBUG kernel: [ 1813.465770] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000
```

FAIL case
2022-10-26T12:01:59.399860Z DEBUG kernel: [ 1811.336836] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T12:01:59.399863Z DEBUG kernel: [ 1811.337023] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T12:01:59.399865Z DEBUG kernel: [ 1811.337218] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T12:01:59.399868Z DEBUG kernel: [ 1811.337408] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T12:01:59.399869Z DEBUG kernel: [ 1811.337622] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40020000
2022-10-26T12:02:01.380028Z DEBUG kernel: [ 1813.465770] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000
2022-10-26T12:02:01.401225Z DEBUG kernel: [ 1813.479839] snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.29.SSP1.IN freed
2022-10-26T12:02:01.449898Z DEBUG kernel: [ 1813.513956] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000
2022-10-26T12:02:01.962621Z ERR kernel: [ 1814.043734] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x30020000 (msg/reply size: 12/12)
2022-10-26T12:02:09.613173Z DEBUG kernel: [ 1821.701262] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40010000
2022-10-26T12:02:10.762189Z INFO kernel: [ 1822.845534] PM: suspend entry (s2idle)
2022-10-26T12:02:49.675106Z DEBUG kernel: [ 1823.028655] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30100000
2022-10-26T12:02:49.675109Z DEBUG kernel: [ 1823.028925] snd_sof:sof_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.29.SSP1.IN setup complete
2022-10-26T12:02:49.675110Z DEBUG kernel: [ 1823.028938] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000
2022-10-26T12:02:49.675112Z DEBUG kernel: [ 1823.029582] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000
2022-10-26T12:02:49.675113Z DEBUG kernel: [ 1823.030063] snd_sof:sof_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: widget SSP1.IN setup complete

OK case:
2022-10-26T11:57:28.430481Z DEBUG kernel: [ 1540.333981] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430484Z DEBUG kernel: [ 1540.334174] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430518Z DEBUG kernel: [ 1540.334333] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430520Z DEBUG kernel: [ 1540.334522] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430523Z DEBUG kernel: [ 1540.334705] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430525Z DEBUG kernel: [ 1540.334869] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430528Z DEBUG kernel: [ 1540.335031] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430531Z DEBUG kernel: [ 1540.335222] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430533Z DEBUG kernel: [ 1540.335385] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430536Z DEBUG kernel: [ 1540.335590] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430538Z DEBUG kernel: [ 1540.335783] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000
2022-10-26T11:57:28.430539Z DEBUG kernel: [ 1540.335984] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40020000
2022-10-26T11:57:29.994080Z DEBUG kernel: [ 1542.070562] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000
2022-10-26T11:57:32.035186Z DEBUG kernel: [ 1544.120273] snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.29.SSP1.IN freed
2022-10-26T11:57:32.245961Z DEBUG kernel: [ 1544.323824] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000
[... cut ...]
2022-10-26T11:57:32.699443Z DEBUG kernel: [ 1544.774999] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000
2022-10-26T11:57:32.722328Z DEBUG kernel: [ 1544.797890] snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF12.2 freed
2022-10-26T11:57:32.744949Z DEBUG kernel: [ 1544.820776] snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30210000
2022-10-26T11:57:32.791439Z DEBUG kernel: [ 1544.867276] snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget BUF12.1 freed

kv2019i, 2022-10-26:
This is not a fix, but seems to have impact to reproduction rate -> https://github.com/thesofproject/linux/pull/3965
When kernel added support for dynamic pipelines, the code to handle static pipelines was also extended to free DSP elements (pipes and components) at suspend. Older FW versions were skipped at runtime as freeing the elements led to failures. The referred patch modifies kernel to use the legacy patch for also SOF2.0 based firmware versions (ABI3.20) like is used in this bug report.

kv2019i, 2022-10-27:
I've been bisecting the issue on FW side (with unmodified SOF kernel, i.e. https://github.com/thesofproject/linux/pull/3965 NOT applied). It seems newer FW releases handle this better and at least the reproduction rate is lower:

```
2e045a49172fa0 topology1: CMakeLists: add     # PASS        (2.2 based, latest stable-v2.2)
fc4180899619ec rtnr: Support setting data blo # FAILURES (2.0 based, latest adl-004-drop-stable)
1153b51a42de5 config:ADL:Add DTS config       # FAILURES (2.0 based Intel ADL branch, this is the FW version in original bug report)
```

I will continue the bisect work, but early results show the failures could be caused a by a bug that is already fixed in FW mainline. Bisecting is complicated as reproduction rate is low, so the results need to be verified multiple times. So take above data as early indication, to be verified by more testing.

kv2019i, 2022-10-28:
Based on test results done with various older SOF firmware releases, I've pushed a polished kernel-side patch for upstream consideration: https://github.com/thesofproject/linux/pull/3965/

kv2019i, 2022-10-31:
Did a test today with CONFIG_DEBUG_BLOCK_FREE=y and did not find any invalid frees on heap with the FW version in original bug report.

kv2019i, 2022-11-02:
Based on test data, this issue is fixed in SOF mainline. Due to low repro rate, bisecting to a specific fix was not possible. There have been multiple fixes to dangling pointers in SOF mainline and in general more testing this year as dynamic pipelines has been enabled in more and more topologies used in regular daily/PR CI. Closing this issue as already fixed.