thesofproject/sof #6492: [BUG] TPLG2: the dai_index setting of Object.Dai does not take effect for the build

yongzhi1, 2022-10-26:
**Describe the bug**
When we set dai index for DAI object in the conf file, the setting does not take effect for alsatplg build, the issue is generic for all DAI types.

**To Reproduce**
Use SSP as an example:

```
--- a/tools/topology/topology2/cavs/cavs-nocodec.conf
+++ b/tools/topology/topology2/cavs/cavs-nocodec.conf
@@ -96,7 +96,7 @@ Object.Dai {
        }
        SSP."2" {
                id              2
-               dai_index       2
+               dai_index       3
                direction       "duplex"
                name            NoCodec-2
                default_hw_conf_id      0
```
Even dai_index is set to 3, the dai_index remains 2 in the build result, i.e. DAI instance number is always being used as dai_index.

**Reproduction Rate**
100%
**Expected behavior**
We should be able to use dai_index.

ranj063, 2022-10-26:
@yongzhi1 you should be able to change the dai_index with
```
SSP.3 {
....
}
```

For the SSP DAI class, dai_index is the unique attribute and needs to be set when initializing like above

yongzhi1, 2022-10-26:
Hi, @ranj063 , so if the unique instance number N is the dai_index and we can set otherwise, should we remove all dai_index from conf level?

ranj063, 2022-10-26:
> Hi, @ranj063 , so if the unique instance number N is the dai_index and we can not set otherwise, should we remove all dai_index from conf level?

yes please

yongzhi1, 2022-10-27:
Sure, https://github.com/thesofproject/sof/pull/6493, thanks @ranj063 for clarification!!

lgirdwood, 2022-10-27:
@juimonen fyi -related to your DAI ID issue ?

juimonen, 2022-10-28:
> @juimonen fyi -related to your DAI ID issue ?

no this is different and @yongzhi1 has a fix for this.

closed 2022-10-28 by yongzhi1