thesofproject/sof #6516: [BUG] ipc timed out for 0x13000004|0x1 when playback on MTL platforms

miRoox, 2022-11-02:
**Describe the bug**
playback at first time fails with ipc timed out for 0x13000004|0x1.

NOTE: On the main branch, the runtime power management is not ready, so we disabled runtime PM on our MTL DUTs with the kernel option `options snd-sof-pci sof_pci_debug=0x01`.

**To Reproduce**
```
aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/69c5960521f2
    * SOF: https://github.com/thesofproject/sof/commit/ef62a508930c
2) Name of the topology file
    * Topology: cavs-mtl-sdw-dmic.tplg, sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_SDW, MTLP_RVP_NOCODEC
      (MTLP_RVP_HDA should be same but we are blocked by https://github.com/thesofproject/linux/issues/3985)

**Screenshots or console output**

**dmesg**
```
[  205.020033] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  205.523729] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13000004|0x1
[  205.524270] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  205.524282] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  205.524352] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x1|0x0, target: 0x33000000|0x1|0x0, ctl: 0x3
[  205.524431] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  205.524499] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  205.524554] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  205.524595] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  205.524656] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  205.524707] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  205.524769] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  205.524814] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  205.524883] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[  205.525000] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[  205.525074] kernel:  Port0: ASoC: trigger FE cmd: 1 failed: -110
```

keqiaozhang, 2022-11-29:
This issue disappeared after the Zephyr update. But let's keep it open for one more day, I will close it if it doesn't happen tomorrow.

keqiaozhang, 2022-11-30:
This issue is not reproducible in CI, closing it.

closed 2022-11-30 by keqiaozhang