thesofproject/sof #6526: ADL-PS: DSP failed to boot with extended rom status:  0x6000021

gongjun-song, 2022-11-03:
**Describe the bug:**
After the platform boot, the DSP initialization fails and the audio does not work.
```
test@ADL01:~$ sudo aplay -l && sudo arecord -l
aplay: device_list:274: no soundcards found...
arecord: device_list:274: no soundcards found...

test@ADL01:~$ sudo lspci -s 1f.3 -v
0000:00:1f.3 Multimedia audio controller: Intel Corporation Device 51c9 (rev 01)
        Subsystem: Intel Corporation Device 7270
        Flags: bus master, fast devsel, latency 64, IRQ 16
        Memory at 6005448000 (64-bit, non-prefetchable) [size=16K]
        Memory at 6005100000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [50] Power Management version 3
        Capabilities: [80] Vendor Specific Information: Len=14 <?>
        Capabilities: [60] MSI: Enable- Count=1/1 Maskable- 64bit+
        Kernel driver in use: sof-audio-pci-intel-tgl
        Kernel modules: snd_hda_intel, snd_sof_pci_intel_tgl
```

**dmesg:**
```
[   23.828604] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire machine driver sof_sdw topology sof-adl-rt711-4ch.tplg
[   23.828605] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[   23.828616] sof-audio-pci-intel-tgl 0000:00:1f.3: loading firmware
[   23.830776] sof-audio-pci-intel-tgl 0000:00:1f.3: request_firmware intel/sof/sof-adl.ri successful
[   23.830781] sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 1 size 0x1a0
[   23.830784] sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 0 size 0x50
[   23.830785] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:0:0-5a883
[   23.830787] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:24:0 Kernel ABI 3:23:0
[   23.830788] sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 2 size 0x70
[   23.830790] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: used compiler XCC 12:0:8 xcc used optimization flags -O2
[   23.830795] sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 3 size 0x30
[   23.830797] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   23.830798] sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 4 size 0x20
[   23.830799] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: DBG_ABI 5:3:0
[   23.830800] sof-audio-pci-intel-tgl 0000:00:1f.3: found sof_ext_man header type 5 size 0x20
[   23.830801] sof-audio-pci-intel-tgl 0000:00:1f.3: manifest can hold up to 3 config elements
[   23.830802] sof-audio-pci-intel-tgl 0000:00:1f.3: get index 0 token 1 val 384
[   23.830803] sof-audio-pci-intel-tgl 0000:00:1f.3: get index 1 token 2 val 0
[   23.830804] sof-audio-pci-intel-tgl 0000:00:1f.3: get index 2 token 0 val 0
[   23.830805] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 1 -> 2
[   23.830825] sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[   23.830832] sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   23.830846] sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   23.830851] sof-audio-pci-intel-tgl 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[   23.830852] sof-audio-pci-intel-tgl 0000:00:1f.3: booting DSP firmware
[   23.830950] sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x0
[   23.830952] sof-audio-pci-intel-tgl 0000:00:1f.3: periods:1
[   23.831079] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x40000 successful
[   23.831098] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x40000 successful
[   23.831100] sof-audio-pci-intel-tgl 0000:00:1f.3: period_bytes:0x0
[   23.831101] sof-audio-pci-intel-tgl 0000:00:1f.3: periods:1
[   23.831148] sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting iteration 0 of Core En/ROM load...
[   23.831691] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
[   23.831718] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0e successful
[   23.831721] sof-audio-pci-intel-tgl 0000:00:1f.3: unstall/run core: core_mask = 1
[   23.831725] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   23.832240] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0xd4]=0x80000000 successful
[   23.841431] calling  rt711_sdca_sdw_driver_init+0x0/0x1000 [snd_soc_rt711_sdca] @ 531
[   23.841549] rt711-sdca sdw:0:025d:0711:01: rt711_sdca_init
[   23.841584] rt711-sdca sdw:0:025d:0711:01: probe complete
[   23.841616] initcall rt711_sdca_sdw_driver_init+0x0/0x1000 [snd_soc_rt711_sdca] returned 0 after 171 usecs
[   24.132289] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x80000]=0x6000021 timedout
[   24.132321] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
[   24.132328] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[   24.132330] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   24.132331] sof-audio-pci-intel-tgl 0000:00:1f.3: Attempting iteration 1 of Core En/ROM load...
[   24.132336] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0f successful
[   24.132345] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf010f0e successful
[... 36 lines cut ...]
[   24.785684] sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1d003c timedout
[   24.785691] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   24.785694] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[   24.785909] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 3 -> 0
[   24.785910] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -110
```

**To Reproduce:**
After booting, DSP init failed.

**Reproduction Rate:**
100%

**Environment:**
Kernel Branch: topic/sof-dev
Kernel Commit: 69c5960521f220ead

**SOF Branch:** 
sof-v2.2

**Platform:**
ADL-PS

**dmesg:**
[dmesg.log](https://github.com/thesofproject/linux/files/9910093/dmesg.log)

plbossart, 2022-11-03:
@mwasko was this firmware tested with the slim driver and test suite on ADL-PS RVP? I'd like to check if there is an issue in general or if this is an issue with Linux flow. Thanks!

mwasko, 2022-11-04:
> @mwasko was this firmware tested with the slim driver and test suite on ADL-PS RVP? I'd like to check if there is an issue in general or if this is an issue with Linux flow. Thanks!

We have tested the FW binaries with ADL-P, but as long as the RVP is not ADL-N the key authentication should pass. @gongjun-song  have you checked whether you ADL-PS RVP is a production HW? If it is not then you need Intel debug signed binaries and those are not available to public.

plbossart, 2022-12-13:
@gongjun-song is this issue still relevant? How can we have a P1 issue with no update in one month? @lgirdwood FYI

gongjun-song, 2022-12-13:
@plbossart, @lgirdwood, @mengdonglin 
I contacted the engineer of IOTG team and they were very slow to reply.
I think we can close this issue for now; I will contact IOTG again and wait for their reply.

closed 2022-12-13 by plbossart

samtinel, 2022-12-20:
I am encountering the same issue on Linux 6.0 and whatever stabile 5.15. I dont

samtinel, 2022-12-20:
I am encountering the same issue on Linux 6.0 and stable 5.15. I dont have reason to believe it ever worked though.
This laptop is a fairly new Cannon Lake laptop.

alsa: no soundcards
lspci: https://paste.debian.net/hidden/2627d90b
relevant log lines: https://paste.debian.net/hidden/61cb02de/

If I can aid in debugging this despite my limited knowledge I am happy to help!

gongjun-song, 2022-12-21:
@samtinel , they verified it on Yocto and it works well. But I am not sure whether they validated it on Ubuntu. 
I'm still waiting for a reply, I'll update if I have new message.

JockeTF, 2023-01-17:
I fixed this by re-enabling Intel Management Engine, [as hinted to in the Gentoo Forums](https://forums.gentoo.org/viewtopic-p-8752857.html).

plbossart, 2023-01-17:
Yes, it's a well-known dependency https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#make-sure-the-me-is-enabled