thesofproject/sof #6534: [BUG] mtrace doesn't work on MTL

miRoox, 2022-11-04:
**Describe the bug**
mtrace reports `[Errno 2] No such file or directory: '/sys/kernel/debug/sof/mtrace/core0'` on MTL

**To Reproduce**
```
sudo ./mtrace-reader.py
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/5af7c07c632a
    * SOF: https://github.com/thesofproject/sof/commit/1dffde13ba4b
2) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_SDW and MTLP_RVP_NOCODEC

**Screenshots or console output**
```
Traceback (most recent call last):
  File "/usr/local/bin/mtrace-reader.py", line 20, in <module>
    fd = os.open(MTRACE_FILE, os.O_RDONLY)
FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/debug/sof/mtrace/core0'
```

mengdonglin, 2022-11-04:
Trace is enabled on kernel side
https://github.com/thesofproject/kconfig/blob/master/sof-dev-defconfig
CONFIG_SND_SOC_SOF_DEBUG_ENABLE_FIRMWARE_TRACE=y

How can we make sure mtrace is enabled when building MTL FW?  @miRoox @kv2019i @juimonen @jsarha

kv2019i, 2022-11-04:
It should be enabled by default. Checking current SOF "main" and building for mtl gives me:

$ grep MTRACE ../build-mtl/zephyr/.config 
CONFIG_LOG_BACKEND_ADSP_MTRACE=y
CONFIG_LOG_BACKEND_ADSP_MTRACE_OUTPUT_TEXT=y

Let me test this myself as well.

kv2019i, 2022-11-04:
Issue solved, kernel was disabling mtrace via sof_debug.

closed 2022-11-04 by kv2019i