thesofproject/sof #6544: [BUG][IPC4][MTL] FW reported error: 104 when testing capture on MTL-SDW platform

keqiaozhang, 2022-11-07:
**Describe the bug**
This issue happened on MTL-SDW platform with `mtl-002-drop-stable`. 
FW reported error: 104 - Other failure of module instance initialization request. This issue only happens when testing capture. No such issue when testing playback. The reproduction rate is 100%

**dmesg**
```
[  273.585953] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget copier.ALH.3.1 instance 0 - pipe 3 - core 0
[  273.585957] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x40000004|0x36: MOD_INIT_INSTANCE [data size: 216]
[  273.586255] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x60000068|0x36: MOD_INIT_INSTANCE
[  273.586259] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 104 - Other failure of module instance initialization request
[  273.586563] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x40000004|0x36
[  273.586804] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to create module copier.ALH.3.1
[  273.586829] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12000000|0x0: GLB_DELETE_PIPELINE
[  273.586963] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x32000000|0x0: GLB_DELETE_PIPELINE
[  273.586982] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x12000000|0x0: GLB_DELETE_PIPELINE
[  273.586987] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-mtl 0000:00:1f.3: widget pipeline.3 freed
[  273.586990] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to set up connected widgets
```

**mtrace**
```
[00:02:49.616,000] <inf> ipc: ipc_cmd: rx	: 0x11000002|0x0
[00:02:49.616,000] <inf> pipe: pipeline_new: pipeline new pipe_id 0 priority 0
[00:02:49.616,000] <inf> ipc: ipc_cmd: rx	: 0x40000004|0x36
[00:02:49.616,000] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:02:49.616,000] <err> dai_comp: dai_new(): dai_get() failed to create DAI.
[00:02:49.616,000] <err> copier: comp:0 0x40000 failed to create dai
[00:02:49.616,000] <err> copier: unenable to create dai
[00:02:49.616,000] <err> ipc: error: failed to init module 4 : 0
[00:02:49.616,000] <err> ipc: ipc4: MODULE_MSG failed with err 104
[00:02:49.617,000] <inf> ipc: ipc_cmd: rx	: 0x12000000|0x0
[00:02:49.617,000] <inf> pipe: pipeline_free: pipe:0 0x0 pipeline_free()Terminated
```

**To Reproduce**
`arecord   -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/null -v -q`

**Reproduction Rate**
100%

**TPLG**
[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel Branch: topic/sof-dev
    * Kernel Commit: [cb13f81e00b4](https://github.com/thesofproject/linux/commit/cb13f81e00b4)

    * SOF Branch: mtl-002-drop-stable
    * SOF Commit: [9fbcce13d9b8](https://github.com/thesofproject/sof/commit/9fbcce13d9b8)

2) Name of the topology file
    * Topology: cavs-mtl-sdw-dmic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: mtl-rvp-sdw

ranj063, 2022-11-07:
@juimonen @keqiaozhang this is the fix for this issue that needs to be submitted to zephyr
```
diff --git a/dts/xtensa/intel/intel_adsp_cavs25.dtsi b/dts/xtensa/intel/intel_adsp_cavs25.dtsi
index 94f28bc675..ed589dc532 100644
--- a/dts/xtensa/intel/intel_adsp_cavs25.dtsi
+++ b/dts/xtensa/intel/intel_adsp_cavs25.dtsi
@@ -287,9 +287,16 @@
                 *
                 * https://github.com/zephyrproject-rtos/zephyr/pull/50287#discussion_r974591009
                 */
-               alh0: alh1: alh@71000 {
+               alh0: alh0@71000 {
                        compatible = "intel,alh-dai";
-                       reg = <0x00071000 0x00071200>;
+                       reg = <0x00071000 0x00071800>;
+
+                       status = "okay";
+               };
+
+               alh1: alh1@71000 {
+                       compatible = "intel,alh-dai";
+                       reg = <0x00071000 0x00071800>;
 
                        status = "okay";
                };
```

juimonen, 2022-11-07:
Ok, I'll try to take care of this.

keqiaozhang, 2022-11-09:
Confirmed that this issue has been fixed on mtl-002-stable-drop branch.
https://github.com/thesofproject/sof/commit/b7d911d3ebde5f893e1b03723b0a176e66c684ee

closed 2022-11-09 by keqiaozhang