thesofproject/sof #6545: [BUG] xrun error when simply playback on MTLP_RVP_NOCODEC

miRoox, 2022-11-07:
**Describe the bug**
CI observed xrun error when simply playback on MTLP_RVP_NOCODEC.

**To Reproduce**
```
aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
```

**Reproduction Rate**
50%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev | https://github.com/thesofproject/linux/commit/5af7c07c632a
    * SOF: main | https://github.com/thesofproject/sof/commit/1dffde13ba4b
2) Name of the topology file
    * Topology: sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_NOCODEC

**Screenshots or console output**

**mtrace**
```
[00:03:28.987,000] <inf> pipe: pipeline_prepare: pipe:0 0x0 pipe prepare
[00:03:28.987,000] <inf> dai_intel_ssp: dai_ssp_pre_start: dai_ssp_pre_start
[00:03:28.987,000] <inf> dai_intel_ssp: dai_ssp_start: dai_ssp_start
[00:03:28.987,000] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 29000, reg 30, mask 100, value 100
[00:03:28.987,000] <wrn> dai_comp: comp:1 0x40001 dai_copy(): nothing to copy
[00:03:28.987,000] <inf> pipe: pipeline_trigger: pipe:0 0x0 pipe trigger cmd 7
[00:03:28.987,000] <inf> ll_schedule: zephyr_ll_task_schedule_common: task add 0xa00dd9c0 0x20180U priority 0 flags 0x0
[00:03:28.988,000] <wrn> dai_comp: comp:1 0x40001 dai_copy(): nothing to copy
[00:03:28.988,000] <err> copier: comp:0 0x40000 failed to find dai comp or sink pipeline not running.
[00:03:28.988,000] <err> copier: comp:0 0x40000 failed to find dai comp or sink pipeline not running.
[00:03:28.988,000] <inf> host_comp: host_get_copy_bytes_normal: comp:0 0x40000 no bytes to copy, 384 free in buffer, 0 available in DMA
[00:03:28.989,000] <wrn> dai_comp: comp:1 0x40001 dai_copy(): Copy_bytes 0 + avail bytes 240 < period bytes 384, possible glitch
[00:03:28.989,000] <wrn> dai_comp: comp:1 0x40001 dai_copy(): nothing to copy
[00:03:28.990,000] <err> dma_dw_common: xrun detected
[00:03:28.990,000] <err> dai_comp: comp:1 0x40001 dai_report_xrun(): underrun due to no data available
[00:03:28.990,000] <err> component: comp:1 0x40001 comp_underrun(): dev->comp.id = 262145, source->avail = 384, copy_bytes = 0
[00:03:28.990,000] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 6
[00:03:28.990,000] <err> pipe: pipe:1 0x0 pipeline_copy(): ret = -61, start->comp.id = 262145, dir = 1
[00:03:28.990,000] <err> pipe: pipe:1 0x0 pipeline_task(): xrun recovery failed! pipeline is stopped.
```
[full_mtrace.txt](https://github.com/thesofproject/sof/files/9948300/full_mtrace.txt)

miRoox, 2022-11-07:
This error also happened on `mtl-002-drop-stable` branch (https://github.com/thesofproject/sof/commit/9fbcce13d9b8)

mengdonglin, 2022-11-09:
This error can be reproduced on mtl-002-drop-stable branch (commit https://github.com/thesofproject/sof/commit/b7d911d3ebde5f893e1b03723b0a176e66c684ee). Report ID 17485.

miRoox, 2022-11-15:
@ranj063 provided a zephyr patch for mtl-002 to verify if this issue is related to DW DMA driver:
```diff
diff --git a/drivers/dma/dma_dw_common.c b/drivers/dma/dma_dw_common.c index 66957f905a..bcbf12f5d0 100644
--- a/drivers/dma/dma_dw_common.c
+++ b/drivers/dma/dma_dw_common.c
@@ -782,8 +782,12 @@ int dw_dma_get_status(const struct device *dev, uint32_t channel,
        }
#if CONFIG_DMA_DW_HW_LLI
        if (!(dw_read(dev_cfg->base, DW_DMA_CHAN_EN) & DW_CHAN(channel))) {
-               LOG_ERR("xrun detected");
-               return -ENODATA;
+               dw_write(dev_cfg->base, DW_DMA_CHAN_EN, DW_CHAN_UNMASK(channel));
+
+               if (!(dw_read(dev_cfg->base, DW_DMA_CHAN_EN) & DW_CHAN(channel))) {
+                       LOG_ERR("xrun detected");
+                       return -ENODATA;
+               }
        }
#endif
        return 0;
```

I tried this patch and the issue will be gone. 
Most playback test cases passed while many capture cases still failed.
The internal CI test report (https://sof-ci.sh.intel.com/#/result/planresultdetail/17679) shows the first failed case is check-playback-3times, but I cannot reproduce the error by running it independently, it seems related to the previous cases.

[sof-mtl.zip](https://github.com/thesofproject/sof/files/10011903/sof-mtl.zip)
[mtrace-check-playback-3times.txt](https://github.com/thesofproject/sof/files/10011907/mtrace-check-playback-3times.txt)

lgirdwood, 2022-11-16:
@ranj063 @juimonen @lyakh re the above patch, IIUC the DMAC state machine may require time between channel enable/disable changes. i.e. it may have to clear FIFOS and pending requests.

abonislawski, 2022-11-18:
Issue fixed here:

- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/52354

miRoox, 2022-11-22:
It's fixed on mtl-002 and currently cannot be reproduced on main branch (due to #6516).
We can reopen it if we see it again.

closed 2022-11-22 by miRoox