thesofproject/sof #6561: sof error on computestick stck1a32wfc lubuntu sound crackling[BUG]

naxil, 2022-11-09:
**Describe the bug**
this error on dmesg and sound is dirty (and alsa mixer not show volume etc)
sudo dmesg | grep error
[   15.884563] sof-audio-acpi-intel-byt 80860F28:00: error: no matching ASoC machine driver found - aborting probe
[   15.885929] sof-audio-acpi-intel-byt 80860F28:00: error: failed to get machine info -19
[   15.887282] sof-audio-acpi-intel-byt 80860F28:00: error: sof_probe_work failed err: -19

**To Reproduce**
install lubuntu 22.10 on intel computestick  stck1a32wfc

**Reproduction Rate**
always but sound works

**Expected behavior**
clear sound? alsamixer show something?

**Impact**
little problem)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA} 5.19.0-23-generic
    * SOF: {SHA} idk how to show it
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lubuntu 22.10

plbossart, 2022-11-10:
can you please attach the log of 'alsa-info'?

I am not sure how the sound could be 'crackling', with this error there would be no sound card created, so how would one even play a stream?

naxil, 2022-11-10:
yes it upload the verbose log on http://alsa-project.org/db/?f=b52686199dfd4e1f81cde2449adcdbcd86eefed7

plbossart, 2022-11-10:
There's no usable information in the logs @naxil, I didn't find any known ACPI HID for audio codecs and the dmesg log is empty.

Let's try something different. please install acpica-tools and extract the DSDT table

````
sudo apt install acpica-tools
sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
iasl -d dsdt.dat
cp dsdt.dsl dsdt.dsl.txt
````

and attach dsdt.dsl.txt

plbossart, 2022-11-10:
Also please clarify in your compute stick has a 3.5mm jack or not. If you only have an HDMI connector, then the issue is with the snd_hdmi_lpe_audio driver. The SOF driver does not deal with HDMI on BYT/CHT.

naxil, 2022-11-10:
[dsdt.dsl.txt](https://github.com/thesofproject/sof/files/9984404/dsdt.dsl.txt)
yes this is my file... after your command... sorry but what i have do? its a code about firmware drivers? 
yes i think the stick only send audio vai hdmi and i liste it via jack but conntected to the tv... BT pods have same crackling...

plbossart, 2022-11-10:
The DSDT confirms that there is no audio codec on this device

````
Device (RTEK)
            {
                Name (_ADR, Zero)  // _ADR: Address
                Name (_HID, "10EC5640" /* Realtek I2S Audio Codec */)  // _HID: Hardware ID
                Name (_CID, "10EC5640" /* Realtek I2S Audio Codec */)  // _CID: Compatible ID
                Name (_DDN, "RTEK Codec Controller ")  // _DDN: DOS Device Name
                Name (_UID, One)  // _UID: Unique ID
                Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
                {
                    Name (SBUF, ResourceTemplate ()
                    {
                        I2cSerialBusV2 (0x001C, ControllerInitiated, 0x00061A80,
                            AddressingMode7Bit, "\\_SB.I2C2",
                            0x00, ResourceConsumer, , Exclusive,
                            )
                        GpioInt (Edge, ActiveHigh, ExclusiveAndWake, PullNone, 0x0000,
                            "\\_SB.GPO2", 0x00, ResourceConsumer, ,
                            )
                            {   // Pin list
                                0x0004
                            }
                    })
                    Return (SBUF) /* \_SB_.I2C2.RTEK._CRS.SBUF */
                }

                Method (_STA, 0, NotSerialized)  // _STA: Status
                {
                    Return (Zero)
                }
````

The _STA return of zero means the device is not exposed, and hence the SOF driver cannot find a codec. The error you see is normal - it's a feature not a bug.

I am not sure what the userpace setup is, what you should try is to use "speaker-test -Dhw:0,0 -c2 -r48000 -t sine"  and check the results. Try first with the Bluetooth device using the output of your TV or receiver. If you hear a nice sine wave, then the problem is either in userspace or with your BT headset.

Clearly it's not an SOF bug at all.

naxil, 2022-11-11:
thank you very much for your help, and sorry if my noob.situation take your time.... 
so i can't use sof at all on this device...
i

closed 2022-11-11 by lgirdwood