thesofproject/sof #6570: [BUG] ipc timed out for 0x45010004|0x8020006 when multiple-pipeline-capture on ADLP_RVP_NOCODEC_IPC4ZPH

miRoox, 2022-11-10:
**Describe the bug**
CI observed ipc timed out for 0x45010004|0x8020006 when multiple-pipeline-capture on ADLP_RVP_NOCODEC_IPC4ZPH 

**To Reproduce**
```
TPLG=/lib/firmware/intel/avs-tplg/cavs-tgl-nocodec.tplg MODEL=ADLP_RVP_NOCODEC_IPC4ZPH ~/sof-test/test-case/multiple-pipeline.sh -f c -l 50
```

**Reproduction Rate**
About 50%? (need further investigation)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev | https://github.com/thesofproject/linux/commit/551c11c310ac
    * SOF: main | https://github.com/thesofproject/sof/commit/0857ed68b903
2) Name of the topology file
    * Topology: cavs-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP_RVP_NOCODEC_IPC4ZPH 

**Screenshots or console output**

**dmesg**
```
[ 3708.921272] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x45010004|0x8020006: MOD_BIND
[ 3709.428479] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x45010004|0x8020006
[ 3709.428715] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[ 3709.428847] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 3709.428856] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 3709.428860] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3709.428887] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 3709.428906] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3709.428931] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0xc5010004|0x8020006|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 3709.428948] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3709.428958] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3709.428967] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 3709.428977] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 3709.428996] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 3709.429029] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 3709.429038] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3709.429053] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to bind modules copier.module.8.2:1 -> gain.17.1:0
[ 3709.429056] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
```

keqiaozhang, 2022-11-14:
@miRoox , please help to check with the latest main branch.

miRoox, 2022-11-15:
No reproduction. It should be fixed by #6578

closed 2022-11-15 by miRoox