thesofproject/sof #6574: [BUG] 10_05_TestIpcStressUnderLoad24000Hz24b32b8ch BBB value is smaller than acceptable value

antonirx, 2022-11-10:
**Describe the bug**
BBB value is smaller than acceptable value in 10_05_TestIpcStressUnderLoad24000Hz24b32b8ch
Glitch in wave.

**To Reproduce**
Run tests with Diagnostic driver:
10_05_TestIpcStressUnderLoad24000Hz24b32b8ch

**Expected behavior**
No bugs as shown below

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: 4ab225b5a7c730e5a78e4538624d3953dd83ac6f
    * REPO: ec2dc6cec7ef6615beda3cbc6091d49d4eee2a6e
2) Name of the platform(s) on which the bug is observed.
    TGL IPC3

**Screenshots or console output**
[image]

lgirdwood, 2022-11-11:
@antonirx I think for main branch its best we focus on IPC4 tests. @mwasko fyi.

wszypelt, 2023-01-02:
TGL issue does not occur
Tested FW hash: [4943aaa](https://github.com/thesofproject/sof/commit/4943aaa8907149cf6c02e9046e2ef5d40ba72d74)

closed 2023-01-02 by wszypelt