thesofproject/sof #6576: [FEATURE] Clean-up topology2/cavs to make it user-friendly and maintainable

plbossart, 2022-11-10:
The topology2 framework allows for code reuse, modularity, macros and arithmetic to determine settings.

Sadly, our existing cavs topologies are 'developer' level and a number of ground rules were forgotten.

a) all controls and PCM names SHALL have a meaningful name. 

"volume playback 1' is not acceptable, each PGA and control shall be clearly identified with a pipeline information and position in the pipeline.
The use of 'master' SHALL be revisited. There SHALL not be more than one 'master' control on a path to an endpoint.

when possible, all PCM and control names SHALL be backwards compatible with existing UCM files. Don't be creative.

b) topology2/cavs files SHALL have the same level of modularity as topology1. We know e.g. for HDaudio/SoundWire/I2S that we'll have tons of variations (jack, no jack, echo reference, one amp, 2 amps, BT, HDMI with 3/4 streams). We have to track and make sure all variability in topology1 is supported in topology2

c) the indices used for pipelines and components matter. in topology1 we could easily identify when a pipeline was connected to SSP0. This information is lost in topology2. so the rule is that just by using at the topology viewer, anyone SHALL be able to know which PCM device to use to play on a specific DAI.

d) all generated topology2 files SHALL be named with the conventions used in the kernel. A topology file that is not known to the kernel SHALL be moved to a developer directory. 

Common sense really.

plbossart, 2022-11-10:
@lgirdwood @ranj063 @bardliao @marc-hb @yongzhi1  FYI

marc-hb, 2022-11-10:
I think examples would be great. Pointers to PRs fixing this sort of issues even better.

ranj063, 2022-11-14:
@plbossart this is the first clean up to build the right binaries in the right folder https://github.com/thesofproject/sof/pull/6592

ranj063, 2022-11-17:
https://github.com/thesofproject/sof/pull/6627 addresses the third bullet in this requirement

ranj063, 2022-11-17:
https://github.com/thesofproject/sof/pull/6592 is for the 4th

ranj063, 2022-12-19:
> "volume playback 1' is not acceptable, each PGA and control shall be clearly identified with a pipeline information and position in the pipeline.

@plbossart Is the position in the pipeline really necessary? We do not have a case where we have more than one valud component in the same pipeline. And what should the protocol be for naming. "<Direction> Volume <Pipeline ID>" where Direction will be playback/capture OK?

plbossart, 2022-12-19:
What we agreed is that the pipeline ID is not needed for the host/dai, but will be present for all other components - with the pipeline ID + instance ID. This is different to the position in the pipeline as written above, that is clearly not a good idea in hindsight. I insist that we need to have something else than the pipeline ID for other components, some unique ID such as the instance ID is necessary. We could have a gain that prevents saturation before a processing and compensation later on, it's classic for fixed-point processing

marc-hb, 2022-12-20:
>  all generated topology2 files SHALL be named with the conventions used in the kernel. 

If a name is wrong, validation MUST NOT silently rename it and "paint" the test results green and leave customers stuck with a non-working system. The purpose of validation is to find bugs before customers do, not to hide them.

If a name is wrong, tests must do their job and FAIL in order to report the issue.

lgirdwood, 2022-12-20:
@ranj063 @jsarha pls update the status check boxes for what has been merged etc. Thanks

plbossart, 2022-12-21:
Not following the question @ranj063, sorry

ranj063, 2022-12-21:
> Not following the question @ranj063, sorry

@plbossart how would you like the names to look like. Something like "Playback Volume 5 1" , "Capture Volume 6 1" etc?

plbossart, 2022-12-21:
It's certainly better than what we currently have

````
numid=39,iface=MIXER,name='gain.0.1 1 Playback Volume 0'
numid=40,iface=MIXER,name='gain.1.1 2 Main Playback Volume'
````
 or

````
numid=60,iface=MIXER,name='PGA1.0 1 Master Playback Volume'
numid=62,iface=MIXER,name='PGA2.0 2 Master Capture Switch'
numid=61,iface=MIXER,name='PGA2.0 2 Master Capture Volume'
numid=63,iface=MIXER,name='PGA3.1 3 Master Playback Volume'
numid=71,iface=MIXER,name='PGA30.0 30 Playback Volume'
numid=72,iface=MIXER,name='PGA31.0 31 Playback Volume'
numid=66,iface=MIXER,name='PGA5.0 5 Master Capture Switch'
numid=65,iface=MIXER,name='PGA5.0 5 Master Capture Volume'
numid=68,iface=MIXER,name='PGA6.0 6 Master Playback Volume'
numid=69,iface=MIXER,name='PGA7.0 7 Master Playback Volume'
numid=70,iface=MIXER,name='PGA8.0 8 Master Playback Volume'

````

The only open in my mind is whether we want a "Master" volume for 'dai' pipelines. that would certainly be easier to identify when we have multiple PCMs, but without additional information on the DAI itself it may be difficult to use.

ranj063, 2022-12-21:
hmm I wonder where the "gain.0.1" or "gain.1." is coming from. Looks like they are the widget names which have the mixer. Is this something alsatplg appends automatically?

plbossart, 2022-12-21:
no idea where this comes from @ranj063, but the topology was sof-tgl-rt711-rt1316-rt714.tplg so it's probably faster if you look at the conf file and how the names are created.

kv2019i, 2023-03-09:
Two days left before branch, I think we need to move this to v2.6? @jsarha @plbossart @lgirdwood ?

jsarha, 2023-03-10:
> Two days left before branch, I think we need to move this to v2.6? @jsarha @plbossart @lgirdwood ?

There is one merged PR [1] and couple of more PRs [2] ready for review, and after them there is finally all the tools ready to solve the first tic-box "all controls and PCM names SHALL have a meaningful name.", however I do not have the knowledge to come up with these meaningful names.

Then there is couple of more tic-boxes that have not even been started, so yes this wont get solved in two days.

[1] https://github.com/alsa-project/alsa-utils/pull/189
[2] https://github.com/thesofproject/sof/pull/7227
      https://github.com/thesofproject/linux/pull/4226

lgirdwood, 2023-05-10:
@jsarha @ranj063 @plbossart on track for v2.6 ?

ranj063, 2023-05-17:
#7631 and #7640 complete the third bullet

alex-cri, 2023-05-30:
@plbossart, with PR #7682 merged is there anything further required before closing this bug?

ranj063, 2023-05-30:
> @plbossart, with PR #7682 merged is there anything further required before closing this bug?

@alex-cri 2 more pending https://github.com/thesofproject/sof/pull/7694 and @7704. And I think @jsarha has a couple of more coming for the chrome topology and the sdw topologies

closed 2023-06-28 by kv2019i