thesofproject/sof #6589: [BUG i.MX] warning: the int symbol CORE_COUNT has a non-int default MP_NUM_CPUS

dbaluta, 2022-11-14:
**Describe the bug**

When compiling the firmware for i.MX8 platforms we get the following warning:
```
(warning: the int symbol CORE_COUNT (defined at src/platform/Kconfig:307) has a non-int default MP_NUM_CPUS (undefined))
```

**To Reproduce**

$ ./scripts/xtensa-build-all.sh imx8m

**Reproduction Rate**

Always

**Expected behavior**
Compiling the i.MX8 firmware should not produce a warning.

**Impact**
It is annoying.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: dont'care
    * SOF: main
2) Name of the topology file
    * Topology: don't care
3) Name of the platform(s) on which the bug is observed.
    * Platform: i.MX8/i.MX8X/i.MX8M

paulstelian97, 2022-11-14:
I have done a little bit of digging, the warning comes from the fact that kconfiglib does not check the "if" conditions on defaults when validating. When actually generating the configuration a correct .config file is created anyway, as this validation is a separate pass.

https://github.com/thesofproject/sof/blob/b9623d0fee3b0af2e68ab20f2cba3a4035918c1c/scripts/kconfig/kconfiglib.py#L3420 is the origin of the warning message. A simple visual inspection of the code reveals that no such checking of the "if" condition of any defaults is done. Further, I have searched for the original Kconfiglib implementation, and it _also_ manifests this issue.

I have not verified how the Linux kernel (C based) implementation of Kconfig behaves yet, so I don't know if it's a bug or an intended feature of Kconfiglib.

I have reproduced it on i.MX8QM as well.

CC @iuliana-prodan

paulstelian97, 2022-11-14:
I'm seeing two different solutions. A simpler one makes for a wordier Kconfig, and a more complicated one to be done inside kconfiglib.py itself to skip checks for defaults that are false.

lgirdwood, 2022-11-14:
@paulstelian97 is there any Zephyr Kconfig symbol we can reuse and deprecate any duplication we have ?

paulstelian97, 2022-11-15:
> @paulstelian97 is there any Zephyr Kconfig symbol we can reuse and deprecate any duplication we have ?

We can probably define MP_NUM_CPUS manually in our xtos builds as well, which makes the conditional default be unnecessary, and have MAX_CORE_COUNT be an alias for that. Or define MAX_CORE_COUNT in terms of MP_NUM_CPUS on Zephyr and manually otherwise (but with care, otherwise we just move this issue somewhere else)

paulstelian97, 2022-11-16:
A proposed solution (partial) is pull request #6614. It needs more work but it's a start and it's already sufficient to remove the warning.

marc-hb, 2022-12-07:
This is not specific to IMX, I have seen this warning on all Intel platforms since forever.

marc-hb, 2022-12-07:
>  inside kconfiglib.py 

BTW I suspect ours is out of date compared to the Zephyr one.

paulstelian97, 2022-12-07:
> > inside kconfiglib.py
> 
> BTW I suspect ours is out of date compared to the Zephyr one.

While I haven't checked the Zephyr one, I have looked at an upstream one and it still seems to have the code structured the same way, which leads to still having the issue. Do you know of Zephyr having its own patches on top of kconfiglib.py?

marc-hb, 2022-12-07:
I forgot kconfiglib.py had its own git repo, sorry for the confusion. In the past kconfiglib.py was designed by and for Zephyr so the Zephyr copy would have always been fairly up to date and with very few "downstream" changes if any. Then the developer left and the project became orphaned (https://github.com/ulfalizer/Kconfiglib/issues/121). So not sure about now, should be easy to check with the usual `diff`, `git log` etc.

I see some recent changes in https://github.com/zephyrproject-rtos/zephyr/commits/main/scripts/kconfig but not many.

paulstelian97, 2022-12-07:
The actual code interesting to us looks basically identical so the Zephyr kconfiglib _should_ also manifest the issue (at least that's what I'd expect, but it can be checked I guess)

kv2019i, 2023-03-09:
No progress, tagging to v2.6.

kv2019i, 2023-11-24:
@dbaluta @paulstelian97 We've been bumping this forward since v2.5 -- shall we do this once again?

dbaluta, 2023-11-27:
Let's close this as we plan to deprecate Xtensa OS anyway.

closed 2023-11-27 by dbaluta