thesofproject/sof #6593: [BUG] Firmware boot failure at start of playback on MTL RVP

yongzhi1, 2022-11-14:
**Describe the bug**
aplay fails from cmd line.

**To Reproduce**
1. Power up RVP
2. Check aplay -l for spk
3. Run and observe dmesg log
```
aplay -Dhw:0,1 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
```

**Reproduction Rate**
100%

**Expected behavior**
playback should not fail.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://chromium.googlesource.com/a/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.15
    * SOF: https://github.com/thesofproject/sof/tree/mtl-002-drop-stable/
2) Name of the topology file
    * Topology: 
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP

**Screenshots or console output**

```
[  145.604451] sof-audio-pci-intel-mtl 0000:00:1f.3: Turning i915 HDAC power 1
[  145.610142] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  145.613075] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  145.616420] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  145.616467] sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[  145.618215] sof-audio-pci-intel-mtl 0000:00:1f.3: Turning i915 HDAC power 0
[  145.619078] sof-audio-pci-intel-mtl 0000:00:1f.3: Current DSP power state: D0I0
[  145.619091] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 0 -> 1
[  145.619100] sof-audio-pci-intel-mtl 0000:00:1f.3: loading firmware
[  145.619106] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 1 -> 2
[  145.620240] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x1010000 successful
[  145.621354] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1d1c]=0x8151 successful
[  145.621375] sof-audio-pci-intel-mtl 0000:00:1f.3: booting DSP firmware
[  145.621383] sof-audio-pci-intel-mtl 0000:00:1f.3: IMR restore supported, booting from IMR directly
[  145.622468] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000101 successful
[  145.622479] sof-audio-pci-intel-mtl 0000:00:1f.3: Primary core power up successful
[  145.622489] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x73214]=0x80000000 successful
[  145.622531] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x41 successful
[  145.622561] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x1 successful
[  145.622587] sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x114c]=0xf successful
[  147.678462] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  147.687993] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware boot failure due to timeout
[  147.696885] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[  147.705882] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[  147.715618] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000001c, ROM debug error: 0x2328
[  147.726134] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[  147.734121] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  147.743369] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
[  147.753597] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 2 -> 3
[  147.753614] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_pm_runtime_get on 0000:00:1f.3: -5
[  147.765897]  SSP1-Codec: __soc_pcm_open() failed (-5)
[  147.771549]  Speakers: ASoC: dpcm_be_dai_startup() failed at SSP1-Codec (-5)
[  147.779433]  Speakers: dpcm_fe_dai_startup() failed (-5)

```

fredoh9, 2022-11-15:
I have same problem with this. Some of tests are done with this branch already. Looks there are some gaps, will fill the gap and clarify the problem soon.

yongzhi1, 2022-11-17:
Observed on some SoC stepping consistently, thus re-open.

yongzhi1, 2022-11-17:
Seems the issue is IMR related, do not know why imr boot failed upon playback, the error is not seen if I turn off imr boot:
```
diff --git a/sound/soc/sof/intel/hda-loader.c b/sound/soc/sof/intel/hda-loader.c
index eb22eb3f6fee..7405e906d463 100644
--- a/sound/soc/sof/intel/hda-loader.c
+++ b/sound/soc/sof/intel/hda-loader.c
@@ -481,7 +481,7 @@ int hda_dsp_cl_boot_firmware(struct snd_sof_dev *sdev)
        ret = hda_cl_copy_fw(sdev, hext_stream);
        if (!ret) {
                dev_dbg(sdev->dev, "Firmware download successful, booting...\n");
-               hda->skip_imr_boot = false;
+               hda->skip_imr_boot = true;
        } else {
                snd_sof_dsp_dbg_dump(sdev, "Firmware download failed",
                                     SOF_DBG_DUMP_PCI | SOF_DBG_DUMP_MBOX);

```

RDharageswari, 2022-11-17:
I am also observing the same issue:
 123.046270] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  123.055775] sof-audio-pci-intel-mtl 0000:00:1f.3: Firmware boot failure due to timeout
[  123.064644] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[  123.073603] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xffffffff, ROM error: 0xffffffff
[  123.083344] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xd000001c, ROM debug error: 0x2328
[  123.093858] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit not enabled
[  123.101848] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  123.111108] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to boot DSP firmware after resume -5
[  123.121331] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_pm_runtime_get on 0000:00:1f.3: -5
[  123.133600]  SSP0-Codec: __soc_pcm_open() failed (-5)
[  123.139260]  Headset: ASoC: dpcm_be_dai_startup() failed at SSP0-Codec (-5)
[  123.147046]  Headset: dpcm_fe_dai_startup() failed (-5)
[  123.973742] cros-usbpd-charger cros-usbpd-charger.2.auto: Port 2: default case!

RDharageswari, 2022-11-17:
[dmesg.txt](https://github.com/thesofproject/sof/files/10036526/dmesg.txt)

kv2019i, 2022-11-18:
@marcinszkudlinski  Can you confirm whether IMR boot should be used with mtl-002 yet? 

@yongzhi1 Kernel should retry coldboot if IMR boot fails. I was checking chromeos-5.15 backports and at least "3b99852f4c874062295704dd483b03cab61301fe ASoC: SOF: Intel: hda: Skip IMR boot after a firmware crash or boot failure
" is missing. This could help (IMR boot would still fail, but it should retry coldboot).

Otherwise kernel can be configured to skip IMR boot with "options snd_sof sof_debug=128".

marcinszkudlinski, 2022-11-18:
It should not, D3 flow is not implemented in 002

yongzhi1, 2022-11-18:
Thanks @kv2019i , will try your suggestion.

mengdonglin, 2022-12-06:
Closing. Please verify when D3 flow is ready on next release.

closed 2022-12-06 by mengdonglin