thesofproject/sof #6595: [BUG] no sound coming from speakers with 16-bit playback on DeepBuffer PCM dev

yongzhi1, 2022-11-15:
**Describe the bug**
16-bit playback from DeepBufer dev generates no audible sound. 

**To Reproduce**
Run the following cmd and can hear no sound from speakers:
```
aplay -Dhw:0,11 -c2 -r48000 -fS16_LE /home/sample.wav -vv
```

**Reproduction Rate**
100%

**Expected behavior**
Expect to hear some sound quality as if we are playing from _sofrt5682 [sof-rt5682], device 1: Speakers_

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://chromium.googlesource.com/a/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.15  + https://github.com/thesofproject/linux/pull/3928
    * SOF: https://github.com/thesofproject/sof/tree/mtl-001-drop-stable
2) Name of the topology file
    * Topology: https://github.com/yongzhi1/sof/tree/deepbuffer
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP

**Screenshots or console output**

RanderWang, 2022-11-16:
@yongzhi1 it is a sdw platform ? do you have kernel log ?  And is it caused by 16bits ?

yongzhi1, 2022-11-16:
Hi @RanderWang , it's I2S rt5682+max, 32bit playback works fine although I only tested once.

RanderWang, 2022-11-16:
It works on SDW and HDA platform. I only tested with nocodec tplg since we don't have a platform with I2S codec. It is not caused by deep buffer setting, instead about SSP format selection in driver. I will debug it.

````c
static int sof_ipc4_pcm_dai_link_fixup(struct snd_soc_pcm_runtime *rtd,
				       struct snd_pcm_hw_params *params)
{
        ....

	switch (ipc4_copier->dai_type) {
	case SOF_DAI_INTEL_SSP:
		ipc4_ssp_dai_config_pcm_params_match(sdev, (char *)rtd->dai_link->name, params);
		break;
	default:
		break;
	}

	return 0;
}

yongzhi1, 2022-11-16:
Hi, @RanderWang, thanks for the analysis, made the following changes, but did not see improvement.

```
--- a/tools/topology/topology2/cavs/cavs-rt5682.conf
+++ b/tools/topology/topology2/cavs/cavs-rt5682.conf
@@ -163,12 +163,29 @@ Object.Pipeline {
                        copier_type "SSP"
                        stream_name "SSP0-Codec"
                        node_type $I2S_LINK_OUTPUT_CLASS
+
+                       num_audio_formats 1
+                       # 16-bit input format 48KHz 2ch. Output sample format is always 32-bit for playback
+                       Object.Base.audio_format.1 {
+                               out_bit_depth           32
+                               out_valid_bit_depth     32
+                               dma_buffer_size "$[$ibs * $DEEPBUFFER_FW_DMA_MS]"
+                       }
```
Also the volume level from Spk sound fine w/o above changes:

card 0: sofrt5682 [sof-rt5682], device 1: Speakers (*) [] <= vol OK
card 0: sofrt5682 [sof-rt5682], device 11: Deepbuffer Amps (*) [] <= silent playing observed

RanderWang, 2022-11-17:
@yongzhi1 can you share the kernel msg ?  we don't have a platform with I2S codec so we can't try. And how about my PR 4016 ? Thanks!

yongzhi1, 2022-11-17:
Second try, still did not make any difference.[dmesg-deep-override.txt](https://github.com/thesofproject/sof/files/10027487/dmesg-deep-override.txt)
```
        }
+
+       deepbuffer-playback.15 {
+               index   1
+
+               Object.Widget.copier."1" {
+                       node_type $I2S_LINK_OUTPUT_CLASS
+                       num_audio_formats 2
+
+                       Object.Base.audio_format.1 {
+                               out_bit_depth           32
+                               out_valid_bit_depth     32
+                               dma_buffer_size "$[$obs * $DEEPBUFFER_FW_DMA_MS]"
+                       }
+                       Object.Base.audio_format.2 {
+                               in_bit_depth            32
+                               in_valid_bit_depth      32
+                               out_bit_depth           32
+                               out_valid_bit_depth     32
+                               # 100ms buffer
+                               dma_buffer_size "$[$ibs * $DEEPBUFFER_FW_DMA_MS]"
+                       }
+               }
+       }
 }
 
 Object.PCM {
```
Log for PR4016, [dmesg-pr4016.txt](https://github.com/thesofproject/sof/files/10027488/dmesg-pr4016.txt), no sound still.

RanderWang, 2022-11-17:
@yongzhi1  thanks !   Format 24/32 is used in [dmesg-deep-override.txt](https://github.com/thesofproject/sof/files/10027487/dmesg-deep-override.txt) but 32/32 is used by [dmesg-pr4016.txt](https://github.com/thesofproject/sof/files/10027488/dmesg-pr4016.txt).  Both no sound ?  this PR https://github.com/thesofproject/sof/pull/6583 can work ?

RanderWang, 2022-11-17:
Can you try closed source fw ?

RanderWang, 2022-11-17:
@yongzhi1 please try this. I tried your change  https://github.com/thesofproject/sof/issues/6595#issuecomment-1317386461 and driver reported a error that no matched format
````c
--- a/tools/topology/topology2/cavs/cavs-nocodec.conf
+++ b/tools/topology/topology2/cavs/cavs-nocodec.conf
@@ -168,6 +168,16 @@ Object.Pipeline {
                        copier_type "SSP"
                        stream_name "NoCodec-0"
                        node_type $I2S_LINK_OUTPUT_CLASS
+
+                       num_audio_formats 1
+                       # 24-bit input format 48KHz 2ch. Output sample format is always 32-bit for playback
+                       Object.Base.audio_format.1 {
+                               in_bit_depth    32
+                               in_valid_bit_depth 24
+                               out_bit_depth           32
+                               out_valid_bit_depth     32
+                               dma_buffer_size "$[$ibs * $DEEPBUFFER_FW_DMA_MS]"
+                       }
                }
````

RanderWang, 2022-11-17:
we need to set output bits to 32 according to be compatible with SSP.  Why 24bit is used in deep buffer ? 24bits is faster than 32bits in FW implementation. The original non-deepbuffer use 32bit is not a optimal way and I will refine them later

````c
	SSP."0" {
		id 		0
		dai_index	0
		direction	"duplex"
		name		SSP0-Codec
		default_hw_conf_id	0
		sample_bits		32  //  set valid bit to 32
		io_clk		$MCLK

		Object.Base.hw_config."SSP0" {
			id	0
			mclk_freq	$MCLK
			bclk_freq	3072000
			tdm_slot_width	32
		}
	}
````

yongzhi1, 2022-11-17:
Hi, @RanderWang, tried closed src FW, it did not help. 
Here is the log with PR4016 + tplg changes, still no sound from 'Deepbuffer Amps' dev.
[dmesg-pr4016+tplg.txt](https://github.com/thesofproject/sof/files/10034072/dmesg-pr4016%2Btplg.txt)
[0001-tplg-changes.txt](https://github.com/thesofproject/sof/files/10034070/0001-tplg-changes.txt)

With https://github.com/thesofproject/sof/pull/6583, yes I can hear sound, [dmesg-pr6583.txt](https://github.com/thesofproject/sof/files/10034178/dmesg-pr6583.txt).

RanderWang, 2022-11-18:
@yongzhi1  do you use dmic to check sound ?  And can I use your device ?

RanderWang, 2022-11-18:
with tplg and 4016 change, why the ouput is still 0x1802 ( 2ch 24 bit), it should b 0x2002 (32bit 2ch)

[  233.152302] sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.SSP.4.1, type 27
[  233.152312] sof-audio-pci-intel-mtl 0000:00:1f.3: sample rate: 48000 sample width: 32 channels: 2
[  233.152322] sof-audio-pci-intel-mtl 0000:00:1f.3: dai index 1 nhlt type 3 direction 0
[  233.152335] sof-audio-pci-intel-mtl 0000:00:1f.3: matching audio format index for 48000Hz, 24bit, 2 channels: 0
[  233.152344] sof-audio-pci-intel-mtl 0000:00:1f.3: Init input audio formats for copier.SSP.4.1
[  233.152351] sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x1802)
[  233.152362] sof-audio-pci-intel-mtl 0000:00:1f.3: Init output audio formats for copier.SSP.4.1
[  233.152369] sof-audio-pci-intel-mtl 0000:00:1f.3:  **#0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x1802)**

yongzhi1, 2022-11-22:
Hi @RanderWang , compared with your nocodec.conf, changed DEEP_BUFFER_PCM_ID from 11 to 31, now it works as expected:[dmesg-deep-tplg-override-02.txt](https://github.com/thesofproject/sof/files/10062376/dmesg-deep-tplg-override-02.txt), thanks!! 

With same kernel and tplg, when I switched to mtl-002-drop-stable, now it failed for different reason:
```
[  118.656628] sof-audio-pci-intel-mtl 0000:00:1f.3: matching audio format index for 48000Hz, 24bit, 2 channels: 0
[  118.656638] sof-audio-pci-intel-mtl 0000:00:1f.3: Init input audio formats for copier.SSP.4.1
[  118.656644] sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x1802)
[  118.656655] sof-audio-pci-intel-mtl 0000:00:1f.3: Init output audio formats for copier.SSP.4.1
[  118.656663] sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002)
[  118.656674] sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.SSP.4.1, IPC size is 164
[  118.656688] sof-audio-pci-intel-mtl 0000:00:1f.3: pipeline: 15 memory pages: 5
[  118.656697] sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget pipeline.15 instance 0 - pipe 15 - core 0
...
[  118.667396] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x130f0004|0x0
[  118.667458] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 4 state 3
[  118.667475] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13040003|0x0
[  118.667886] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0
[  118.667960] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13040003|0x0
[  118.667996] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 4 state 4
[  118.668013] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13040004|0x0
[  118.668842] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0
[  118.668864] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[  118.679527] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13040004|0x0
[  118.688009] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set state 4 for pipeline 4
[  118.696864] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  118.707763]  Deepbuffer Amps: ASoC: trigger FE cmd: 1 failed: -22
```
[mtrace-deep-play-failure.txt](https://github.com/thesofproject/sof/files/10062387/mtrace-deep-play-failure.txt)

kfrydryx, 2022-11-22:
We are encountering issues with unsupported formats in internal testing. AFAIK someone on our team will be working on this now. 
@RanderWang Please help to mitigate the issue for now, proper support for currently unsupported formats should be coming soon.

yongzhi1, 2022-11-23:
Double checked the same with our backported series, with DEEPBUFFER_FW_DMA_MS=100, failed to run playback on either Speakers or Deepbuffer Amps, but both works fine with DEEPBUFFER_FW_DMA_MS=50. Since the original silent playing issue was fixed by format override, let's close this one, thanks all for the great support!!

closed 2022-11-23 by yongzhi1