thesofproject/sof #6598: [BUG] FW reported error: 7 when playback on MTLP_RVP_NOCODEC/MTLP_RVP_SDW

miRoox, 2022-11-15:
**Describe the bug**
CI observed an IPC error when playback on MTLP_RVP_NOCODEC and MTLP_RVP_SDW.

**To Reproduce**
```
aplay   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 10 /dev/zero -v -q
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/ec7f15280902
    * SOF: https://github.com/thesofproject/sof/commit/1e3a4cfe9b26
2) Name of the topology file
    * Topology: sof-mtl-nocodec.tplg, cavs-mtl-sdw-dmic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_NOCODEC/MTLP_RVP_SDW

**Screenshots or console output**

**dmesg**
```
[  251.578281] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[  251.581900] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000007|0x1: GLB_SET_PIPELINE_STATE
[  251.581908] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[  251.581988] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13000004|0x1
[  251.582014] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[  251.582035] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  251.582058] kernel:  Port0: ASoC: trigger FE cmd: 1 failed: -22
```

[full_dmesg.txt](https://github.com/thesofproject/sof/files/10009485/full_dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/10009482/mtrace.txt)

lgirdwood, 2022-11-16:
Relevant part of trace are 
```
[00:03:21.409,000] <inf> ipc: buffer_new: buffer new size 0x300 id 1.0 flags 0x0
[00:03:21.409,000] <inf> dai_intel_ssp: dai_ssp_pm_runtime_en_ssp_power: dai_ssp_pm_runtime_en_ssp_power en_ssp_power index 0
[00:03:21.409,000] <inf> dai_intel_ssp: dai_ssp_pm_runtime_en_ssp_power: dai_ssp_pm_runtime_en_ssp_power I2SLCTL
[00:03:21.409,000] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 28000, reg 8, mask 80, value 80
[00:03:21.410,000] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[00:03:21.410,000] <inf> ipc: dai_config: comp:1 0x40001 dai_config() dai type = 1 index = 0 dd 0x400dfd80
[00:03:21.410,000] <inf> dai_intel_ssp: dai_ssp_set_config_blob: dai_ssp_set_config_blob sscr0 = 0xc1d0077f, sscr1 = 0xd0700004, ssto = 0x00000000, sspsp = 0x2200000
[00:03:21.410,000] <inf> dai_intel_ssp: dai_ssp_set_config_blob: dai_ssp_set_config_blob sscr2 = 0x00004002, sspsp2 = 0x00000000, sscr3 = 0x07070000
[00:03:21.410,000] <err> dai_intel_ssp: dai_ssp_set_config_blob ssioc = 0x00000020, ssrsa = 0x00000003, sstsa = 0x00000003
[00:03:21.410,000] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 28000, reg 4, mask 300000, value 300000
[00:03:21.410,000] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 28000, reg 0, mask 80, value 80
[00:03:21.410,000] <inf> pipe: pipeline_connect: comp:1 0x40001 connect buffer 0 as source
[00:03:21.411,000] <inf> ipc: ipc_cmd: rx	: 0x45000004|0x6
[00:03:21.411,000] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:03:21.411,000] <inf> pipe: pipeline_connect: comp:0 0x40000 connect buffer 0 as sink
[00:03:21.411,000] <inf> pipe: pipeline_connect: comp:0 0x60000 connect buffer 0 as source
[00:03:21.411,000] <inf> ipc: ipc_cmd: rx	: 0x45000006|0x2
[00:03:21.412,000] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:03:21.412,000] <inf> pipe: pipeline_connect: comp:0 0x60000 connect buffer 0 as sink
[00:03:21.412,000] <inf> pipe: pipeline_connect: comp:0 0x20000 connect buffer 0 as source
[00:03:21.412,000] <inf> ipc: ipc_cmd: rx	: 0x45000002|0x3
[00:03:21.412,000] <inf> ipc: buffer_new: buffer new size 0x300 id 0.0 flags 0x0
[00:03:21.413,000] <inf> pipe: pipeline_connect: comp:0 0x20000 connect buffer 0 as sink
[00:03:21.413,000] <inf> pipe: pipeline_connect: comp:1 0x30000 connect buffer 0 as source
[00:03:21.413,000] <inf> ipc: ipc_cmd: rx	: 0x45000003|0x10006
[00:03:21.413,000] <inf> ipc: buffer_new: buffer new size 0x300 id 1.0 flags 0x0
[00:03:21.413,000] <inf> pipe: pipeline_connect: comp:1 0x30000 connect buffer 0 as sink
[00:03:21.413,000] <inf> pipe: pipeline_connect: comp:1 0x60001 connect buffer 0 as source
[00:03:21.414,000] <inf> ipc: ipc_cmd: rx	: 0x45010006|0x10004
[00:03:21.414,000] <inf> ipc: buffer_new: buffer new size 0x300 id 1.0 flags 0x0
[00:03:21.414,000] <inf> pipe: pipeline_connect: comp:1 0x60001 connect buffer 0 as sink
[00:03:21.414,000] <inf> pipe: pipeline_connect: comp:1 0x40001 connect buffer 0 as source
[00:03:21.415,000] <inf> ipc: ipc_cmd: rx	: 0x13000003|0x1
[00:03:21.416,000] <inf> pipe: pipeline_complete: pipe:1 0x0 pipeline complete, clock freq 400000000Hz
[00:03:21.416,000] <wrn> ipc: ipc_pipeline_complete(): no scheduling component specified, use comp 131072
[00:03:21.416,000] <inf> pipe: pipeline_complete: pipe:0 0x0 pipeline complete, clock freq 400000000Hz
[00:03:21.416,000] <inf> ipc: ipc_cmd: rx	: 0x13000004|0x1
[00:03:21.416,000] <inf> ipc: ipc_dai_data_config: comp:1 0x40001 dai_data_config() dai type = 1 index = 0 dd 0x400dfd80
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties dai_index 0
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties fifo 163856
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties handshake 2
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties init delay 0
[00:03:21.417,000] <inf> dai_comp: dai_playback_params: comp:1 0x40001 dai_playback_params() dest_dev = 2 stream_id = 0 src_width = 4 dest_width = 4
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties dai_index 0
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties fifo 163856
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties handshake 2
[00:03:21.417,000] <inf> dai_intel_ssp: dai_ssp_get_properties: dai_ssp_get_properties init delay 0
[00:03:21.417,000] <inf> dai_comp: dai_playback_params: comp:1 0x40001 dai_playback_params() fifo 0x28010
[00:03:21.418,000] <inf> pipe: pipeline_prepare: pipe:1 0x0 pipe prepare
[00:03:21.418,000] <inf> dai_comp: dai_prepare: comp:1 0x40001 dai_prepare()
[00:03:21.418,000] <inf> dai_comp: dai_config_prepare: comp:1 0x40001 dai_config_prepare(), channel = 0
[00:03:21.418,000] <inf> dai_comp: dai_config_prepare: comp:1 0x40001 dai_config_prepare(): new configured dma channel index 0
[00:03:21.418,000] <err> dai_comp: comp:1 0x40001 dai_prepare(): Missing dd->config.elem_array.elems.
[00:03:21.419,000] <err> pipe: pipe:1 0x0 pipeline_prepare(): ret = -22, dev->comp.id = 196608
[00:03:21.419,000] <err> ipc: ipc: pipe 1 comp 0 prepare failed -22
[00:03:21.419,000] <inf> pipe: pipeline_reset: pipe:1 0x0 pipe reset
[00:03:21.419,000] <inf> dai_comp: dai_reset: comp:1 0x40001 dai_reset()
[... cut ...]
[00:03:21.928,000] <inf> pipe: pipeline_disconnect: comp:1 0x60001 disconnect buffer 0 as source
[00:03:21.929,000] <inf> pipe: pipeline_disconnect: comp:1 0x40001 disconnect buffer 0 as source
[00:03:21.929,000] <inf> dma: dma_put: dma_put(), dma = 0x400c84a0, sref = 0
[00:03:21.929,000] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power index 0
[00:03:21.929,000] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power I2SLCTL
[00:03:21.930,000] <inf> pipe: pipeline_free: pipe:1 0x0 pipeline_free()
Terminated
```

miRoox, 2022-11-22:
The error is changed back to #6516 after 5db5f95cb01d8c18cbb2ae3c7e60aed88da63a8f

kfrydryx, 2022-11-22:
Should be fixed. Please reopen if reappears after https://github.com/thesofproject/sof/issues/6516 is fixed

closed 2022-11-22 by kfrydryx