thesofproject/sof #6617: [BUG] mtrace log timestamp accuracy is low

kv2019i, 2022-11-16:
**Describe the bug**
When Zephyr mtrace log mechanism is used with SOF, the log timestamps have limited timestamp accuracy:

[    0.014000] <inf> ipc: rx    : 0x13000003|0x1
[    0.014000] <inf> pipe: pipe:1 0x0 pipeline complete, clock freq 400000000Hz
[    0.014000] <wrn> ipc: ipc_pipeline_complete(): no scheduling component specified, use comp 262144
[    0.014000] <inf> pipe: pipe:0 0x0 pipeline complete, clock freq 400000000Hz
[    0.014000] <inf> ipc: rx    : 0x13000004|0x1
[    0.014000] <inf> ipc: comp:1 0x40001 dai_data_config() dai type = 1 index = 0 dd 0x9e0af5c0

To debug events in the audio loop (repeats every 1ms), this granularity is not accurate enough.

Current settings we use:
CONFIG_LOG_TRACE_SHORT_TIMESTAMP=y
CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP=y
CONFIG_LOG_OUTPUT_FORMAT_LINUX_TIMESTAMP=y
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=38400000
CONFIG_SYS_CLOCK_TICKS_PER_SEC=15000

One way to increase precision is to modify CONFIG_SYS_CLOCK_TICKS_PER_SEC, but that does not seem ideal as that changes system behaviour.
 
**To Reproduce**
Run any test with SOF Zephyr builds and observe mtrace logs.

**Reproduction Rate**
100%

**Expected behavior**
Microsecond level accuracy of timestamps in FW logs.

**Impact**
High

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: main of 2022-11-16
    * SOF: main of 2022-11-16
2) Name of the topology file
    * Topology: any
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL, MTL

kv2019i, 2022-11-16:
FYI @nashif @lgirdwood . Started with a SOF issue as this could be incorrect usage of Zephyr options.

dbaluta, 2022-11-16:
Can you check if you have CONFIG_TIMER_HAS_64BIT_CYCLE_COUNTER=y?

kv2019i, 2022-11-16:
Thanks @dbaluta No, it's not set, but I do remember checking this and that it didn't help (haven't check very recently though, so will check again).

I think the problem is possibly this setting:
```
»       /* Set default timestamp. */
»       if (sys_clock_hw_cycles_per_sec() > 1000000) {
»       »       log_set_timestamp_func(default_lf_get_timestamp, 1000U);
»       } else {
»       »       log_set_timestamp_func(default_get_timestamp,
»       »       »       »              sys_clock_hw_cycles_per_sec());
»       }
```

... HW cycles per sec is 19.2/38.4Mhz on many SOF configs, so we get to the first path and get the low accuracy.

But hmm, I guess SOF could set the timestamp function again in its init. Let me take a look at this.

kv2019i, 2022-11-16:
Ta-daa, that does work. Let me upload a PR soon. I stumbled on a minor obstacle as including the right header from Zephyr blows up due to SOF's panic() definition. But oh well, long overdue to fix this, so I'll do this in the same PR. Stay tuned.

closed 2022-11-18 by lgirdwood