thesofproject/sof #6635: Support out-of-tree platform,driver,audio .... definitions

hongshui3000, 2022-11-18:
I want to be able to develop my own programs in the same way as zephyrrtos (out of tree), but still keep upstream compatibility. For example, I have my own private audio algorithm code that cannot be open sourced or merged with upstream, but I also find it difficult to handle when I also use sof's audio library

dbaluta, 2022-11-18:
You can have a look at module interface at:

https://github.com/thesofproject/sof/blob/main/src/include/sof/audio/module_adapter/module/module_interface.h#L73

lgirdwood, 2022-11-18:
@hongshui3000 recommend to use the module interface as this means you also will also work on Windows with Intel platforms in the future. The module adapter today supports C and a the C++ binding is being upstreamed now.

hongshui3000, 2022-11-19:
I mainly want to get out of the upstream and maintain my own code independently, similar to the zephyr app development approach(https://github.com/zephyrproject-rtos/example-application).
I use the module interface, how should I add my own application or driver or platform, outside of upstream.
Is the method of adding modules out of tree(zephyrrtos)also suitable for sof?

paulstelian97, 2022-11-21:
Yeah, currently they would have to be added in your own SOF fork. Due to the format of the firmware files there's no good way to do it completely out of tree as of now (the build system needs to be more serious if we want stuff that is _actually_ out of tree to be included in the final firmware image).

hongshui3000, 2022-11-21:
> Yeah, currently they would have to be added in your own SOF fork. Due to the format of the firmware files there's no good way to do it completely out of tree as of now (the build system needs to be [...]

Is it possible to use zephyrrtos' build system to achieve this?