thesofproject/sof #6657: [BUG] Recorded file from echo-reference dev has distortion and noise

yongzhi1, 2022-11-23:
**Describe the bug**
Run echo ref unit test with NOCODEC config, the captured file shows distortion and noise when examined on Audacity. 

**To Reproduce**
Start playback first:
```
aplay -Dhw:0,31 -c2 -r48000 -fS16_LE /home/sample.wav
```
Run capture and check the captured file on desktop:
```
arecord -Dhw:0,13 /home/loop.wav -c2 -r48000 -fs16_le -d10 
```

**Reproduction Rate**
100%

**Expected behavior**
The issue is not reproducible with closed src ACE FW.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/bardliao/linux/commits/echo-reference-nocodec
    * SOF: https://github.com/thesofproject/sof/tree/mtl-002-drop-stable
2) Name of the topology file
    * Topology: https://github.com/bardliao/sof/commits/test-echo-reference-nocodec-2
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP

**Screenshots or console output**
[mtrace-echo-ref-mtl-002.txt](https://github.com/thesofproject/sof/files/10079534/mtrace-echo-ref-mtl-002.txt)
https://user-images.githubusercontent.com/19287706/203658196-f7212407-3407-497d-b893-ab293443ce5d.mp4

lgirdwood, 2022-11-25:
> Run echo ref unit test with NOCODEC config, the captured file shows distortion and noise when examined on Audacity.

@yongzhi1 can you take a screenshot of the audacity waveform and also attach a small raw PCM clip. Compressed audio cant be used to root cause distortion.

mengdonglin, 2022-11-28:
@yongzhi1 Which Platform and codec are you using in the test?

yongzhi1, 2022-11-28:
Hi @lgirdwood, here is the screenshot for the waveform, the lower track was captured with ACE, in both cases LR_48k.wav was used

[image]
And the captured file [loop-mtl-002.zip](https://github.com/thesofproject/sof/files/10106116/loop-mtl-002.zip)

lgirdwood, 2022-11-30:
@kfrydryx @mengdonglin sounds like some clipping/distortion on the signal. Does not look like a DMA issue, but maybe format conversion or signal clipping in the copier ?

mengdonglin, 2022-12-01:
@yongzhi1 @kfrydryx @lgirdwood Deprioritize this issue after sync with @bardliao . Current driver support for echo-reference is a temporary solution, and is being re-implemented. Let's test the new implementation when it's ready. Thanks!

keqiaozhang, 2022-12-05:
Will test EC in new implementation when it's ready, closing this one.

closed 2022-12-05 by keqiaozhang