thesofproject/sof #6667: [BUG] Build fail on mt8186/v0.2 if using newest alsa-lib

Zames-Chang, 2022-11-25:
I encounter this problem with 

**Describe the bug**
If you use newest alsa-lib to build topology will causing build failure.
It can be workaround by checkout alsa-lib to commit 152983f01b0bc1178ea0d461ebf66e2d2a8e2e02

**To Reproduce**
git checkout mt8186/v0.2
./scripts/build-tools.sh -T

**Reproduction Rate**
100%

**Expected behavior**
Can build

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: d3d49dce58d3d316cd7044ee84996e48b1457df4
2) Name of the topology file
    * Topology: I build all tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: mt8186

**Screenshots or console output**
```
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- No CMAKE_BUILD_TYPE, defaulting to Debug
-- Configuring done
-- Generating done
-- Build files have been written to: /sof/sof/tools/build_tools
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'glib-2.0'
-- Configuring done
-- Generating done
-- Build files have been written to: /sof/sof/tools/build_tools/fuzzer
+ cmake --build /sof/sof/tools/build_tools -- -j 128 topologies
[  0%] Generating sof-mtl-nocodec.tplg
[  0%] Generating cavs-mixin-mixout-hda-2ch-cnl.tplg
[  0%] Generating cavs-mixin-mixout-hda-2ch-tgl.tplg
[  0%] Generating cavs-mixin-mixout-hda-4ch-tgl.tplg
[  1%] Generating cavs-mixin-mixout-hda-4ch-cnl.tplg
[  1%] Generating cavs-mixin-mixout-hda.tplg
[  1%] Generating cavs-passthrough-hdmi.tplg
[  1%] Generating cavs-tgl-nocodec.tplg
[  1%] Generating cavs-sdw.tplg
find_modes(): fs not set
find_modes(): fs not set
find_modes(): fs not set
saving nhlt as binary in nhlt-ace-mtl-nocodec.bin
find_modes(): fs not set
saving nhlt as binary in nhlt.bin
saving nhlt as binary in nhlt.bin
saving nhlt as binary in nhlt.bin
[  1%] Built target abi_v1
[... 241 lines cut ...]
ALSA lib pcm.c:1523:(tplg_parse_hw_config) deprecated fsync value 'codec_slave', use 'codec_consumer'
[ 37%] Generating production/sof-byt-es8316-ssp0.tplg
ALSA lib pcm.c:1471:(tplg_parse_hw_config) deprecated bclk value 'codec_slave', use 'codec_consumer'
gmake[3]: *** [topology/topology1/kernel_dependent/v5.19/CMakeFiles/topology_sof-adl-rt711-l2-rt1316-l01.dir/build.make:73: topology/topology1/kernel_dependent/v5.19/sof-adl-rt711-l2-rt1316-l01.tplg] Error 139
ALSA lib pcm.c:1523:(tplg_parse_hw_config) deprecated fsync value 'codec_slave', use 'codec_consumer'
gmake[2]: *** [CMakeFiles/Makefile2:8267: topology/topology1/kernel_dependent/v5.19/CMakeFiles/topology_sof-tgl-rt711-rt1308-rt715.dir/all] Error 2
[ 37%] Generating production/sof-byt-da7213-ssp0.tplg
[ 37%] Generating production/sof-ehl-nocodec.conf
ALSA lib pcm.c:1471:(tplg_parse_hw_config) deprecated bclk value 'codec_slave', use 'codec_consumer'
ALSA lib pcm.c:1523:(tplg_parse_hw_config) deprecated fsync value 'codec_slave', use 'codec_consumer'
[ 38%] Generating production/sof-icl-dmic-4ch.conf
[ 38%] Generating production/sof-hda-generic-4ch.tplg
ALSA lib pcm.c:1471:(tplg_parse_hw_config) deprecated bclk value 'codec_slave', use 'codec_consumer'
ALSA lib pcm.c:1523:(tplg_parse_hw_config) deprecated fsync value 'codec_slave', use 'codec_consumer'
[ 38%] Generating production/sof-byt-rt5682-ssp0.tplg
gmake[2]: *** [CMakeFiles/Makefile2:8423: topology/topology1/kernel_dependent/v5.19/CMakeFiles/topology_sof-adl-rt711-l2-rt1316-l01.dir/all] Error 2
[ 39%] Generating production/sof-adl-nocodec.conf
[ 39%] Generating production/sof-byt-cx2072x-ssp0.tplg
[ 39%] Generating production/sof-byt-wm5102-ssp0.tplg
ALSA lib pcm.c:1471:(tplg_parse_hw_config) deprecated bclk value 'codec_slave', use 'codec_consumer'
ALSA lib pcm.c:1523:(tplg_parse_hw_config) deprecated fsync value 'codec_slave', use 'codec_consumer'
Segmentation fault (core dumped)
Segmentation fault (core dumped)
Segmentation fault (core dumped)
```

lgirdwood, 2022-11-25:
@juimonen @ranj063 any ideas - looks like a topology compiler crash.
@Zames-Chang which topology file - the build output is in parallel so difficult to see which topology file is crashing the compiler ?

juimonen, 2022-11-29:
@perexg FYI, something in latest alsa-lib head is segfaulting all our topology1 builds....

I see the segfault as:

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7ee20d7 in snd_interval_empty (i=<optimized out>) at /home/juimonen/work/alsa-lib/src/pcm/interval_inline.h:47
47		return i->empty;
(gdb) bt
#0  0x00007ffff7ee20d7 in snd_interval_empty (i=<optimized out>) at /home/juimonen/work/alsa-lib/src/pcm/interval_inline.h:47
#1  snd1_pcm_hw_param_get (params=0x0, var=var@entry=10, val=val@entry=0x7fffffffda2c, dir=dir@entry=0x0) at pcm_params.c:144
#2  0x00007ffff7edc57f in __snd_pcm_hw_params_get_channels (params=<optimized out>, val=val@entry=0x7fffffffda2c) at pcm.c:4492
#3  0x00007ffff7edfd8e in __old_snd_pcm_hw_params_get_channels (params=<optimized out>) at pcm.c:7885
#4  0x00007ffff7f88040 in parse_access_values (hdr=0x4b45e0, cfg=0x45d300) at ctl.c:78
#5  parse_access (cfg=cfg@entry=0x45cc50, hdr=hdr@entry=0x4b45e0) at ctl.c:99
#6  0x00007ffff7f88bb6 in tplg_parse_control_bytes (tplg=tplg@entry=0x4282f0, cfg=cfg@entry=0x45cc50, private=private@entry=0x0) at ctl.c:543
#7  0x00007ffff7f86902 in tplg_parse_compound (tplg=tplg@entry=0x4282f0, cfg=cfg@entry=0x45cbe0, fcn=0x7ffff7f88800 <tplg_parse_control_bytes>, private=private@entry=0x0) at parser.c:138
#8  0x00007ffff7f86bc4 in tplg_parse_config (cfg=0x43bb90, tplg=0x4282f0) at parser.c:186

and this lead me to commit 78b20e3caa7bba930095e05f, which does seems little fishy, at least what reads in the commit message.

But even after using the get_access method I still get segfault. So something perhaps in the ALSA_PCM_OLD_HW_PARAMS_API things, but not sure...

juimonen, 2022-11-29:
@perexg thanks! builds now.

@Zames-Chang update alsa-lib to latest head.

lgirdwood, 2022-11-30:
@Zames-Chang can we close ?

Zames-Chang, 2022-12-01:
Hi 

Sorry for late reply. I can build topology now with latest head. Thanks for the help.

closed 2022-12-01 by Zames-Chang