thesofproject/sof #6673: [BUG] [mt8186] DSP panic on sof-logger filter update

msyrota, 2022-11-27:
**Describe the bug**
Attempting to update trace filters for any component on mt8186 causes DSP FW panic.

**To Reproduce**
```
localhost ~ # sof-logger -l /lib/firmware/mediatek/sof/sof-mt8186.ldc -F debug=dai -t
```

**Reproduction Rate**
100%

**Expected behavior**
Successful trace log level update.

**Impact**
Annoyance (have to edit log levels manually and rebuild FW)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Kernel: Linux localhost 5.15.64-09059-gc972c537e297-dirty aarch64 GNU/Linux (ChromeOS 15014.0.0)
    * SOF: mt8186\v0.2 [d3d49dc](https://github.com/thesofproject/sof/commit/d3d49dce58d3d316cd7044ee84996e48b1457df4)
2) Name of the topology file
    * Topology: sof-mt8186-mt6366
3) Name of the platform(s) on which the bug is observed.
    * Platform: mt8186

**Screenshots or console output**
```
[  156.239109] sof-audio-of-mt8186 10680000.adsp: ipc tx: 0x90040000: GLB_TRACE_MSG: FILTER_UPDATE
[  156.239161] sof-audio-of-mt8186 10680000.adsp: snd_sof_dsp_panic: dsp_oops_offset 5253120 differs from panic offset 233492486
[  156.239179] sof-audio-of-mt8186 10680000.adsp: DSP panic!
[  156.239185] sof-audio-of-mt8186 10680000.adsp: fw_state change: 6 -> 7
[  156.744667] sof-audio-of-mt8186 10680000.adsp: ipc tx timed out for 0x90040000 (msg/reply size: 60/12)
[  156.744722] sof-audio-of-mt8186 10680000.adsp: preventing DSP entering D3 state to preserve context
[  156.744745] sof-audio-of-mt8186 10680000.adsp: Firmware exception
[  156.744773] sof-audio-of-mt8186 10680000.adsp: error: fail in sof_ipc_trace_update_filter -110

localhost ~ # sof-logger -d /lib/firmware/mediatek/sof/sof-mt8186.ldc
logger ABI Version is   5:3:0
ldc_file ABI Version is 5:3:0
ldc_file src checksum           0x9e4b9298
Loaded FW expects checksum      0x9e4b9298

Components uuid dictionary size:        1632 bytes
Components uuid base address:           0x4e180000
Components uuid entries:
           ADDRESS                                    UUID NAME
        0x4e180000  <53863428-9a72-44df-af0f-fe45ea2348ba> clkdrv
        0x4e180030  <be60f97d-78df-4796-a0ee-435cb56b720a> ipc
                                    ...
        0x4e180150  <c2b00d27-ffbc-4150-a51a-245c79c5e54b> dai
                                    ...
```

lgirdwood, 2022-11-30:
@thesofproject/mediatek fyi - looks like the debug filter updates are causing a panic() on mt8186. @cujomalainey fyi.

Zames-Chang, 2022-12-01:
@tinghan-shen

kuanhsuncheng, 2022-12-01:
@chunxu-li

kuanhsuncheng, 2022-12-02:
@msyrota Hi,

Could you please tell us more about "Platform: mt8186"? Thanks.

msyrota, 2022-12-04:
Hi, @kuanhsuncheng, platform is Steelix chromebook (Lenovo 300e), ChromeOS is official 15014.0.0

chunxu-li, 2022-12-05:
Hi @msyrota 
I can not reproduce DSP panic, and I add some debug log to test log filter, it woks well.
before set `dai` loglevel
```
[    43123849.953079] (          21.145832) c0 dai          1.2     src/audio/dai-legacy.c:669  INFO dai_prepare()
[    43123857.557245] (           7.604167) c0 dai          1.2     src/audio/dai-legacy.c:633  INFO dai_config_prepare(), channel = 0
[    43123864.640578] (           7.083333) c0 dai          1.2     src/audio/dai-legacy.c:634  WARN dai_config_prepare(), channel = 0
[    43123874.015578] (           9.375000) c0 dai          1.2     src/audio/dai-legacy.c:635  ERROR dai_config_prepare(), channel = 0
[    43123881.203078] (           7.187500) c0 dai          1.2     src/audio/dai-legacy.c:654  INFO dai_config_prepare(): new configured dma channel index 0
[    43123895.682244] (          14.479166) c0 memif              ...../mt8186/afe-memif.c:469  INFO memif_set_config, direction:8, afe_dir:1
[    43123903.286410] (           7.604167) c0 afedrv             ......./mt8186/afe-drv.c:271  INFO afe_dai_get_config, id:0
[    43123910.369743] (           7.083333) c0 afedrv             ......./mt8186/afe-drv.c:272  WARN afe_dai_get_config, id:0
[    43123919.744743] (           9.375000) c0 afedrv             ......./mt8186/afe-drv.c:273  ERROR afe_dai_get_config, id:0
[    43123926.203076] (           6.458333) c0 afedrv             ......./mt8186/afe-drv.c:286  INFO dai:0 get: format:0, rate:48000, channel:2
```
after set loglevel `sof-logger -l sof-mt8186.ldc -F error="dai" -t`
```
[   143880265.636879] (           5.833333) c0 ipc                       src/trace/trace.c:437  INFO Adaptive filtering disabled by user
[   143880270.532712] (           4.895833) c0 ipc                       src/trace/trace.c:440  INFO filter loglevel 1
[   143880280.428545] (           9.895833) c0 ipc                       src/trace/trace.c:453  INFO ret 5
[   143880285.532711] (           5.104167) c0 ipc                  src/ipc/ipc3/handler.c:927  INFO trace_filter_update for UUID key 0x4E180150, comp -1.-1 affected 5 components
```
only `error` level `dai` log shows
```
[   393588504.881031] (          20.520832) c0 pipe         1.3   ....../pipeline-params.c:346  INFO pipe prepare
[   393588518.735197] (          13.854166) c0 dai          1.2     src/audio/dai-legacy.c:635  ERROR dai_config_prepare(), channel = 0
[   393588531.131030] (          12.395833) c0 memif              ...../mt8186/afe-memif.c:469  INFO memif_set_config, direction:8, afe_dir:1
[   393588536.547697] (           5.416667) c0 afedrv             ......./mt8186/afe-drv.c:271  INFO afe_dai_get_config, id:0
[   393588541.547697] (           5.000000) c0 afedrv             ......./mt8186/afe-drv.c:272  WARN afe_dai_get_config, id:0
[   393588549.256030] (           7.708333) c0 afedrv             ......./mt8186/afe-drv.c:273  ERROR afe_dai_get_config, id:0
[   393588553.526863] (           4.270833) c0 afedrv             ......./mt8186/afe-drv.c:286  INFO dai:0 get: format:0, rate:48000, channel:2
```

SOF: mt8186\v0.2 [d3d49dc](https://github.com/thesofproject/sof/commit/d3d49dce58d3d316cd7044ee84996e48b1457df4)
test on
krabby:`CHROMEOS_RELEASE_DESCRIPTION=15183.52.0 (Official Build) dev-channel corsola test` 
and 
steelix: `CHROMEOS_RELEASE_DESCRIPTION=15130.0.0 (Official Build) dev-channel corsola test`

closed 2025-07-24 by github-actions[bot]