thesofproject/sof #6679: [BUG] MOD_DX IPC timeout

ranj063, 2022-11-28:
**Describe the bug**
The MOD_DX IPC times out when entering D3

**To Reproduce**
check-suspend-resume-with-audio.sh -l 1 -m capture

**Reproduction Rate**
Very low < 1-5%. Only seen it a handful of times so far.

**Expected behavior**
MOD_DX IPC should always send a reply back to the host.

3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

https://sof-ci.01.org/sofpr/PR6672/build2523/devicetest/index.html?model=TGLU_RVP_NOCODEC_IPC4ZPH&testcase=check-suspend-resume-with-capture

miRoox, 2022-12-06:
This issue is also found in MTL platforms, and the reproduction rate seems higher than TGL.

---

Found in test report 18581:
Kernel Commit: 
[321028577de9](https://github.com/thesofproject/linux/commit/321028577de9)
SOF Commit: 
[4a9a99aeb24a](https://github.com/thesofproject/sof/commit/4a9a99aeb24a)

[dmesg.txt](https://github.com/thesofproject/sof/files/10162059/dmesg.txt)

ranj063, 2022-12-06:
even on TGL check-suspend-resume-with-audio.sh -l 20 -m capture fails 100% of the time

keqiaozhang, 2023-01-09:
This issue is not reproducible in CI. Closing this bug.

closed 2023-01-09 by keqiaozhang