thesofproject/sof #6681: [BUG] ipc timed out for 0x13020004|0x0 when multiple pipeline playback on MTLP_RVP_SDW

miRoox, 2022-11-29:
**Describe the bug**
CI observed an ipc timed out for 0x13020004|0x0 when multiple pipeline playback on MTLP_RVP_SDW

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-ace-tplg/cavs-mtl-sdw-dmic.tplg MODEL=MTLP_RVP_SDW ~/sof-test/test-case/multiple-pipeline.sh -f p
```

**Reproduction Rate**
Only CI observe it, need further investigation.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev | https://github.com/thesofproject/linux/commit/62c1ccce6c65
    * SOF: main | https://github.com/thesofproject/sof/commit/ff7861f2229d
2) Name of the topology file
    * Topology: cavs-mtl-sdw-dmic.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_SDW

**Screenshots or console output**

**dmesg**
```
[ 2070.254060] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 2 state 4
[ 2070.254066] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13020004|0x0: GLB_SET_PIPELINE_STATE
[ 2070.484226] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 1 dir 1
[ 2070.484237] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: period min 192 max 16384 bytes
[ 2070.484242] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: period count 2 max 16
[ 2070.484246] kernel: snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: buffer max 65536 bytes
[ 2070.486324] kernel: snd_soc_rt711_sdca:rt711_sdca_pcm_hw_params: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_pcm_hw_params rt711-sdca-aif1
[ 2070.489223] kernel: snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 1 dir 1
[ 2070.489237] kernel: snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[ 2070.490166] kernel: snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x80]=0x40000 successful
[ 2070.490173] kernel: snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-mtl 0000:00:1f.3: period_bytes:0x4000
[ 2070.490177] kernel: snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-mtl 0000:00:1f.3: periods:4
[ 2070.490200] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.ALH.3.1, type 28
[ 2070.490204] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-mtl 0000:00:1f.3: matching audio format index for 48000Hz, 32bit, 2 channels: 1
[ 2070.490209] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-mtl 0000:00:1f.3: Init input audio formats for copier.ALH.3.1
[ 2070.490213] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002)
[ 2070.490219] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-mtl 0000:00:1f.3: Init output audio formats for copier.ALH.3.1
[ 2070.490222] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002)
[ 2070.490234] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.ALH.3.1, IPC size is 216
[ 2070.490239] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.host.4.1, type 24
[ 2070.490242] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-mtl 0000:00:1f.3: matching audio format index for 48000Hz, 16bit, 2 channels: 0
[ 2070.490246] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-mtl 0000:00:1f.3: Init input audio formats for copier.host.4.1
[ 2070.494846] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x1002)
[ 2070.494852] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-mtl 0000:00:1f.3: Init output audio formats for copier.host.4.1
[ 2070.494855] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-mtl 0000:00:1f.3:  #0: 48000KHz, 16bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x1002)
[ 2070.494859] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.host.4.1, IPC size is 84
[ 2070.494866] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: pipeline: 3 memory pages: 2
[ 2070.494871] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: Create widget pipeline.3 instance 3 - pipe 3 - core 0
[ 2070.651034] kernel: soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x40000
[ 2070.651608] kernel: snd_soc_rt711_sdca:rt711_sdca_interrupt_callback: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_interrupt_callback control_port_stat=4, sdca_cascade=1
[ 2070.652709] kernel: soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x20000
[ 2070.683857] kernel: snd_soc_rt711_sdca:rt711_sdca_headset_detect: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_headset_detect, detected_mode=0x3
[ 2070.683866] kernel: snd_soc_rt711_sdca:rt711_sdca_jack_detect_handler: rt711-sdca sdw:0:025d:0711:01: in rt711_sdca_jack_detect_handler, jack_type=0x1
[ 2070.683873] kernel: snd_soc_rt711_sdca:rt711_sdca_jack_detect_handler: rt711-sdca sdw:0:025d:0711:01: in rt711_sdca_jack_detect_handler, btn_type=0x0
[ 2070.683879] kernel: snd_soc_rt711_sdca:rt711_sdca_jack_detect_handler: rt711-sdca sdw:0:025d:0711:01: in rt711_sdca_jack_detect_handler, scp_sdca_stat1=0x1, scp_sdca_stat2=0x0
[ 2070.757263] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13020004|0x0
[ 2070.757505] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 2070.757509] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2070.757523] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x93020004|0x0|0x0, target: 0x33000000|0x0|0x0, ctl: 0x3
[ 2070.757531] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2070.757535] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2070.757539] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 2070.757543] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 2070.757553] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[ 2070.757558] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 2070.757563] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 2070.757567] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 2070.757580] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 2070.757586] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 2070.757596] kernel:  DeepBuffer: ASoC: trigger FE cmd: 1 failed: -110
```
[full dmesg.txt](https://github.com/thesofproject/sof/files/10109272/full_dmesg.txt)

juimonen, 2022-11-29:
so is this with deepbuffer and do we have mtrace?

miRoox, 2022-11-29:
> so is this with deepbuffer and do we have mtrace?

I think it's not with deep buffer. I've updated the console log.

mtrace seems almost empty due to #6669:
```
[00:01:05.163,118] <inf> ipc: ipc_cmd: rx	: 0x11000007|0x0
[00:01:05.163,243] <inf> pipe: pipeline_new: pipeline new pipe_id 0 priority 0
[00:01:05.163,765] <inf> ipc: ipc_cmd: rx	: 0x40000004|0x15
[00:01:05.164,050] <inf> ipc: buffer_new: buffer new size 0x180 id 0.0 flags 0x0
[00:01:05.164,393] <inf> dma: dma_get: dma_get() ID 0 sref = 1 busy channels 0
[00:01:05.164,556] <inf> pipe: pipeline_connect: comp:0 0x40000 connect buffer 0 as sink
[00:01:05.168,153] <inf> ipc: ipc_cmd: rx	: 0x40000006|0x10
[00:01:05.168,396] <inf> module_adapter: module_init: comp:0 0x60000 module_init() start
[00:01:05.168,566] <inf> module_adapter: module_init: comp:0 0x60000 module_init() done
[00:01:05.168,921] <inf> ipc: ipc_cmd: rx	: 0x44000006|0x30000018
[00:01:05.169,481] <inf> ipc: ipc_cmd: rx	: 0x40000002|0xa
[00:01:05.170,496] <inf> ipc: ipc_cmd: rx	: 0x11010006|0x0
Terminated
```

miRoox, 2022-12-01:
As #6543 got fixed, we found this issue also happened on MTLP_RVP_HDA

miRoox, 2022-12-06:
Not reproduced