thesofproject/sof #6700: [BUG] ipc timed out for 0x47000000| MOD_SET_DX when testing suspend/resume with audio

keqiaozhang, 2022-12-01:
**Describe the bug**
This error happened on TGLU_RVP_NOCODEC_IPC4ZPH when testing suspend/resume with capture. The reproduction rate is low. 
There's no obvious error in mtrace.

**dmesg**
```
[ 6605.003135] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[ 6605.003425] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[ 6605.003482] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[ 6605.003489] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[ 6605.509719] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[ 6605.509724] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 6605.509726] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 6605.509769] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[ 6605.509770] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 6605.509817] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0xc7000000|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 6605.509820] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 6605.509821] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 6605.509821] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 6605.509823] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 6605.509845] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xffffffff: unknown ROM status value
[ 6605.509884] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff
[ 6605.509885] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 6605.509887] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ctx_save IPC error: -110, proceeding with suspend
```

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/sof-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC_IPC4ZPH ~/sof-test/test-case/check-suspend-resume-with-audio.sh -l 5 -m capture

**Reproduction Rate**
Low, no reproductions so far by manually check.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[3869d8a05995](https://github.com/thesofproject/linux/commit/3869d8a05995)

SOF Branch: main
SOF Commit: 
[0bdf8e3d318f](https://github.com/thesofproject/sof/commit/0bdf8e3d318f)

Zephyr Commit: 
[ffed8388e7cb](https://github.com/zephyrproject-rtos/zephyr/commit/ffed8388e7cb)
Platform: TGLU_RVP_NOCODEC_IPC4ZPH
TPLG:sof-tgl-nocodec.tplg

[dmesg.txt](https://github.com/thesofproject/sof/files/10130443/dmesg.txt)

[mtrace.txt](https://github.com/thesofproject/sof/files/10130449/mtrace.txt)

ranj063, 2022-12-03:
This is a duplicate of #6679

miRoox, 2022-12-06:
Let's track it on #6679.

closed 2022-12-06 by miRoox