thesofproject/sof #6704: [BUG][IPC3][Zephyr] heap corruption during probing

lyakh, 2022-12-01:
**Describe the bug**
Firmware probing fails with an "ipc tx timed out" message in the kernel. The error log contains something like
```
ASSERTION FAIL [chunk_used(h, c)] @ WEST_TOPDIR/zephyr/lib/os/heap.c:175
        unexpected heap state (double-free?) for memory at 0x9e0817c0
[00:00:05.190,000] <err> os:  ** FATAL EXCEPTION
[00:00:05.190,000] <err> os:  ** CPU 2 EXCCAUSE 63 (zephyr exception)
[00:00:05.190,000] <err> os:  **  PC 0xbe0174b6 VADDR (nil)
[00:00:05.190,000] <err> os:  **  PS 0x60220
[00:00:05.191,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:2 CALLINC:2)
[00:00:05.191,000] <err> os:  **  A0 0xbe015ee0  SP 0xbe071b50  A2 0x4  A3 0xbe071b60
[00:00:05.191,000] <err> os:  **  A4 0xbe071b40  A5 0x4  A6 0xbe071b40  A7 0x2
[00:00:05.191,000] <err> os:  **  A8 0xbe014569  A9 0xbe071ab0 A10 0xbe04f29c A11 0xbe071b60
[00:00:05.192,000] <err> os:  ** A12 0xbe071b40 A13 0x4 A14 0x20000000 A15 (nil)
[00:00:05.192,000] <err> os:  ** LBEG 0xbe04cb9c LEND 0xbe04cba6 LCOUNT (nil)
[00:00:05.192,000] <err> os:  ** SAR 0x5
Backtrace:0xbe0174b3:0xbe071b50 0xbe015edd:0xbe071b60 0xbe014eaf:0xbe071b80 0xbe0287d0:0xbe071ba0 0xbe028a25:0xbe071be0 0xbe025239:0xbe071c00 0xbe039bcb:0xbe071c50 0xbe02645a:0xbe071c80 0xbe0262f2:0xbe071cc0 0xbe026422:0xbe071d00 0xbe0262f2:0xbe071d40 0xbe026422:0xbe071d80 0xbe0262f2:0xbe071dc0 0xbe026422:0xbe071e00 0xbe0262f2:0xbe071e40 0xbe026422:0xbe071e80 0xbe025fc4:0xbe071ec0 0xbe02c9fc:0xbe071f10 0xbe0462d5:0xbe0720b0 0xbe02838b:0xbe072100 0xbe01601d:0xbe072140 0xbe014ab7:0xbe072170 
[00:00:05.194,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 2
[00:00:05.194,000] <err> os: Current thread: 0x9e079158 (unknown)
[00:00:05.194,000] <err> os: Halting system
```

**To Reproduce**
Boot the firmware several times (rebooting the system seems to work a bit better than just reloading modules) and see probing fail with an 

**Reproduction Rate**
Somewhere between 20 and 50%

**Environment**
- Platform: TGL
- IPC version: IPC3
- Topology: nocodec
- OS: enable PulseAudio
- SOF: `CONFIG_ASSERT=y`

Observed with all recent versions, e.g.
SOF: 68808e854bec3711e4dca23247ea0ca2f5d2aa6d
Zephyr: e6ef1f3434a2f7c26bd9d1bbdba8537ff541ef01
Linux kernel: 6992e1c067144d28e6461237aae84798898e6787

an attempt to find a working SOF version went as far back as e135633636f88f53c175d33cc8703edcaa784ae7 which still failed the same way.

**Analysis**

The failed assertion means that a heap header got corrupted in Zephyr. Often it's detected when freeing input or output buffers in `module_adapter_reset()`. The header can likely be overwritten either by a cache write-back or by run-away DMA. Added debugging to check heap allocation headers upon allocation revealed invalid headers much earlier typically in 2 stream buffers. It has been noted, that buffers are always allocated on core 0 even when the pipeline runs on a different core. Fixing this to allocate buffers on the same core didn't help.

lgirdwood, 2022-12-02:
@lyakh do you know if DMA was running for host, trace or DAI during any part of these tests ? i.e. could DMA be clobbering

lyakh, 2022-12-02:
@lgirdwood yes, DMA was running, but I very much suspect a cache problem now. Particularly it can be explained by data prefetch if the DSP does that? I seem to observe memory being read into cache speculatively without the DSP actually actively accessing it. E.g. I added a function to check whether a certain address is in cache. I call that function when allocating and freeing memory from heap and I see it confirming cached data where none should be there. If that's really the case and the DSP does data prefetch we need to rethink some code paths.

lgirdwood, 2022-12-02:
@lyakh TGL config.h
```
#define XCHAL_HAVE_PREFETCH		1	/* PREFCTL register */
#define XCHAL_HAVE_PREFETCH_L1		1	/* prefetch to L1 dcache */
#define XCHAL_PREFETCH_CASTOUT_LINES	2	/* dcache pref. castout bufsz */
#define XCHAL_PREFETCH_ENTRIES		8	/* cache prefetch entries */
#define XCHAL_PREFETCH_BLOCK_ENTRIES	0	/* prefetch block streams */
#define XCHAL_HAVE_CACHE_BLOCKOPS	0	/* block prefetch for caches */
#define XCHAL_HAVE_ICACHE_TEST		1	/* Icache test instructions */
#define XCHAL_HAVE_DCACHE_TEST		1	/* Dcache test instructions */
#define XCHAL_HAVE_ICACHE_DYN_WAYS	1	/* Icache dynamic way support */
#define XCHAL_HAVE_DCACHE_DYN_WAYS	1	/* Dcache dynamic way support */
```
What happens if you disable prefetch ?

lgirdwood, 2022-12-02:
Prefect should be an opt-in usage where and when needed, not a generic speculative usage.

lgirdwood, 2022-12-02:
@lyakh some grepping
```
src/arch/xtensa/include/xtensa/config/core.h:763:/*  Default PREFCTL value to enable prefetch.  */
src/arch/xtensa/include/xtensa/config/core.h:765:#define XCHAL_CACHE_PREFCTL_DEFAULT	0x00044	/* enabled, not aggressive */
```
usage in reset vector (for xtos config)
```
#if XCHAL_HAVE_PREFETCH
	/* Enable cache prefetch if present.  */
#if CONFIG_APOLLOLAKE

#if CONFIG_SKYLAKE || CONFIG_KABYLAKE
	movi.n	a2, 0	/* skylake and kabylake */
#else
	movi.n	a2, 34  /* apollolake */
#endif

#else
	movi.n	a2, 68  /* eveything else */
#endif
	wsr	a2, PREFCTL
#endif
```

lgirdwood, 2022-12-02:
@andyross @nashif - fyi, may need to fine tune cache prefetcher in Zephyr configs.
@lyakh the 68 above is 0x44 and the non aggressive default. Can you try 0 and rerun the test.

lyakh, 2022-12-05:
@lgirdwood the idea is good, thanks. Tried setting PREFCTL to 0 on all cores and it didn't help. So, I assume it isn't the reason.
OTOH I also was able to narrow it down a bit:
disabling `CONFIG_INCOHERENT` didn't help
but disabling `CONFIG_SOF_ZEPHYR_HEAP_CACHED` does seem to help
i.e. disabling the coherent API doesn't help, but making all heap allocations uncached does. Then instead of the letter I tried "just" making stream buffer allocations (`rballoc()`) uncached - and that alone seems to be enough too. I'm saying "seems" because the reproduction rate isn't 100% and with some modifications it can be about 50%, with others about 5%, so, it isn't certain

lyakh, 2022-12-05:
@andyross @nashif Ouch! Prefetch does play a role and not a good one! And it affects Zephyr Xtensa heap handling directly I think. I wrote a function that explicitly checks if heap headers end up in cache - and they do. And if I disable prefetch - they don't any more. So, we probably have something like
```
  hdr0            buf0               hdr1          buf1
|---------|----------------------|----------|------------------------|
```
We allocate `buf0`, then `buf1`, then we use `buf0` via its cached alias, at which point `hdr1` gets prefetched into cache too. Apart from triggering my checks this doesn't seem to cause errors by itself so far, because we never dirty that cache, but it's good to be aware of this, maybe there are other cases, when this can become a problem.

lyakh, 2022-12-06:
> @lyakh there is some work being done to fix some linking issues that can impact the aliases and does this impact IPC4 config ?

@lgirdwood you probably mean the effort to fix which sections are cached, and which aren't in the rimage context? Don't think this is related. This is a heap, the heap itself is in an uncached .data section and that's correct, and then cached and uncached allocations are decided at run-time.
IPC4 - yes, I think I have learned several things, that are also applicable to IPC4. E.g. about the heap - the above `hdr0-buf0-hdr1-buf1` chart has some important hidden features. We always align allocations on cache line boundaries, so `buf0` and `buf1` always begin on a cache-line boundary. But we don't align the end. So, say, if `buf0` is only some 20 bytes into a cache-line, then `hdr1` can be placed into the same cache line length. Then if we decide to access `buf0` via a cached alias, `hdr1` can be corrupted. So, e.g. we have to allocate buffer objects *as if* for cached alias, to make sure the end of the allocation is also on a cache-line boundary.
Then also allocating on the same core, prefetch - all these "lessons learned" apply to IPC4 too.

closed 2022-12-20 by kfrydryx