thesofproject/sof #6709: [BUG] HDA audio output garbled on cAVS2.5 with native Zephyr DMA+DAI

kv2019i, 2022-12-02:
**Describe the bug**
Test audio playback to HDA headset codec. 

**To Reproduce**
aplay file.wav

**Reproduction Rate**
100%

**Expected behavior**
Clean audio output

**Impact**
Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:  c693d0caf6fa
    * SOF: f6f9b62d5a1
2) Name of the topology file
    * Topology: sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**

**Notes**
Originally reported by @singalsu  on UP i11.

kv2019i, 2022-12-02:
FYI @mengdonglin @lgirdwood , this potentially affects sof-2.4 stable branch as well.

marc-hb, 2022-12-02:
@singalsu , @singalsu is this too small for https://github.com/thesofproject/sof-test/blob/main/test-case/check-alsabat.sh to notice?

kv2019i, 2022-12-03:
@marc-hb alsabat should definitely pick this up.

kv2019i, 2022-12-05:
Hmm, something is off here. @singalsu , @ujfalusi both got the audio problem. I just retried building kernel (sof-dev of today 41f9d4fe31be334ece6ec0a1829b730342f9c900), rebuild tplg (main of today 7029fb7b015912e9459f7146c2b23153281e98af) and rebuild FW (main of today 7029fb7b015912e9459f7146c2b23153281e98af) and I still get the garbled audio.

Dmic seems to be ok, so at least some of the DMAs are running as expected. The FW log also seems ok, performance is at expected level and no xruns are reported.

mengdonglin, 2022-12-05:
@kv2019i @singalsu @ujfalusi @keqiaozhang Do you have PulseAudio or Pipewire enabled during your test?

kv2019i, 2022-12-05:
@mengdonglin Not enabled, this happens with plain aplay to native PCM, so audio servers should not interfere.

Also tested today:
- forcing DMI-L1 off (-> no impact)
- older test branch for  native drivers: SOF c04b406fd0a1085ba5b1dee888c2bbcf0c601ba0 , Zephyr bd60b9289a635aedf80fc997e6b2a2de56fe5786 (-> issue still seen)

lgirdwood, 2022-12-05:
Do FW logs show any issues re underruns ?
Is audio garbled with less gain i.e. overall gain at -3dB or -6dB
Is there a wav capture to append ?

ujfalusi, 2022-12-05:
@lgirdwood, this is what I have in firmware log when playing audio to HDMI. The audio is garbled and it is playing in slow motion (1.5 times slower):
```
[    0.000000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7d000 initialized
[    0.000000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7c000 initialized
*** Booting Zephyr OS build ffed8388e7cb ***
[    0.000000] <inf> main: SOF on intel_adsp_cavs25
[    0.015530] <inf> init: FW ABI 0x301a000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-1798-g8c16ce371dbf zephyr:ffed8388e7cb src hash 0xdeb39d20 (ref hash 0xdeb39d20)
[    0.015550] <inf> power: pm_runtime_enable_dsp dsp_d0_sref 1
[    0.015648] <inf> ll_schedule: dma_single_chan_domain_init(): num_dma 2, clk 4
[    0.015668] <inf> ipc: ipc_init()
[    0.015686] <inf> idc: idc_init()
[    0.015701] <inf> dma_trace: dma_trace_init_complete()
[    0.015713] <inf> dma: dma_get() ID 4 sref = 1 busy channels 0
[    0.015845] <inf> main: SOF initialized
[    0.015921] <inf> idc: idc_init()
[    0.016133] <inf> ipc: rx    : 0x44000000|0x31400008
[    0.016295] <inf> ipc: rx    : 0x44000000|0x3060004c
[    0.046951] <inf> ipc: rx    : 0xe070000|0x180
[    0.046963] <inf> pipe: pipeline new pipe_id 8388608 priority 0
[    0.047005] <inf> dma: dma_get() ID 5 sref = 1 busy channels 0
[    0.047063] <inf> dma: dma_get() ID 7 sref = 1 busy channels 0
[    0.047066] <inf> ipc: comp:8388608 0x800002 dai_config() dai type = 3 index = 15 dd 0x9e080bc0
[    0.047081] <inf> ipc: buffer new size 0x180 id 8388608.8388611 flags 0x0
[    0.047105] <inf> pipe: comp:8388608 0x800001 connect buffer 8388611 as sink
[    0.047110] <inf> pipe: comp:8388608 0x800002 connect buffer 8388611 as source
[    0.047118] <inf> pipe: pipe:8388608 0x0 pipeline complete, clock freq 400000000Hz
[    0.047151] <inf> ipc: comp:8388608 0x800002 dai_data_config() dai type = 3 index = 15 dd 0x9e080bc0
[    0.047170] <inf> dai_comp: comp:8388608 0x800002 dai_playback_params() dest_dev = 0 stream_id = 0 src_width = 2 dest_width = 2
[    0.047175] <inf> dai_comp: comp:8388608 0x800002 dai_playback_params() fifo 0x0
[    0.047193] <inf> pipe: pipe:8388608 0x0 pipe prepare
[    0.047211] <inf> dai_comp: comp:8388608 0x800002 dai_prepare()
[    0.047215] <inf> dai_comp: comp:8388608 0x800002 dai_config_prepare(), channel = 0
[    0.047220] <inf> dai_comp: comp:8388608 0x800002 dai_config_prepare(): new configured dma channel index 0
[    0.047235] <inf> pipe: pipe:8388608 0x0 pipe trigger cmd 7
[    0.047241] <inf> ll_schedule: task add 0xbe081a40 0x20180U priority 0 flags 0x0
[    0.047261] <inf> ll_schedule: zephyr_domain_register domain->type 1 domain->clk 4 domain->ticks_per_ms 38400 period 1000
[    0.047400] <inf> host_comp: comp:8388608 0x800001 no bytes to copy, 384 free in buffer, 0 available in DMA
[    0.047406] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 0 + avail bytes 16 < period bytes 192, possible glitch
[    0.047411] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): nothing to copy
[    0.048380] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 128 + avail bytes 0 < period bytes 192, possible glitch
[    0.049353] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 128 + avail bytes 0 < period bytes 192, possible glitch
[    0.050350] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 128 + avail bytes 0 < period bytes 192, possible glitch
[    0.051350] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 128 + avail bytes 0 < period bytes 192, possible glitch
[    0.052350] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 128 + avail bytes 0 < period bytes 192, possible glitch
[    0.053350] <wrn> dai_comp: comp:8388608 0x800002 dai_copy(): Copy_bytes 128 + avail bytes 0 < period bytes 192, possible glitch
```
The <wrn> lines are flooding the log ^

Analog playback (garbled, distorted, but correct speed):
```
[    0.000000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7d000 initialized
[    0.000000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7c000 initialized
*** Booting Zephyr OS build ffed8388e7cb ***
[    0.000000] <inf> main: SOF on intel_adsp_cavs25
[    0.015351] <inf> init: FW ABI 0x301a000 DBG ABI 0x5003000 tags SOF:v2.0-rc1-1798-g8c16ce371dbf zephyr:ffed8388e7cb src hash 0xdeb39d20 (ref hash 0xdeb39d20)
[    0.015371] <inf> power: pm_runtime_enable_dsp dsp_d0_sref 1
[    0.015471] <inf> ll_schedule: dma_single_chan_domain_init(): num_dma 2, clk 4
[    0.015491] <inf> ipc: ipc_init()
[    0.015508] <inf> idc: idc_init()
[    0.015523] <inf> dma_trace: dma_trace_init_complete()
[    0.015536] <inf> dma: dma_get() ID 4 sref = 1 busy channels 0
[    0.015668] <inf> main: SOF initialized
[    0.015743] <inf> idc: idc_init()
[    0.015803] <inf> ipc: rx    : 0x44000000|0x31400008
[    0.015905] <inf> ipc: rx    : 0x44000000|0x3060004c
[    0.184833] <inf> ipc: rx    : 0x11000005|0x0
[    0.184843] <inf> pipe: pipeline new pipe_id 0 priority 0
[    0.185081] <inf> ipc: rx    : 0x40000004|0x15
[    0.185115] <inf> ipc: buffer new size 0x180 id 0.0 flags 0x0
[    0.185148] <inf> dma: dma_get() ID 5 sref = 1 busy channels 0
[... 42 lines cut ...]
[    0.191961] <inf> dai_comp: comp:1 0x40001 dai_playback_params() dest_dev = 0 stream_id = 0 src_width = 4 dest_width = 4
[    0.191966] <inf> dai_comp: comp:1 0x40001 dai_playback_params() fifo 0x0
[    0.191993] <inf> pipe: pipe:1 0x0 pipe prepare
[... cut ...]
[    0.194235] <err> copier: comp:0 0x40000 failed to find dai comp or sink pipeline not running.
[    0.194250] <inf> host_comp: comp:0 0x40000 no bytes to copy, 384 free in buffer, 0 available in DMA
[    1.215231] <inf> ll_schedule: ll timer avg 2666, max 3992, overruns 0
[    2.239231] <inf> ll_schedule: ll timer avg 2666, max 3088, overruns 0
[    3.263231] <inf> ll_schedule: ll timer avg 2666, max 3069, overruns 0
[    4.287231] <inf> ll_schedule: ll timer avg 2664, max 3081, overruns 0
```
Nothing special to my eyes in the log.

bardliao, 2022-12-06:
I can reproduce the issue on my Up Xtreme board (TGL) with the specific recipe
Kernel: c693d0caf6fa
SOF: https://github.com/thesofproject/sof/commit/f6f9b62d5a17a416d203ae531ae652ae131837c2

bardliao, 2022-12-06:
Reverting below change can fix the issue on my Up Xtreme board
24c0aad83 tgl: ipc4: zephyr: move to native drivers

bardliao, 2022-12-06:
HDA headset capture has the same issue. 
Captured wav: [hs.zip](https://github.com/thesofproject/sof/files/10165744/hs.zip)

keqiaozhang, 2022-12-07:
Thanks @bardliao for debugging. I can reproduce the issue with speaker-test on my TGL-UP board and this issue cannot be detected by a single frequency sine wav.

singalsu, 2022-12-07:
Need to use some sine wave frequency that does not have a period that matches with driver and buffer sizes. E.g. 1 kHz is bad for testing.  Take a test frequency that s a prime number near 1 kHz. E.g. 997 Hz is a prime but if it doesn't drift enough vs 1 ms multiple buffers, take frequency that is more off e.g. 919 Hz. You can check with Octave if a number is a prime

```
>> isprime(919)
ans = 1
```

keqiaozhang, 2022-12-07:
This issue can be preproduced on MTL-HDA with main branch, but with mtl-002-stable-drop, this issue cannot be reproduced.

marc-hb, 2022-12-07:
> E.g. 1 kHz is bad for testing.

What does `check-alsabat.sh` do by default and should we change it?

kv2019i, 2022-12-07:
@abonislawski @softwarecki Does this ring any bell? We have distortion with HDA link DMA with main but not with mtl-002. I was checking the delta and at least "dma: hda: calculate correct size if block_count > 1" is missing from Zephyr main, but that did not solve this issue. Any other patches that could explain the difference?

kv2019i, 2022-12-07:
During failing test case, DMA link driver programming looks like:

```
[    0.275365] <inf> dma_hda: hdalinkout: size=768, blk_count=2, channel 0, regblocksize=u
[    0.275371] <inf> dma_hda: hdalinkout: source_data_size=4
[    0.275551] <inf> dma_hda: hdalinkout: reload regblock=32, ch=0, size=144
[    0.276510] <inf> dma_hda: hdalinkout: reload regblock=32, ch=0, size=352
[    0.277566] <inf> dma_hda: hdalinkout: reload regblock=32, ch=0, size=384
```

Without above patch, block-count of 2 is not handled by the driver and thus audio output is garbled.

keqiaozhang, 2022-12-07:
> 997 Hz is a prime but if it doesn't drift enough vs 1 ms multiple buffers, take frequency that is more off e.g. 919 Hz.

Thanks @singalsu , I checked the alsabat with 919Hz and it did catch the error. I think we can add more special frequency sine wavs in alsabat test.
```
ubuntu@sh-tglu-up-hda-04:~$ ~/sof-test/test-case/check-alsabat.sh -p hw:sofhdadsp,0 -c hw:Device,0
2022-12-07 13:25:31 UTC [INFO] ktime=3257 sof-test PID=2479: starting
2022-12-07 13:25:31 UTC [WARNING] NO MODEL is defined. Please define MODEL to run alsa_settings/MODEL.sh
2022-12-07 13:25:31 UTC [INFO] Starting /home/ubuntu/.local/bin/mtrace-reader.py
2022-12-07 13:25:31 UTC [INFO] check the PCMs before alsabat test
2022-12-07 13:25:33 UTC [COMMAND] alsabat -Phw:sofhdadsp,0 --standalone -n 240000 -r 48000 -c 2 -F 919
alsa-utils version 1.2.2

Entering playback thread (ALSA).
Get period size: 2048  buffer size: 16384
Playing generated audio sine wave
2022-12-07 13:25:34 UTC [COMMAND] alsabat -Chw:Device,0 -c 1 -r 48000 -F 919
alsa-utils version 1.2.2

Entering capture thread (ALSA).
Get period size: 3000  buffer size: 24000
Recording ...
Capture completed.

BAT analysis: signal has 65536 frames at 48000 Hz, 1 channels, 2 bytes per sample.

Channel 1 - Checking for target frequency 919.00 Hz
Amplitude: 23195.4; Percentage: [70]
Detected peak at 418.95 Hz of 28.98 dB
 Total 30.0 dB from 418.21 to 419.68 Hz
 FAIL: Peak freq too low 418.95 Hz
Detected peak at 580.81 Hz of 23.57 dB
 Total 31.3 dB from 580.81 to 581.54 Hz
 FAIL: Peak freq too low 580.81 Hz
Detected peak at 919.19 Hz of 36.81 dB
 Total 40.0 dB from 913.33 to 925.05 Hz
 PASS: Peak detected at target frequency
Detected peak at 1419.43 Hz of 27.79 dB
 Total 40.7 dB from 1417.24 to 1420.90 Hz
 FAIL: Peak freq too high 1419.43 Hz
Detected peak at 1581.30 Hz of 21.80 dB
 Total 40.7 dB from 1581.30 to 1581.30 Hz
 FAIL: Peak freq too high 1581.30 Hz
Detected peak at 2419.19 Hz of 24.27 dB
 Total 40.8 dB from 2419.19 to 2419.19 Hz
 FAIL: Peak freq too high 2419.19 Hz
Detected peak at 2581.05 Hz of 20.20 dB
 Total 40.9 dB from 2581.05 to 2581.05 Hz
 FAIL: Peak freq too high 2581.05 Hz
Detected peak at 3418.95 Hz of 20.19 dB
 Total 40.9 dB from 3418.95 to 3419.68 Hz
 FAIL: Peak freq too high 3418.95 Hz
Detected peak at 4419.43 Hz of 20.52 dB
 Total 41.0 dB from 4419.43 to 4419.43 Hz
 FAIL: Peak freq too high 4419.43 Hz
Detected at least 9 signal(s) in total

Return value is -1003
2022-12-07 13:25:36 UTC [ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2022-12-07 13:25:36 UTC [ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-alsabat.sh
2022-12-07 13:25:36 UTC [INFO] pkill -TERM -f mtrace-reader.py
2022-12-07 13:25:36 UTC [INFO] ktime=3262 sof-test PID=2479: ending
2022-12-07 13:25:36 UTC [WARNING] Process(es) started by /home/ubuntu/sof-test/test-case/check-alsabat.sh are still active, killing these process(es):
2022-12-07 13:25:36 UTC [WARNING] Catch pid: 2564 alsabat -Phw:sofhdadsp,0 --standalone -n 240000 -c 2 -r 48000 -F 919
2022-12-07 13:25:36 UTC [WARNING] Kill cmd:'alsabat -Phw:sofhdadsp,0 --standalone -n 240000 -c 2 -r 48000 -F 919' by kill -9
2022-12-07 13:25:36 UTC [INFO] Test Result: FAIL!
```

kv2019i, 2022-12-07:
@keqiaozhang I submitted https://github.com/thesofproject/sof-test/pull/990 for sof-test

singalsu, 2022-12-07:
@keqiaozhang If there are other tests with 1 kHz it's a good idea to replace in all of them the frequency. When the frequency is a prime it will ensure all possible PCM codes are exercised in the test over time. With 1 kHz @48 kHz only 25 different PCM code values are used.

keqiaozhang, 2022-12-19:
@kv2019i this issue is still reproducible on main branch, any Zephyr change missed on main branch?

kv2019i, 2022-12-19:
@keqiaozhang This bug will be fixed with this fix, which is still in review: https://github.com/thesofproject/sof/pull/6823

marc-hb, 2023-01-04:
Stupid Github auto-close not waiting for daily tests results.

keqiaozhang, 2023-01-09:
Confirmed that this issue is fixed by https://github.com/thesofproject/sof/pull/6823

closed 2023-01-09 by keqiaozhang