thesofproject/sof #6710: [BUG] TGL-H IPC4 Zephyr build is broken in main

kv2019i, 2022-12-02:
**Describe the bug**
The IPC4 native drivers is build is broken since commit "tgl: ipc4: zephyr: move to native drivers" as the IPC4 overlay for TGL-H was also updated to native drivers, but upstream Zephyr does not have all the needed definitions for intel_adsp_cavs25_tgph.

**To Reproduce**
Build SOF with
./scripts/xtensa-build-zephyr.py -i IPC4 tgl-h

**Reproduction Rate**
100%

**Expected behavior**
Firmware can be built

**Impact**
Showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel:  c693d0caf6fa
    * SOF: f6f9b62d5a1
2) Name of the topology file
    * Topology: sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL-H

**Screenshots or console output**

**Notes**

kv2019i, 2022-12-02:
FYI @mengdonglin @lgirdwood  most probably affects sof-2.4 stable branch.

marc-hb, 2022-12-07:
Why is this very recent tgl-h build successful?
https://sof-ci.01.org/sofpr/PR6733/build2695/build/tgl-h_zph.txt

marc-hb, 2022-12-07:
Adding -i IPC4 tgl-h is a 4 characters addition in https://github.com/thesofproject/sof/blob/main/.github/workflows/zephyr.yml

kv2019i, 2022-12-15:
@marc-hb wrote:
> Adding -i IPC4 tgl-h is a 4 characters addition in https://github.com/thesofproject/sof/blob/main/.github/workflows/zephyr.yml

I tried but I had to use more than four:
https://github.com/thesofproject/sof/pull/6816

kv2019i, 2022-12-15:
Fixed with #6807 , closing.

closed 2022-12-15 by kv2019i