thesofproject/sof #6711: [BUG] failed to bind modules copier.module.8.2:1 -> copier.module.17.2:0 when test multiple-pipeline-capture with IPC4 Tester on MTL_RVP_NOCODEC

miRoox, 2022-12-05:
**Describe the bug**
CI observed it failed to failed to bind modules copier.module.8.2:1 -> copier.module.17.2:0 when test multiple-pipeline-capture with **IPC4 Tester** on MTL_RVP_NOCODEC

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-mtl-nocodec.tplg MODEL=MTLP_RVP_NOCODEC_ACE ~/sof-test/test-case/multiple-pipeline.sh -f c
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/41f9d4fe31be
    * firmware: IPC4 tester (closed firmware)
    * SOF (topology): https://github.com/thesofproject/sof/commit/8c16ce371dbf
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL_RVP_NOCODEC

**Screenshots or console output**

**dmesg**
```
[ 2515.636027] kernel: snd_sof:sof_ipc4_route_setup: sof-audio-pci-intel-mtl 0000:00:1f.3: bind copier.module.8.2:1 -> copier.module.17.2:0
[ 2515.636032] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x45010004|0x8040004: MOD_BIND
[ 2515.637145] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx      : 0x1b0a0000|0x8060: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 2515.637149] kernel: snd_sof:sof_ipc4_rx_msg: sof-audio-pci-intel-mtl 0000:00:1f.3: Unhandled DSP message: 0x1b0a0000|0x8060
[ 2515.637153] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx done : 0x1b0a0000|0x8060: GLB_NOTIFICATION|EXCEPTION_CAUGHT
[ 2516.142349] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x45010004|0x8040004
[ 2516.142394] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 2516.142398] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2516.142431] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc5010004|0x8040004|0x0, target: 0x1b0a0000|0x8060|0x0, ctl: 0x3
[ 2516.142460] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 2516.142478] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 2516.142497] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 2516.142511] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 2516.142543] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0xc0000005, ROM error: 0xfff
[ 2516.142562] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[ 2516.142586] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 2516.142601] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 2516.142623] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to bind modules copier.module.8.2:1 -> copier.module.17.2:0
[ 2516.142659] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules copier.module.17.2:0 -> gain.17.1:0
[ 2516.142677] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-mtl 0000:00:1f.3: widget copier.module.17.2 freed
[ 2516.142681] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules gain.17.1:0 -> copier.host.17.1:0
[ 2516.142686] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-mtl 0000:00:1f.3: widget gain.17.1 freed
[ 2516.142694] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12020000|0x0: GLB_DELETE_PIPELINE
[ 2516.646317] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x12020000|0x0
[ 2516.646346] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 2516.646363] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.17
[ 2516.646387] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
[ 2516.646409] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed widget list set up for pcm 12 dir 1
[ 2516.646432] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -110
[ 2516.646485] kernel:  ssp-capture: ASoC: error at __soc_pcm_hw_params on ssp-capture: -110
[ 2516.646512] kernel:  ssp-capture: ASoC: error at dpcm_fe_dai_hw_params on ssp-capture: -110
```

bardliao, 2022-12-05:
Based on my experiment, it can be fixed if we set sink format before binding the widgets.

```patch
diff --git a/sound/soc/sof/ipc4-topology.c b/sound/soc/sof/ipc4-topology.c
index 4cab0f0d1241..07f290ef4745 100644
--- a/sound/soc/sof/ipc4-topology.c
+++ b/sound/soc/sof/ipc4-topology.c
@@ -2139,6 +2139,17 @@ static int sof_ipc4_route_setup(struct snd_sof_dev *sdev, struct snd_sof_route *
                return sroute->dst_queue_id;
        }

+       /* Pin 0 format is already set during copier module init */
+       if (sroute->src_queue_id > 0 && WIDGET_IS_COPIER(src_widget->id)) {
+               ret = sof_ipc4_set_copier_sink_format(sdev, src_widget, sink_widget,
+                                                     sroute->src_queue_id);
+               if (ret < 0) {
+                       dev_err(sdev->dev, "failed to set sink format for %s source queue ID %d\n",
+                               src_widget->widget->name, sroute->src_queue_id);
+                       goto out;
+               }
+       }
+
        dev_dbg(sdev->dev, "bind %s:%d -> %s:%d\n",
                src_widget->widget->name, sroute->src_queue_id,
                sink_widget->widget->name, sroute->dst_queue_id);
@@ -2165,17 +2176,6 @@ static int sof_ipc4_route_setup(struct snd_sof_dev *sdev, struct snd_sof_route *
                goto out;
        }

-       /* Pin 0 format is already set during copier module init */
-       if (sroute->src_queue_id > 0 && WIDGET_IS_COPIER(src_widget->id)) {
-               ret = sof_ipc4_set_copier_sink_format(sdev, src_widget, sink_widget,
-                                                     sroute->src_queue_id);
-               if (ret < 0) {
-                       dev_err(sdev->dev, "failed to set sink format for %s source queue ID %d\n",
-                               src_widget->widget->name, sroute->src_queue_id);
-                       goto out;
-               }
-       }
-
        return ret;

 out:

```

I think we should set sink format first. But I didn't find the related code on IPC4 tester yet. @RanderWang @ranj063 FYI

lgirdwood, 2022-12-05:
@miRoox does this work on main branch ?

bardliao, 2022-12-06:
> @bardliao I think it makes sense to set the sink format before binding the modules as well. But I thought it was @RanderWang's recommendation to do it the other way around

I checked the IPC4 tester source code. It needs to get pin ibs and obs while binding. So, we need to set sink format before binding. I will submit a PR for it.
The change will fix the issue when we run arecord on pcm 0 and pcm 12 manually. However, the ` ~/sof-test/test-case/multiple-pipeline.sh -f c` still fail with IO error.

RanderWang, 2022-12-06:
> @bardliao I think it makes sense to set the sink format before binding the modules as well. But I thought it was @RanderWang's recommendation to do it the other way around

yes, it was my advice. I checked the TGL FW code and thought it was ok, but it doesn't work on MTL.