thesofproject/sof #6723: [BUG] I/O error when check-pause-resume-capture on MTL platforms

miRoox, 2022-12-06:
**Describe the bug**
CI observed an Input/output error when check-pause-resume-capture-100 on MTLP_RVP_SDW and MTLP_RVP_HDA.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-mtl-rt711-4ch.tplg MODEL=MTLP_RVP_SDW ~/sof-test/test-case/check-pause-resume.sh -c 100 -m capture
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/321028577de9
    * SOF: https://github.com/thesofproject/sof/commit/4a9a99aeb24a
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-mtl-rt711-4ch.tplg, sof-ace-tplg/sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_SDW, MTLP_RVP_HDA

**Screenshots or console output**

**console output**
```
=== PAUSE ===                                                            
(100/100) Wait for 121 ms before pause

(100/100) Wait for 181 ms before resume

declare -- cmd="journalctl_cmd --since=@1670281836"
2022-12-05 23:11:08 UTC [REMOTE_INFO] Entering expect script with:
      arecord   -D hw:0,10 -r 48000 -c 4 -f S32_LE -vv -i /dev/null -q
spawn arecord -D hw:0,10 -r 48000 -c 4 -f S32_LE -vv -i /dev/null -q
Hardware PCM card 0 'sof-soundwire' device 10 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 4
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 4096
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 4096
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0

#+                                                 | 00%
(1/100) Wait for 142 ms before pause
 
#+                                                 | 00%
=== PAUSE ===                                                            
(1/100) Wait for 107 ms before pause

(1/100) Wait for 104 ms before resume
                                                                           
 
#+                                                 | 00%
=== PAUSE ===                                                            
(2/100) Wait for 189 ms before pause

(2/100) Wait for 181 ms before resume
                                                                           
 arecord: pcm_read:2221: read error: Input/output error
2022-12-05 23:11:10 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2022-12-05 23:11:10 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/check-pause-resume.sh
```

keqiaozhang, 2023-02-03:
This issue also happened on MTL-NOCODEC when testing signal-stop-start.

marc-hb, 2023-02-03:
... since @fredoh9 just fixed signal-stop-start's "green failure" in
- thesofproject/sof-test#1000

keqiaozhang, 2023-02-20:
This issue also can be reproduced with signal stop/start on MTL platforms. Changing the priority to P1.

mszleszy, 2023-02-24:
@miRoox Unfortunately logs contain only two pipelines creation and there are no payloads from ipc messages.
Are you able to capture ipc logs (ipc messages with payloads) of all pipelines necessary for reproduction?

lgirdwood, 2023-03-01:
@keqiaozhang are you able to help here ?

marc-hb, 2023-03-01:
The reproduction steps in the description seem very thorough to me. Firmware aside, the main requirement is of course to compile and install the Linux kernel version referenced; I don't see how this Linux test failure could be investigated and fixed without using Linux.

mengdonglin, 2023-03-02:
> @keqiaozhang are you able to help here ?

@lgirdwood I think we need the driver feature to dump IPC flows. It's hard to collect manually.

mengdonglin, 2023-03-02:
This issue is observed with all audio interfaces (HD-Audio, I2S, SDW): https://sof-ci.ostc.intel.com/#/result/planresultdetail/21529

keqiaozhang, 2023-03-03:
I defined DEBUG_VERBOSE in ipc4.c. 

Here is the ipc message with payload on MTL-NOCODEC platform:
**kernel log**
<details>
<code><pre>
<summary>ipc with payload </summary>
[  229.017510] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[  229.017522] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  229.020782] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[  229.020792] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[  229.020801] snd_sof_intel_hda_common:hda_dsp_state_log: sof-audio-pci-intel-mtl 0000:00:1f.3: Current DSP power state: D0I0
[  229.020805] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 0 -> 2
[  229.020809] snd_sof:snd_sof_load_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: loading firmware
[  229.020811] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 2 -> 3
[  229.021869] snd_sof_intel_hda_common:mtl_dsp_pre_fw_run: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x1010000 successful
[  229.022933] snd_sof_intel_hda_common:mtl_dsp_pre_fw_run: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1d1c]=0x8151 successful
[  229.022955] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: booting DSP firmware
[  229.022964] snd_sof_intel_hda_common:hda_dsp_cl_boot_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: IMR restore supported, booting from IMR directly
[  229.024020] snd_sof_intel_hda_common:mtl_dsp_core_power_up: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000101 successful
[  229.024033] snd_sof_intel_hda_common:mtl_dsp_cl_init: sof-audio-pci-intel-mtl 0000:00:1f.3: Primary core power up successful
[  229.024047] snd_sof_intel_hda_common:mtl_dsp_cl_init: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x73214]=0x80000000 successful
[  229.024072] snd_sof_intel_hda_common:mtl_enable_interrupts: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1800]=0x41 successful
[  229.024085] snd_sof_intel_hda_common:mtl_enable_interrupts: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1140]=0x1 successful
[  229.134368] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx      : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  229.134385] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 3 -> 6
[  229.134397] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc rx done : 0x1b080000|0x0: GLB_NOTIFICATION|FW_READY
[  229.134584] snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-mtl 0000:00:1f.3: firmware boot complete
[  229.134597] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 6 -> 7
[  229.134620] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[  229.134951] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x64000000|0x31400008: MOD_LARGE_CONFIG_SET
[  229.134999] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[  229.135018] Message payload: 00000000: a3e2e3ae ffedcb58
[  229.135028] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  229.135414] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[  229.135664] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  229.135675] Message payload: 00000000: 00000100 00001000 00000001 000001ef
[  229.135680] Message payload: 00000010: 00000000 00000000 00000000 00000000
[  229.135683] Message payload: 00000020: 00000000 00000000 00000000 00000000
[  229.135685] Message payload: 00000030: 00000000 00000000 00000000 00000000
[  229.135688] Message payload: 00000040: 00000000 00000000 00000000
[  229.135713] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: open stream 0 dir 0
[  229.135718] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: period min 192 max 16384 bytes
[  229.135722] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: period count 2 max 16
[  229.135725] snd_sof:sof_pcm_open: sof-audio-pci-intel-mtl 0000:00:1f.3: buffer max 65536 bytes
[  229.136024] snd_sof:sof_pcm_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: hw params stream 0 dir 0
[  229.136035] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1c0]=0x40000 successful
[  229.136060] snd_sof_intel_hda_common:hda_dsp_stream_hw_params: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1c0]=0x40000 successful
[  229.136069] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-mtl 0000:00:1f.3: period_bytes:0x4000
[  229.136072] snd_sof_intel_hda_common:hda_dsp_stream_setup_bdl: sof-audio-pci-intel-mtl 0000:00:1f.3: periods:4
[  229.136096] snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-mtl 0000:00:1f.3: copier copier.host.1.1, type 23
[... cut ...]
[00:00:03.604,165] <inf> ll_schedule: zephyr_domain_unregister: zephyr_domain_unregister domain->type 1 domain->clk 3
[00:00:03.604,798] <inf> ipc: ipc_cmd: rx	: 0x13000002|0x1
[00:00:03.604,808] <inf> pipe: pipeline_trigger: pipe:1 0x0 pipe trigger cmd 0
[00:00:03.604,821] <inf> smart_amp_test: smart_amp_trigger: comp:1 0xf0000 smart_amp_trigger(), command = 0
[00:00:03.605,001] <err> dai_intel_ssp: dai_ssp_poll_for_register_delay poll timeout reg 163848 mask 4 val 4 us 125
</code></pre>
</details>

kv2019i, 2023-03-06:
@kfrydryx @mwasko P1 and nobody assigned, can you follow-up? Keeping on v2.5 milestone.

mwasko, 2023-03-06:
@kv2019i, @miRoox without a log with IPC payloads it is impossible to reproduce the same scenario with FW tests which is a first step for us to start debugging.

@mszleszy can you try to reproduce the scenario using the new logs?

kfrydryx, 2023-03-06:
@kv2019i @mwasko Work on reproduction using given ipc payloads is already in progress.

mszleszy, 2023-03-08:
I found two problems based on logs:
1. Problem with second topology run - I'm sending logs and reproduction script to dev team.
2. First topology looks improper. sink of first module is 32/32bit and it's connected to 16/32bit source.. There is no sample size conversion performed.
[image]

ranj063, 2023-03-09:
@mszleszy my kernel PR fixes.the start/stop playback test on mtl nocodec

lgirdwood, 2023-03-13:
> @mszleszy my kernel PR fixes.the start/stop playback&capture tests on mtl nocodec

@ranj063 @mszleszy everything merged/good now, can we close ?

ranj063, 2023-03-14:
> @ranj063 @mszleszy everything merged/good now, can we close ?

@lgirdwood no my lermel PR https://github.com/thesofproject/linux/pull/4235 is still not merged. But even after this is merged, the issue lingers on the MTL SDW and HDA devices with the same IO error during DMIC capture.

mszleszy, 2023-03-14:
@lgirdwood no changes on fw side, and issue is reproduced on fw testing framework. In my opinion issue s still valid.

plbossart, 2023-03-14:
> > @ranj063 @mszleszy everything merged/good now, can we close ?
> 
> @lgirdwood no my lermel PR [thesofproject/linux#4235](https://github.com/thesofproject/linux/pull/4235) is still not merged. But even after this is merged, the issue lingers on the MTL SDW and HDA devices with the same IO error during DMIC capture.

Can you clarify the status @ranj063: you said in comments both "my kernel PR fixes.the start/stop playback&capture tests on mtl nocodec" and "the issue lingers on the MTL SDW and HDA devices with the same IO error during DMIC capture".

Which is it ?

ranj063, 2023-03-14:
It is exactly the two of them @plbossart . This issue combines 2 separate issues. 1. Pause/release failure on dmic capture 2. Start/stop failure on mtl nocodec. My PR fixes only the second part.

plbossart, 2023-03-14:
Then this issue needs to be split in two...

It's not good if the problems are fudged together.

marc-hb, 2023-03-15:
So how about simply refocusing this issue on pause/release? That's what the initial description was about and it's what most comments seem to have been about. Moreover the start/stop is fixed. Also, the issue is not very long yet, not too hard to skip the few start/stop comments.

juimonen, 2023-03-15:
> It is exactly the two of them @plbossart . This issue combines 2 separate issues. 1. Pause/release failure on dmic capture 2. Start/stop failure on mtl nocodec. My PR fixes only the second part.

I dont get why it fixes it in nocodec, but not in hda/sdw.... isnt the dmic the same?

juimonen, 2023-03-15:
[dmic4.log](https://github.com/thesofproject/sof/files/10980696/dmic4.log)

above the log from mtl device with 4ch dmic recording. This has linux 4235 included. There's prints from dai_copy with dma_cb and host copy (both with free samples in sink and source buffers). Seems that dmic dma just overruns at some point after 0.5-1s of transfer...

ranj063, 2023-03-15:
> I dont get why it fixes it in nocodec, but not in hda/sdw.... isnt the dmic the same?

@juimonen i was wrong about MTL nocodec. the pause-release capture test on nocodec is a pass in our tests but it is only because it has avoided testing the DMIC PCMs all along

juimonen, 2023-03-15:
@ranj063 btw just added some extra logging to dai and host and now the dmic is not overrunning... removed the extra log and overrun is back.... something very fishy here.

serhiy-katsyuba-intel, 2023-03-15:
@juimonen , in dmic4.log host copy_bytes is always less or equal to 768 bytes. It's a period size for 4 ch 32 bit. This is true even for cases when host src_bytes and host sink_bytes are both > 768:

[00:00:00.805,673] <wrn> host_comp: comp:1 0x40002 host src_bytes 1536 sink_bytes 3072 copy_bytes 768

So there is some place in code that limits maximum amount of copied bytes to period size. This eventually will lead to buffer overflow at some point. We should allow to copy more bytes then period size.

serhiy-katsyuba-intel, 2023-03-15:
> ... This eventually will lead to buffer overflow at some point. We should allow to copy more bytes then period size.

Indeed, host DMA buffer overflow happened in the log:
[00:00:01.448,671] <inf> host_comp: host_get_copy_bytes_normal: comp:1 0x40002 no bytes to copy, available bytes: 1536, free_bytes: 0

However, that might be another issue, not related to DMIC problem.

kv2019i, 2023-03-15:
This seems to be MTL specific, removing from v2.5

serhiy-katsyuba-intel, 2023-03-15:
Another problem in dmic4.log:
```
[00:00:01.442,690] <wrn> dai_comp: comp:0 0x40000 dai_copy(): nothing to copy
[00:00:01.443,671] <wrn> host_comp: comp:1 0x40002 host src_bytes 1536 sink_bytes 3072 copy_bytes 768
[00:00:01.443,685] <wrn> dai_comp: comp:0 0x40000 dai_copy(): Copy_bytes 0 + free bytes 64 < period bytes 768, possible glitch
[00:00:01.443,688] <wrn> dai_comp: comp:0 0x40000 dai_copy(): nothing to copy
[00:00:01.444,671] <wrn> host_comp: comp:1 0x40002 host src_bytes 1536 sink_bytes 3072 copy_bytes 768
[00:00:01.444,691] <wrn> dai_comp: comp:0 0x40000 dai_copy(): nothing to copy
[00:00:01.445,715] <wrn> host_comp: comp:1 0x40002 host src_bytes 1536 sink_bytes 2304 copy_bytes 768
[00:00:01.445,731] <err> dma_dw_common: xrun detected
```

There were 3 "dai_copy(): nothing to copy" happened before "xrun detected". "dai_copy(): nothing to copy" means either no source bytes available in DMA buffer or no free space in DAI local buffer. The problem is that after "dai_copy(): nothing to copy" dai_copy() exists and does not call dma_reload(): 

```
	/* return if nothing to copy */
	if (!copy_bytes) {
		comp_warn(dev, "dai_copy(): nothing to copy");
		return 0;
	}

<skip>

	ret = dma_reload(dd->chan->dma->z_dev, dd->chan->index, 0, 0, copy_bytes);
```

dma_reload() calls intel_adsp_gpdma_copy() in Zephyr which is responsible to clear DONE bits from LLI list. If dma_reload() was not called for a time longer then DMA fetches uncleared LLI with DONE bit set, DMA will stop.
In our case dma_reload() was not called for 3 ms. Maybe that was enough.

The easiest solution to test this theory is to hack dai_copy() to call dma_reload() even in case of "nothing to copy" and see if it helps.

juimonen, 2023-03-15:
@serhiy-katsyuba-intel yes adding the dma_reload to "nothing to copy" branch is making the overrun disappear. not sure that is a proper fix but anyway seems to help. The host free_bytes is the dma buffer between dsp and linux, so there is no room in some occasions, but downstream (aka dmic) has bytes to offer. It is kind of hard to follow the flow, but to me it looks like the host dma transfer has some glitch or takes too long to transfer (and make more room for dmic) -> dmic dai copy runs without copy couple of times and does not call dma reload -> dmic dma and pipeline stops eventually.

ranj063, 2023-03-15:
> @serhiy-katsyuba-intel yes adding the dma_reload to "nothing to copy" branch is making the overrun disappear. not sure that is a proper fix but anyway seems to help. The host free_bytes is the dma [...]

@juimonen I think you are testing a different scenario than whats reported in this issue. With the pause/release test calling dma_reload() doesnt help

serhiy-katsyuba-intel, 2023-03-15:
host copy only copies maximum of period size bytes. So if in case of some short temporary problem with sending data out of DSP the amount of data in host source buffer will increase to be more then a period size and will never became 0 as host only copies maximum of period size bytes. Several such temporary problems will lead to host source buffer overflow. I guess we should modify host copy to allow to copy more then period bytes.

ranj063, 2023-03-16:
@serhiy-katsyuba-intel @juimonen I found a potential fix for the problem. This is current sequence during pause/release wrt GPDMA now:

dma_suspend ->dma_stop->dma_config->dma_start()

But if I change this to do this instead, I can pause/release the DMIC capture without issues:
dma_suspend->dma_stop->dma_start().

The changes needed are:

```
diff --git a/src/audio/dai-zephyr.c b/src/audio/dai-zephyr.c
index eb4aa9ca9..6dab720c2 100644
--- a/src/audio/dai-zephyr.c
+++ b/src/audio/dai-zephyr.c
@@ -1056,11 +1056,6 @@ static int dai_comp_trigger_internal(struct comp_dev *dev, int cmd)
                                        return ret;
                        }
 
-                       /* dma_config needed after stop */
-                       ret = dma_config(dd->chan->dma->z_dev, dd->chan->index, dd->z_config);
-                       if (ret < 0)
-                               return ret;
-
                        /* start the DAI */
                        dai_trigger_op(dd->dai, cmd, dev->direction);
                        ret = dma_start(dd->chan->dma->z_dev, dd->chan->index);
```
```
diff --git a/drivers/dma/dma_dw_common.c b/drivers/dma/dma_dw_common.c
index 9dc24f63e4..84ad38c613 100644
--- a/drivers/dma/dma_dw_common.c
+++ b/drivers/dma/dma_dw_common.c
@@ -442,6 +442,15 @@ int dw_dma_start(const struct device *dev, uint32_t channel)
 
        struct dw_dma_chan_data *chan_data = &dev_data->chan[channel];
 
+       if (chan_data->state == DW_DMA_IDLE) {
+               chan_data->state = DW_DMA_ACTIVE;
+
+               /* enable the channel */
+               dw_write(dev_cfg->base, DW_DMA_CHAN_EN, DW_CHAN_UNMASK(channel));
+
+               goto out;
+       }
+
        /* validate channel state */
        if (chan_data->state != DW_DMA_PREPARED) {
                LOG_ERR("%s: dma %s channel %d not ready ena 0x%x status 0x%x",
```

juimonen, 2023-03-16:
@ranj063 hmm ok, did you check what happens with ssp with this change? and ssp/dmic in tgl? I'm little bit puzzled what is different here between mtl/tgl, is the dma working in different way etc?

serhiy-katsyuba-intel, 2023-03-16:
Found a suspicious code in Zephyr dmic.c. In dai_dmic_stop() CIC_CONTROL_SOFT_RESET_BIT bit is set to put DMIC in reset. However, in dai_dmic_start() clearing of CIC_CONTROL_SOFT_RESET_BIT is put into #ifndef CONFIG_SOC_SERIES_INTEL_ACE and so is only done for non MTL platforms.
Tagging @dnikodem 
Removing all "#ifndef CONFIG_SOC_SERIES_INTEL_ACE" (there is a second #ifndef for CIC_CONTROL_MIC_MUTE_BIT which seems also not needed/wrong) from dmic.c and running the test could verify the theory.

singalsu, 2023-03-16:
@dnikodem @serhiy-katsyuba-intel If the unreset is done for MTL it should be done before setting the start bits to use the HW sync start feature.

juimonen, 2023-03-16:
sorry about polluting this bug with start/stop issue, but it is real in my mtl device and it is caused by this:
```
commit 5d210ccb695ad53bfd672f7f7223d1316f4e62a3
Author: Baofeng Tian <baofeng.tian@intel.com>
Date:   Wed Feb 22 11:15:25 2023 +0800

    host-zephyr: limit copy bytes to one period for capture pipeline

    Limit copy bytes to period bytes for both playback and capture
    to avoid high load spike.

    Signed-off-by: Baofeng Tian <baofeng.tian@intel.com>

```
I have no idea why it is not causing issues in CI machines, maybe my device is running so on the limit etc.

@serhiy-katsyuba-intel I guess this recent change is exactly limiting for 1 period instead of more data to be pushed. @btian1 FYI.

So I can't say for sure this is a real issue, or just my borked device, or could we face this issue in other scenarios...

ranj063, 2023-03-16:
@serhiy-katsyuba-intel @juimonen The real issue is that we're not waiting to ensure that the DMA channel is disabled. The fix for this is in the DW DMA driver here in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/55891

On top of this, there are 2 more things we need to make sure:
1. https://github.com/thesofproject/sof/pull/7296: To ensure that suspend_drain is default for MTL
2. https://github.com/thesofproject/sof/pull/7295: To fix the ordering of DAI and DMA starts.

abonislawski, 2023-03-17:
@ranj063 great changes and findings! Thanks for debugging this

lgirdwood, 2023-03-17:
@ranj063 is the a PR for clearing the DONE bits ?

kv2019i, 2023-03-22:
@serhiy-katsyuba-intel @ranj063  We now pull in some fixes to dw-dma driver via https://github.com/thesofproject/sof/pull/7316 . This bug with dmic pause/release does still happen. 

One curious data point is that without "e40859f78712 Revert "dma: dw: Do not program SAR/DAR and CTL_HI/LO when using HW LLI"" the dmic pause/release test PASSES on MTL. So this hints the problem is somehow related to state of the the dmic FIFO dma-dw reads from.

ranj063, 2023-03-24:
@kv2019i @serhiy-katsyuba-intel this issue is now fixed with the zephyr PR 56209.

keqiaozhang, 2023-03-29:
We need Zephyr updates in sof main branch. reopening this bug.

keqiaozhang, 2023-04-03:
This bug is fixed by https://github.com/thesofproject/sof/commit/7bc3bde2036b5dc094e7e666a3590adeae45e5c9. Closing this bug.

closed 2023-04-03 by keqiaozhang