thesofproject/sof #6724: [BUG] failed to free pipeline widget pipeline.8 when simultaneous-playback-capture on MTLP_RVP_NOCODEC

miRoox, 2022-12-06:
**Describe the bug**
CI observed FW reported error: 12 for 0x12020000|0x0 (GLB_DELETE_PIPELINE) with "failed to free pipeline widget pipeline.8" when simultaneous-playback-capture on MTLP_RVP_NOCODEC.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-mtl-nocodec.tplg MODEL=MTLP_RVP_NOCODEC ~/sof-test/test-case/simultaneous-playback-capture.sh -l 25
```

**Reproduction Rate**

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/321028577de9
    * SOF: https://github.com/thesofproject/sof/commit/4a9a99aeb24a
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_NOCODEC 

**Screenshots or console output**
**console output**
```
2022-12-05 23:15:24 UTC [REMOTE_INFO] ===== Testing: (Loop: 4/25) =====
2022-12-05 23:15:24 UTC [REMOTE_COMMAND] aplay -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/zero -q &
2022-12-05 23:15:24 UTC [REMOTE_COMMAND] arecord -D hw:0,0 -c 2 -r 48000 -f S16_LE /dev/null -q &
2022-12-05 23:15:24 UTC [REMOTE_INFO] Preparing to sleep for 5
2022-12-05 23:15:29 UTC [REMOTE_INFO] check pipeline after 5s
2022-12-05 23:15:29 UTC [REMOTE_COMMAND] killing all pipelines
2022-12-05 23:15:29 UTC [REMOTE_COMMAND] aplay -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/zero -q &
2022-12-05 23:15:29 UTC [REMOTE_COMMAND] arecord -D hw:0,1 -c 2 -r 48000 -f S16_LE /dev/null -q &
2022-12-05 23:15:29 UTC [REMOTE_INFO] Preparing to sleep for 5
arecord: pcm_read:2221: read error: Input/output error
2022-12-05 23:15:34 UTC [REMOTE_INFO] check pipeline after 5s
/home/ubuntu/sof-test/test-case/simultaneous-playback-capture.sh: line 107: kill: (29202) - No such process
2022-12-05 23:15:34 UTC [REMOTE_ERROR] Error in arecord process after sleep.
/home/ubuntu/sof-test/test-case/simultaneous-playback-capture.sh: line 67: kill: (29202) - No such process
2022-12-05 23:15:34 UTC [REMOTE_ERROR] Starting func_exit_handler(), exit status=1, FUNCNAME stack:
2022-12-05 23:15:34 UTC [REMOTE_ERROR]  func_error_exit()  @  /home/ubuntu/sof-test/test-case/simultaneous-playback-capture.sh
2022-12-05 23:15:34 UTC [REMOTE_ERROR]  main()  @  /home/ubuntu/sof-test/test-case/simultaneous-playback-capture.sh:108
2022-12-05 23:15:39 UTC [REMOTE_INFO] pkill -TERM -f mtrace-reader.py
2022-12-05 23:15:39 UTC [REMOTE_INFO] ktime=5975 sof-test PID=28790: ending
2022-12-05 23:15:39 UTC [REMOTE_INFO] Test Result: FAIL!
```

**dmesg**
```
[ 5964.856930] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12020000|0x0: GLB_DELETE_PIPELINE
[ 5964.857340] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x3200000c|0x0: GLB_DELETE_PIPELINE
[ 5964.857344] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[ 5964.857392] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12020000|0x0
[ 5964.857416] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.8
[ 5964.857436] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules copier.module.7.2:0 -> gain.7.1:0
[ 5964.857441] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-mtl 0000:00:1f.3: widget copier.module.7.2 freed
[ 5964.857444] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-mtl 0000:00:1f.3: unbind modules gain.7.1:0 -> copier.host.7.1:0
[ 5964.857449] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-mtl 0000:00:1f.3: widget gain.7.1 freed
[ 5964.857454] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12030000|0x0: GLB_DELETE_PIPELINE
[ 5964.857679] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x3200000c|0x0: GLB_DELETE_PIPELINE
[ 5964.857683] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[ 5964.857768] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12030000|0x0
[ 5964.857788] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.7
[ 5964.857808] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
[ 5964.857831] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
```

**mtrace**
```
[00:01:38.916,050] <err> dai_intel_ssp: dai_ssp_poll_for_register_delay poll timeout reg 163848 mask 4 val 4 us 125
[00:01:38.916,996] <err> dai_intel_ssp: dai_ssp_poll_for_register_delay poll timeout reg 163912 mask 63 val 0 us 937
[00:01:38.917,001] <wrn> dai_intel_ssp: dai_ssp_empty_tx_fifo warning: timeout
[00:01:38.917,008] <inf> dai_intel_ssp: dai_ssp_update_bits: dai_ssp_update_bits base 28000, reg 30, mask 100, value 0
[00:01:38.917,020] <inf> dai_intel_ssp: dai_ssp_stop: dai_ssp_stopTX stop
[00:01:38.917,093] <err> dma_dw_common: xrun detected
[00:01:38.917,101] <err> dai_comp: comp:2 0x40002 dai_report_xrun(): overrun due to no space available
[00:01:38.917,110] <err> component: comp:2 0x40002 comp_overrun(): sink->free = 768, copy_bytes = 0
[00:01:38.917,118] <inf> pipe: pipeline_trigger: pipe:2 0x0 pipe trigger cmd 6
[00:01:38.917,133] <err> pipe: pipe:2 0x0 pipeline_copy(): ret = -61, start->comp.id = 262146, dir = 0
[00:01:38.917,140] <inf> pipe: pipeline_reset: pipe:2 0x0 pipe reset
[00:01:38.917,146] <inf> dai_comp: dai_reset: comp:2 0x40002 dai_reset()
[00:01:38.917,156] <inf> ipc: dai_dma_release: comp:2 0x40002 dai_config(): Component is in active state. Ignore resetting
[00:01:38.917,201] <err> component: comp:2 0x40002 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:38.917,211] <err> component: comp:2 0x40002 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[00:01:38.917,220] <err> pipe: pipe:2 0x0 pipeline_task(): xrun recovery failed! pipeline is stopped.
[00:01:38.917,226] <inf> ll_schedule: zephyr_ll_task_done: task complete 0xa00f31c0 0x20180U
[00:01:38.917,231] <inf> ll_schedule: zephyr_ll_task_done: num_tasks 2 total_num_tasks 2
[00:01:38.917,241] <inf> pipe: pipeline_reset: pipe:3 0x0 pipe reset
[00:01:38.917,251] <err> component: comp:3 0x40004 comp_set_state(): wrong state = 5, COMP_TRIGGER_RESET
[...]
[00:01:38.925,280] <inf> ipc: ipc_cmd: rx	: 0x12020000|0x0
[00:01:38.925,290] <inf> pipe: pipeline_disconnect: comp:2 0x40002 disconnect buffer 0 as sink
[00:01:38.925,318] <inf> dma: dma_put: dma_put(), dma = 0x400d3e38, sref = 0
[00:01:38.925,325] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power index 0
[00:01:38.925,460] <err> dai_intel_ssp: dai_ssp_poll_for_register_delay poll timeout reg 165892 mask 256 val 256 us 125
[00:01:38.925,465] <wrn> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power warning: timeout
[00:01:38.925,470] <inf> dai_intel_ssp: dai_ssp_pm_runtime_dis_ssp_power: dai_ssp_pm_runtime_dis_ssp_power I2SLCTL
[00:01:38.925,541] <inf> pipe: pipeline_disconnect: comp:2 0x60002 disconnect buffer 0 as sink
[00:01:38.925,551] <inf> pipe: pipeline_disconnect: comp:2 0x60002 disconnect buffer 0 as source
[00:01:38.925,578] <err> ipc: ipc_comp_free(): comp id: 393218 state is 5 cannot be freed
[00:01:38.925,585] <err> ipc: ipc_pipeline_free(): module free () failed
[00:01:38.925,591] <err> ipc: ipc4: FW_GEN_MSG failed with err 12
```

juimonen, 2022-12-09:
kind of looks like that we are stopping the dai, but the dma is still running after it -> xrun and error. Not sure should we use the CONFIG_DMA_SUSPEND_DRAIN to stop the dma first...

keqiaozhang, 2023-01-11:
This issue is not reproducible in CI. The last time I saw this issue was on December 21, 2022. Closing this bug.

closed 2023-01-11 by keqiaozhang