thesofproject/sof #6734: [BUG] FW reported error: 7 for 0x13000004|0x1 when multiple-pause-resume on MTLP_RVP_NOCODEC

miRoox, 2022-12-07:
**Describe the bug**
CI observed an IPC error "FW reported error: 7 - Unsupported operation requested" for 0x13000004|0x1 when testing multiple-pause-resume on MTLP_RVP_NOCODEC.

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-ace-tplg/sof-mtl-nocodec.tplg MODEL=MTLP_RVP_NOCODEC ~/sof-test/test-case/multiple-pause-resume.sh -r 50
```

**Reproduction Rate**
Lower than 30%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/c78e584e7704
    * SOF: https://github.com/thesofproject/sof/commit/4a9a99aeb24a
2) Name of the topology file
    * Topology: sof-ace-tplg/sof-mtl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_NOCODEC 

**Screenshots or console output**

**dmesg**
```
[ 1029.026338] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 1029.026868] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000007|0x1: GLB_SET_PIPELINE_STATE
[ 1029.026872] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 1029.027039] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13000004|0x1
[ 1029.027064] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 1029.027084] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 1029.027107] kernel:  Port2: ASoC: trigger FE cmd: 1 failed: -22
```

keqiaozhang, 2022-12-19:
The reproduction rate is almost 100% on main branch, change it to P1

miRoox, 2023-02-07:
> @miRoox can you share logs with ipc payload

```
[ 1021.306992] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 1021.306997] kernel: Message payload: 00000000: 00000002 00000001 00000000
[ 1021.307001] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 1021.307648] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000007|0x1: GLB_SET_PIPELINE_STATE
[ 1021.307652] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 1021.307694] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13000004|0x1
[ 1021.307718] kernel: Message payload: 00000000: 00000002 00000001 00000000
[ 1021.307721] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 1021.307742] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 1021.307765] kernel:  Port0: ASoC: trigger FE cmd: 1 failed: -22
```
[dmesg.txt](https://github.com/thesofproject/sof/files/10672318/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/10672317/mtrace.txt)

mszleszy, 2023-02-10:
working on reproduction using FAT framework

lgirdwood, 2023-02-13:
@mszleszy can you try with #7052 now being merged, it's improved and hopefully fixed the pause issue. @lyakh fyi.

mszleszy, 2023-02-16:
No repro found using FAT framework.
Logs starts with clean up of old topology and building the problematic one. Maybe it's necessary to buil&run both topologies to reproduce the error. 
Could we get log with both topologies?

miRoox, 2023-02-17:
> @miRoox any update?

We still meet this issue every day.

mszleszy, 2023-02-24:
> > @miRoox any update?
> 
> We still meet this issue every day.

@miRoox probably you missed my previous comment. Are You able to catch ipc logs of both topologies (topology run before the error and topology with the error)

keqiaozhang, 2023-02-27:
This issue has not appeared in CI recently, will continue to monitor this issue in CI for another 2 weeks.

marc-hb, 2023-03-02:
Note every pause-resume test on every platform (including of course MTL) has been "spamming" the firmware logs with _thousands_ of errors like this:

```
[    0.345803] <inf> dai_intel_ssp: dai_ssp_update_bits base 77000, reg 30, mask 100, value 100
[    0.345973] <err> copier: comp:0 0x40000 failed to find dai comp or sink pipeline not running.
[    0.346005] <err> copier: comp:0 0x40000 failed to find dai comp or sink pipeline not running.
[    0.403148] <inf> ipc: rx	: 0x13000003|0x1
```

Example in https://sof-ci.01.org/sofpr/PR7202/build4268/devicetest/index.html?model=ADLP_RVP_SDW_IPC4ZPH&testcase=multiple-pause-resume-5 + many, many others.

The test is PASSing anyway because sof-test has never cared about firmware errors.

This makes it practically impossible to find other errors.

Should this be a warning instead?

cc: @RanderWang, @softwarecki, @kv2019i , @lyakh ,  @ranj063

keqiaozhang, 2023-03-06:
This issue disappeared in CI for weeks, Closing it.

closed 2023-03-06 by keqiaozhang