thesofproject/sof #6741: [FEATURE][Zephyr] Get rid of fix_elf_addrs.py step

kv2019i, 2022-12-07:
**Is your feature request related to a problem? Please describe.**
Currently SOF Zephyr build requires an extra step between linker and running the rimage tool on Intel ADSP platforms.

The process is currently:

linker -> fix_elf_adds.py -> rimage -> bootable .ri image

The script is maintained in work/zephyr/soc/xtensa/intel_adsp/common/fix_elf_addrs.py

The reason for this script is that in old XTOS linker scripts, all sections are mapped to cached alises, but in Zephyr, linker scripts are already used to assign sections to uncached/cached aliases. rimage does not support this and will emit errors if given a raw Zephyr elf file. To fix this, fix_elf_addrs.py is run on the intermediate elf binary.

This step however has led to hard-to-debug errors when linker script changes are made. This is especially problematic when important generic linker scripts from Zephyr.

**Describe the solution you'd like**
Modify rimage to support the Zephyr linker script output directly.

**Describe alternatives you've considered**
- modify Zephyr linker script to map everything to cached -> does not work as Zephyr kernel code is assuming specific sections are accesible with coherent memory (and on these Intel platforms, this means a uncached alias)

kv2019i, 2022-12-07:
FYI @mmaka1 @marcinszkudlinski ... at least @mmaka1 wasn't listed in Zephyr project, so pinging you in this SOF side enhancement. Let's try to have the main discussion/commentary in the Zephyr ticket.

kv2019i, 2023-03-06:
@lyakh Still on track for V2.5 -- we aim to branch this week?

lyakh, 2023-03-07:
> @lyakh Still on track for V2.5 -- we aim to branch this week?

@kv2019i have we upgraded to use that rimage already? Then I can push a PR to Zephyr to remove the script?

marc-hb, 2023-03-07:
The rimage version is in sof/west.yml (and in git submodules)

kv2019i, 2023-03-13:
This didn't make the cut for v2.5, moving to v2.6.

lgirdwood, 2023-03-17:
> @lyakh rimage west update merged today.

@lyakh What's left to merge now? Can we reference the PRs here.

lyakh, 2023-03-20:
> > @lyakh rimage west update merged today.
> 
> @lyakh What's left to merge now? Can we reference the PRs here.

@lgirdwood we need https://github.com/zephyrproject-rtos/zephyr/pull/55554 merged and then just move SOF to use updated rimage and Zephyr, like https://github.com/thesofproject/sof/pull/7226 but with real git commits from respective main branches

lgirdwood, 2023-03-21:
> > > @lyakh rimage west update merged today.
> > 
> > 
> > @lyakh What's left to merge now? Can we reference the PRs here.
> 
> @lgirdwood we need [zephyrproject-rtos/zephyr#55554](https://github.com/zephyrproject-rtos/zephyr/pull/55554) merged and then just move SOF to use updated rimage and Zephyr, like #7226 but with real git commits from respective main branches

@lyakh Zephyr PR now merged.

lyakh, 2023-03-22:
> @lyakh Zephyr PR now merged.

@lgirdwood yes, but we cannot upgrade to that version of Zephyr yet because of another breakage, or has a fix been merged too by now? @kv2019i

lgirdwood, 2023-03-22:
@lyakh pls also checking with @nashif re breakage fixes.

lyakh, 2023-05-11:
> @lyakh can we close ?

@lgirdwood the SOF part is in the tree, but the Zephyr part has been reverted. As discussed yesterday, I have to resubmit it, will try to do today.

lyakh, 2023-05-12:
https://github.com/zephyrproject-rtos/zephyr/pull/57768 waiting for tests and merge

lyakh, 2023-05-15:
https://github.com/zephyrproject-rtos/zephyr/pull/57768 got merged. As long as it doesn't get reverted again, we can close this feature request. Or we can ask @nashif to confirm that the previously observed issues have now been resolved.