thesofproject/sof #6756: [BUG][mtl-002] alsa_conformance_test: test_rates failed for both accuracy and stability on I2S interfaces

yongzhi1, 2022-12-08:
**Describe the bug**
Run _alsa_conformance_test --test-suites test_rates_ to check whether all estimated rates are the same as what it set.
When --rate-criteria-diff-pct is set to 0.05 (default is 0.01), tests pass consistently with CAVS_FW_ACE_RELEASE but failed for cSOF.

More details about the test:
https://chromium.googlesource.com/chromiumos/platform/audiotest/+/refs/heads/master/alsa_conformance_test.md#Stability-of-rate

**To Reproduce**
```
alsa_conformance_test.py -P hw:0,0  --test-suites  test_rates  --rate-criteria-diff-pct 0.05 --merge-thld-size 48
```
**Reproduction Rate**
100%

**Expected behavior**
alsa_conformance_test.py is a "standard" test that run on all previous platforms. 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://chromium.googlesource.com/a/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.15 with https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/4047328
    * SOF: mtl-002-drop-stable and main https://github.com/thesofproject/sof/commit/0f1434504b952a2920ff054ff7b34d1b08e8bd98
2) Name of the topology file
    * Topology: sof-mtl-max98357a-rt5682, sof-mtl-nocodec.tplg,
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP

**Screenshots or console output**
```
localhost ~ # alsa_conformance_test.py -P hw:0,0  --test-suites  test_rates  --rate-criteria-diff-pct 0.05 --merge-thld-size 48 
0 passed, 1 failed
Device Information
	Name: hw:0,0
	Card: sofrt5682 [sof-rt5682]
	Device: Headset (*) []
	Stream: PLAYBACK
	Format: ['S16_LE', 'S24_LE', 'S32_LE']
	Channels: [2]
	Rate: [48000]
	Period_size range: [24, 4096]
	Buffer_size range: [48, 16384]
Test Rates
	Set rate 48000: fail - Expected rate is 48000, measure 43024.080435000003, difference 4975.9195649999965 > threshold 24.0
```
Note:

With --rate-criteria-diff-pct 0.05 , threshold = 48000 * 0.05/100, expected rate is 48K, measured 43K, off 5K.

lgirdwood, 2022-12-09:
@mwasko @marcinszkudlinski @kv2019i fyi - could we be using the wrong clock frequency for the timer here ?

yongzhi1, 2022-12-11:
Attached test outputs with debug option, we can see there is more irregularity shown in the DIFF column for cSOF:
[ssp-rate-output-cSOF.txt](https://github.com/thesofproject/sof/files/10203610/ssp-rate-output-cSOF.txt)
[ssp0-rate-output-ACE.txt](https://github.com/thesofproject/sof/files/10203611/ssp0-rate-output-ACE.txt)

mwasko, 2022-12-12:
@yongzhi1 can you provide information on what MTL-RVP SOC revision it was reproduced (A0, A3, other)?

yongzhi1, 2022-12-12:
Hi, @mwasko Thanks for your attention, this was observed on A3 with 'S16_LE', 'S24_LE', 'S32_LE', and seems the i2s master link clock selection fix for https://github.com/thesofproject/sof/issues/6727 also worked wonder here, now the test results are on-a par with ACE FW.

juimonen, 2022-12-13:
@yongzhi1 I've had this in my stash for a while: https://github.com/juimonen/sof/tree/ssp_aux_data , so is this what is needed?
If someone can elaborate on the "link clock selection", we can probably try it out (the proper way).

yongzhi1, 2022-12-14:
Hi, @juimonen , probably "link_config" in your ssp_aux_data could be used to pass the clock source selection:
https://github.com/thesofproject/zephyr/pull/66

juimonen, 2022-12-14:
@yongzhi1 ok thanks, I'll make some test/demo about this

yongzhi1, 2023-01-13:
https://github.com/thesofproject/sof/pull/6860 was merged, so close this issue.