thesofproject/sof #6764: [FEATURE][Zephyr] Add stack checks to component info

cujomalainey, 2022-12-09:
**Is your feature request related to a problem? Please describe.**
Currently some components require stacks larger than the default, if you forget to modify the stack you are in for a bad time trying to figure out what you messed up

**Describe the solution you'd like**
Zephyr provides us a better opportunity given its rich APIs. In theory we can actually check our current stack size to make sure we are compatible before we even run the component if the information is packaged with the component.

**Describe alternatives you've considered**
N/A, Stack is based on FW currently but core assignment is in the topology, makes it very hard to validate outside of runtime

**Additional context**
https://github.com/thesofproject/sof/pull/6746#pullrequestreview-1211882309

lgirdwood, 2022-12-12:
@marcinszkudlinski @juimonen I assume we can add stack sizes to rimage toml to build into the manifest ?

lgirdwood, 2022-12-13:
> What about `comp_driver_info`?

I think we have a similar use case for Windows and trying to align the flow around Zephyr/IPC. @mwasko and @marcinszkudlinski working on configuration flow now.

cujomalainey, 2022-12-13:
Understood :)

lgirdwood, 2022-12-14:
Pencil this in for v2.5 at end of Q1/23

kv2019i, 2023-03-06:
V2.5 branching this week and nobody assigned, so moving to v2.6.

lgirdwood, 2023-05-10:
Moved to v2.7 - Loadable modules in progress.

kv2019i, 2023-08-22:
FYI to @jxstelter and @pjdobrowolski working on loadable modules support.

kv2019i, 2023-08-23:
@cujomalainey @lgirdwood @alex-cri I propose we move to v2.8. There is incremental work done that helps to address issues like https://github.com/thesofproject/sof/pull/6746/commits/104ed3e906ac1105060a5bc5487752cde80e61ba , including:
- Zephyr tooling to analyze stack usage (CONFIG_STACK_SENTINEL and CONFIG_STACK_USAGE)
- the new Data Processing module support (https://github.com/thesofproject/sof/pull/8020 and friends) design allows to read stack information from module manifest, and create a dedicated thread with appropriate size (i..e. stack is dimensioned to the module). for now stack is fixed to TASK_DP_STACK_SIZE

But e.g. for the more common low-latency modules, the stack size continues to be a global build time setting and we do not have infra in SOF for modules to declare the stack sizes, and to discover the maximum at build time and use this to size the LL-thread size. I believe this is at the root of the original ask @cujomalainey , right. The loadable modules further complicate this (a loaded LL-module will have to run in the same thread, so we cannot re-dimension the stack size of existing LL-threads).

alex-cri, 2023-08-23:
@kv2019i , I agree with your takeaways. Will give a day for any further comments to come in then plan to move to v2.8.

kv2019i, 2023-11-24:
Bumping to v2.9. If anyone has fresh examples and possible experiences using the Zephyr stack debug tools (and any shortcomings), please add here.

kv2019i, 2024-06-05:
The Zephyr shell support added in v2.10 provides more tools to debug stack usage, but the original ask of providing infra to describe component stack needs, and checking this at build time, remains open.

@cujomalainey with the new (static) tools  in Zephyr, has this still been a problem? I'll clear the milestone now, we can discuss the priority and potentially bring back to v2.10. FYI @lgirdwood

cujomalainey, 2024-06-05:
@andyross who can test this out and see if zephyr catches the issues

kv2019i, 2024-09-06:
Please add feedback on whether the existing tools in Zephyr are sufficient, or should we keep this open and target for better tooling at the audio component/module level. I'll move this to v2.12 for now as no new activity is planned this month. The Zephyr thread analyzer was merged this week, along with SOF debug stream to get this out to host, so this is tool is now available as well.

kv2019i, 2024-12-13:
Feature cutoff for v2.12. No activity and further feedback in 2.10/2.11 cycle, so I'll clear the milestone for now, but keep this open.

jsarha, 2025-02-06:
@cujomalainey , is there anything more to do here beyond the SOF thread_info and debug_stream.py client? The too now produces this kind of output:

root@9049fa09b098:~# debug_stream.py  
CPU 2: Load: 100.0% 2 threads (seqno 0)
    0x40161a40           stack 19.2%    load 0.0%
    2 thread info        stack 37.6%    load 0.0%
CPU 0: Load: 100.0% 9 threads (seqno 0)
    idle 02              stack 8.2%     load 0.0%
    0x40161978           stack 2.0%     load 0.0%
    0x401618b0           stack 2.0%     load 0.0%
    1 thread info        stack 3.9%     load 0.0%
    0 thread info        stack 37.6%    load 0.0%
    edf_workq            stack 8.2%     load 0.0%
    sysworkq             stack 31.8%    load 0.0%
    logging              stack 16.9%    load 0.0%
    idle 00              stack 8.2%     load 0.0%

To get this output the SOF has to be built with debug_stream and thred_info options that are included in app/debug_stream_overlay.conf [1] overlay. And the python tools debug_stream.py [2] and cavstool.py [3] should be copied to the same directory on DUT and give execute rights. Run debug_stream.py as root and that's it.

[1] https://github.com/thesofproject/sof/blob/main/app/debug_stream_overlay.conf
[2] https://github.com/thesofproject/sof/blob/main/tools/debug_stream/debug_stream.py
[3] https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/intel/intel_adsp/tools/cavstool.py

cujomalainey, 2025-02-06:
@johnylin76 as I am no longer part of the SOF team