thesofproject/sof #6774: [BUG] audio device doesn't work on Xiaomibook air 13

hzb-ykpcx, 2022-12-12:
I installed Fedora 37 and here's my dmesg
```
[    9.465213] sof-audio-pci-intel-tgl 0000:00:1f.3: Direct firmware load for intel/sof-tplg/sof-adl-rt711-2ch.tplg failed with error -2
[    9.465222] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg request firmware intel/sof-tplg/sof-adl-rt711-2ch.tplg failed err: -2
[    9.465224] sof-audio-pci-intel-tgl 0000:00:1f.3: you may need to download the firmware from https://github.com/thesofproject/sof-bin/
[    9.465226] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -2
[    9.465228] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -2
[    9.465241] sof_sdw sof_sdw: ASoC: failed to instantiate card -2
[    9.465377] sof_sdw sof_sdw: snd_soc_register_card failed -2
[    9.465396] sof_sdw: probe of sof_sdw failed with error -2
[    9.524244] hid-sensor-hub 001F:8087:0AC2.0003: hid_field_extract() called with n (192) > 32! (kworker/2:0)
[   10.957933] rfkill: input handler disabled
[   11.764330] soundwire_intel soundwire_intel.link.0: Clock stop failed -110
[   11.764338] soundwire_intel soundwire_intel.link.0: cannot enable clock stop on suspend
```

I tried adding `options snd-intel-dspcfg dsp_driver=1` to /etc/modprobe/alas_base.conf and reboot, only speaker works but the microphone doesn't work (not even detected). After I did as the [archwiki](https://wiki.archlinux.org/title/PipeWire#Microphone_is_not_detected_by_PipeWire) says and reboot, I get both input device and output device detected both in the audio settings, but they neither work. I download the latest firmware and install, doesn't work either.

lgirdwood, 2022-12-12:
@hzb-ykpcx looks like you are missing FW and topology binaries. Not sure how to install from your distro but they can be cloned here https://github.com/thesofproject/sof-bin

hzb-ykpcx, 2022-12-12:
@lgirdwood I actually installed the latest v2.2.3 before, but got the same issue.

plbossart, 2022-12-12:
I am afraid this is the same issue as https://github.com/thesofproject/linux/issues/4088, likely a bad BIOS.

Can you enable dynamic debug by copying this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/10209451/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.conf, reboot and attach the full dmesg log. Also attach the result of `alsa-info`.

plbossart, 2022-12-12:
BTW this is very very odd

````
options snd-intel-dspcfg dsp_driver=1 to /etc/modprobe/alas_base.conf and reboot, only speaker works
````

This would mean you have an HDaudio codec and still using a SoundWire topology. That's not quite possible. Let's see with the full dmesg log and alsa-info what your platform is.

hzb-ykpcx, 2022-12-13:
Thx for your reply, here it is.
[alsa-info.txt](https://github.com/thesofproject/sof/files/10216513/alsa-info.txt)
[full_dmesg.txt](https://github.com/thesofproject/sof/files/10216514/full_dmesg.txt)

plbossart, 2022-12-13:
ok, your BIOS is borked, the platform is HDaudio but somehow the BIOS advertises a SoundWire device.

````
[    8.203168] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    8.203180] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    8.203183] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 2 not selected in firmware
[    8.203187] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 3 not selected in firmware
[    8.203191] soundwire_intel:sdw_intel_probe_controller: acpi PRP00001:00: Creating 4 SDW Link devices
[    8.203323] soundwire_bus:sdw_extract_slave_id: soundwire sdw-master-0: SDW Slave Addr: 20025d071100
[    8.203327] soundwire_bus:sdw_extract_slave_id: soundwire sdw-master-0: SDW Slave class_id 0x00, mfg_id 0x025d, part_id 0x0711, unique_id 0x0, version 0x2

[    8.203520] snd_sof_intel_hda:hda_codec_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: HDA codec #0 probed OK: response: 14f120d0
[    8.206480] snd_sof_intel_hda:request_codec_module: hdaudio ehdaudio0D0: loading codec module: hdaudio:v14F120D0r00100001a01

[    8.254351] snd_sof_intel_hda:hda_codec_probe: sof-audio-pci-intel-tgl 0000:00:1f.3: HDA codec #2 probed OK: response: 8086281c
[    8.255147] snd_sof_intel_hda:request_codec_module: hdaudio ehdaudio0D2: loading codec module: hdaudio:v8086281Cr00100000a01
````
we cannot have both HDaudio and SoundWire supported.

can you add this in /etc/modprobe.d/alsa-base.conf

````
options snd-intel-sdw-acpi sdw_link_mask=0x80
````

This should disable all SoundWire links.