thesofproject/sof #6789: [BUG] playback does not work on dell adlp device

btian1, 2022-12-13:
**Describe the bug**
Tried with latest kernel and SOF, playback is not happening.

**To Reproduce**
1. get latest sof dev kernel, dpkg in adlp device.
2. set fw to: /lib/firmware/intel/sof-ipc4/tgl/dbgkey/dsp_basefw.bin
3. set tplg path to: /lib/firmware/intel/avs-tplg/cavs-sdw.tplg
4. playback with aplay, for example: sudo aplay -Dhw:0,2 -i California_Dreaming.wav
5. error reported

**Reproduction Rate**
100%

**Expected behavior**
playback should work normally

**Impact**
block daily development

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 12.12 daily
    * SOF: 12.12 daily
2) Name of the topology file
    * Topology: cavs-sdw.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: adlp dell device

**Screenshots or console output**
from kernel:

[   11.200740] BUG: kernel NULL pointer dereference, address: 0000000000000000
[   11.200743] #PF: supervisor read access in kernel mode
[   11.200745] #PF: error_code(0x0000) - not-present page

From SOF:

[   11.185140] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x60000068|0x40036: MOD_INIT_INSTANCE
[   11.185142] sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 104 - Other failure of module instance initialization request
[   11.185154] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x40040004|0x40036
[   11.185158] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to create module copier.ALH.41.1

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

ujfalusi, 2022-12-13:
@btian1, can you share the full dmesg?

plbossart, 2022-12-13:
And test IPC3 first!

closed 2022-12-14 by btian1

btian1, 2022-12-14:
[dmesg.log](https://github.com/thesofproject/sof/files/10225725/dmesg.log)

ujfalusi, 2022-12-14:
@btian1, thank you!
This patch is going to fix the NULL pointer: https://github.com/thesofproject/linux/pull/4086/commits/9b2996d141e2bb29f23e3844eb65b2b921705cd0

btian1, 2022-12-15:
Thanks, Peter, hope this patch can make ADLP SDW device working, but for me, I already move to TGL UP board for further testing, will check in future.

ujfalusi, 2022-12-15:
@btian1, it should fix the NULL pointer crash, not to make things magically work ;)