thesofproject/sof #6793: [BUG] trigger multiple pipeline start failed for WoV tplg

libinyang, 2022-12-13:
**Describe the bug**
The WoV tplg has below connections:
    DMIC ------> KPB ------------------------------> host_copier
                          |                                                      ^
                          |                                                       |
                          |----> micsel ----> dummywov ----> virtual_widget
ppl0 contains DMIC, KPB
ppl1 contains micsel, dummywov
ppl2 contains host_copier

The driver triggers multiple pipelines with a single IPC with SOF_IPC4_GLB_PIPE_STATE_EXT_MULTI set in the ipc message extension field. The IPC message is: 
ipc tx      : 0x13000004|0x1
payload  : 0x00000003 0x00000002 0x00000001 0x00000000
This sequence doesn't work and the trigger to running state will fail.

However, if we trigger single pipeline in sequence like:
ipc tx      : 0x13000004|0x0
ipc tx      : 0x13010004|0x0
ipc tx      : 0x13020004|0x0
This sequence works.

Use a pipeline, which contains a KPB component, to record. For example, on MTL nocodec platform, run:
```
arecord -Dhw:0,11 -c2 -r16000 -fs32_le -M -N 2.wav -vvv
```

**Reproduction Rate**
100%

**Expected behavior**
Record works well.

**Impact**
The pipelines which contains kpb won't work

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/9b5c7765a81bbd778e2c7b7a32dbe681c00b51d8
    * SOF: https://github.com/thesofproject/sof/commit/0664718a5896abe1e50191218cb80abb7171ad2d + https://github.com/thesofproject/sof/pull/6782 +  https://github.com/thesofproject/sof/pull/6668/commits/8938c3198ff6ab95815d4a42843d864366f44d47
2) Name of the topology file
    * Topology:     sof-ace-tplg/sof-mtl-nocodec-wov.tplg (which can be generated with PR #6782 )

3) Name of the platform(s) on which the bug is observed.
    * Platform: MTLP_RVP_NOCODEC

**dmesg**
[dmesg.txt](https://github.com/thesofproject/sof/files/10219126/dmesg.txt)
[mtrace.txt](https://github.com/thesofproject/sof/files/10219128/mtrace.txt)

libinyang, 2022-12-13:
The mtrace error:
```
[00:01:07.155,301] <inf> kd_test: test_keyword_trigger: comp:1 0xd0000 test_keyword_trigger()
[00:01:07.155,315] <err> module_adapter: comp:1 0x90000 module_process() error 1: for comp 589824
[00:01:07.155,323] <err> module_adapter: comp:1 0x90000 module_adapter_copy() error 1: module processing failed
[00:01:07.155,331] <inf> kpb: kpb_trigger: comp:0 0xe0000 kpb_trigger()
[00:01:07.155,343] <inf> kpb: kpb_trigger: comp:0 0xe0000 kpb_trigger()
```
ps: 0x90000 module is selector module.

The dmesg error:
```
[  219.532633] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x12000000|0x0: GLB_DELETE_PIPELINE
[  219.532971] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x3200000c|0x0: GLB_DELETE_PIPELINE
[  219.532978] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[  219.533252] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12000000|0x0
[  219.533275] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.19
```
ps: The dmesg shows error at deleting pipeline, but the mtrace shows error before this point. mtrace shows error at trigger multiple pipelines to running state.

lgirdwood, 2022-12-14:
@fkwasowi any insights ?

kfrydryx, 2022-12-20:
Can this be reproduced with https://github.com/thesofproject/sof/pull/6827 merged?

libinyang, 2022-12-21:
https://github.com/thesofproject/sof/pull/6827 can fix this issue on my old commit. But I tested on the latest fw code, which has already merged https://github.com/thesofproject/sof/pull/6827, the trigger still failed...

libinyang, 2023-01-10:
The latest code doesn't work because the heap memory is not large enough. https://github.com/thesofproject/sof/pull/6925 can fix this issue.

mengdonglin, 2023-01-10:
Closing as it's no longer a trigger sequence issue. For memory fix, please track https://github.com/thesofproject/sof/pull/6925

closed 2023-01-10 by mengdonglin