thesofproject/sof #6813: [BUG] MOD_SET_DX times out on MTL platform

keqiaozhang, 2022-12-15:
**Describe the bug**
This issue happened on MTL when testing suspend/resume related cases. From the demsg, it failed with MOD_SET_DX timeout.
```
[  142.743284] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  142.743892] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[  142.743966] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[  142.744060] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x47000000|0x0: MOD_SET_DX [data size: 8]
[  143.247354] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x47000000|0x0
[  143.247443] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  143.247454] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  143.247552] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x47000000|0x0|0x0, target: 0x64000000|0x3060004c|0x0, ctl: 0x3
[  143.247634] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  143.247689] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  143.247744] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  143.247785] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[  143.247864] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  143.247932] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  143.247997] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  143.248043] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  143.248101] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: -110 failed to prepare DSP for device removal
```
Before the reproduction rate is about 50%, but after https://github.com/thesofproject/sof/commit/0664718a5896abe1e50191218cb80abb7171ad2d, this issue can be reproduced on MTL 100% at the first round of suspend/resume.

**To Reproduce**
rtcwake -m mem -s 5 

**Reproduction Rate**
100% after https://github.com/thesofproject/sof/commit/0664718a5896abe1e50191218cb80abb7171ad2d, before the repro rate is about 50%.

**Expected behavior**

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: tip sof-dev/eebcfac2c69e
    * SOF: main/99a65fc7d72f
2) Name of the platform(s) on which the bug is observed.
    * Platform: MTL

[dmesg.txt](https://github.com/thesofproject/sof/files/10233416/dmesg.txt)

kfrydryx, 2022-12-20:
DX functionalities on MTL are in progress, issues are expected at the moment. Let's keep it on hold and revisit a bit later.

keqiaozhang, 2022-12-22:
This issue also can be reproduced on TGL-IPC4:
https://sof-ci.sh.intel.com/#/result/planresultdetail/19395?model=TGLU_RVP_SDW_IPC4ZPH&testcase=check-suspend-resume-with-playback-5

miRoox, 2023-01-05:
The reproduction rate become pretty low, and now we usually see https://github.com/thesofproject/linux/issues/4077 instead.

keqiaozhang, 2023-01-09:
This issue is not reproducible in CI, closing this bug.

miRoox, 2023-01-10:
It happened again today, seems the reproduction rate is very low.

mengdonglin, 2023-01-11:
Deprioritize it to P2 due to the very low reproduce rate atm.

keqiaozhang, 2023-02-24:
The error changed from `MOD_SET_DX` to `MOD_SET_D0IX`. This error change seems to have occurred after recent upstream mergers.
```
[ 1260.565574] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[ 1260.566134] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x64000000|0x3060004c: MOD_LARGE_CONFIG_SET
[ 1260.566169] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x44000000|0x3060004c: MOD_LARGE_CONFIG_SET [data size: 76]
[ 1260.566266] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x48000000|0x12: MOD_SET_D0IX
[ 1261.072884] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x48000000|0x12
[ 1261.072894] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 1261.072897] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 1261.072906] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc8000000|0x12|0x0, target: 0x64000000|0x3060004c|0x0, ctl: 0x3
[ 1261.072910] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 1261.072912] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 1261.072913] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[ 1261.072915] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 1261.072922] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[ 1261.072924] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0xa, ROM debug error: 0x0
[ 1261.072927] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[ 1261.072928] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 1261.072934] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: PM_GATE ipc error -110
[ 1261.073020] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set requested target DSP state 0 substate 1
[ 1261.073032] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: setting dsp state 0 substate 1
[ 1261.073042] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to power down DSP during suspend -110
[ 1261.073052] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: PM: pci_pm_suspend(): snd_sof_suspend+0x0/0x20 [snd_sof] returns -110
[ 1261.073157] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: PM: dpm_run_callback(): pci_pm_suspend+0x0/0x160 returns -110
[ 1261.073191] kernel: sof-audio-pci-intel-mtl 0000:00:1f.3: PM: failed to suspend async: error -110
[ 1261.073448] kernel: PM: Some devices failed to suspend, or early wake event detected
```

kv2019i, 2023-03-06:
V2.5 branching this week and nobody assigned, so moving to v2.6.