thesofproject/sof #6818: [BUG][MTL]: Headset capture resuls in I/O error with mtl-002 firmware

RDharageswari, 2022-12-16:
localhost ~ # arecord -Dhw:0,0 hp_cap.wav -c2 -r48000 -fs24_le -d15 -vv
Recording WAVE 'hp_cap.wav' : Signed 24 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-rt5682' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S24_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 8192
  period_size  : 2048
  period_time  : 42666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2048
  period_event : 0
  start_threshold  : 1
##+                                                | 03%arecord: pcm_read:2161:                                                                              read error: Input/output error
localhost ~ # dmesg

[   70.770549] init: cros-camera main process ended, respawning
[   71.538604] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[   71.550366] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x13060003|0x0
[   71.559112] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_dai_trigger on SSP0 Pin: -22
[   71.569077]  Headset: ASoC: error at dpcm_be_dai_trigger on Headset: -22
[   71.576583]  Headset: ASoC: trigger FE cmd: 0 failed: -22
[   71.584724] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 12 - Required resource is in invalid state
[   71.596242] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x12060000|0x0
[   71.604736] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to free pipeline widget pipeline.6
[   71.614659] sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to free connected widgets
[   71.623148] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -22
[   74.151506] sof-audio-pci-intel-mtl 0000:00:1f.3: FW reported error: 4 - The system or resource is busy
[   74.162201] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc error for msg 0x47000000|0x0
[   74.170705] sof-audio-pci-intel-mtl 0000:00:1f.3: ctx_save IPC error during suspend: -16
[   75.967136] init: cros-camera main process (4677) terminated with status 237
[   75.975062] init: cros-camera main process ended, respawning

RDharageswari, 2022-12-16:
@abonislawski @mengdonglin @sathya-nujella @udaymb @yongzhi1

mengdonglin, 2022-12-16:
@RDharageswari Can this issue with RT5682 headset capture be reproduced with previous mtl-002 release (SOF: mtl-002-drop-stable | https://github.com/thesofproject/sof/commit/fd0824097a3e)?

RDharageswari, 2022-12-16:
Able to repro this issue with the no-codec topology also

yongzhi1, 2022-12-16:
The xun in [fw_error_rc15.txt](https://github.com/thesofproject/sof/files/10242724/fw_error_rc15.txt) looks similar to https://github.com/thesofproject/sof/issues/6724 and was observed for capture only as commented by https://github.com/thesofproject/sof/issues/6724#issuecomment-1344029264

RDharageswari, 2022-12-16:
Last fix shared by @abonislawski is working well without any errors.

keqiaozhang, 2022-12-19:
@RDharageswari can we close this issue?

udaymb, 2022-12-19:
> @RDharageswari can we close this issue?

@keqiaozhang - Is the fix part of the mtl-002 branch?

keqiaozhang, 2022-12-19:
> > @RDharageswari can we close this issue?
> 
> @keqiaozhang - Is the fix part of the mtl-002 branch?

As Dhara confirmed that the fix share by @abonislawski can fix the issue, so I wonder if the fix has landed on mtl-002 and we can close this bug.

udaymb, 2022-12-20:
@abonislawski shared this FW [sof-mtl-8cd208873234e9ed02d4caaea7ae224b.ri.txt](https://github.com/thesofproject/sof/files/10267536/sof-mtl-8cd208873234e9ed02d4caaea7ae224b.ri.txt)

topology on max98357a and rt5682 codec - [sof-mtl-max98357a-rt5682-9c077c288605e8c581c6d25ef131a8dc.tplg.txt](https://github.com/thesofproject/sof/files/10267546/sof-mtl-max98357a-rt5682-9c077c288605e8c581c6d25ef131a8dc.tplg.txt)

Below option is set
options snd_sof sof_debug=128

With this the speaker playback, headset playback and capture, dmic capture works fine

udaymb, 2022-12-27:
@abonislawski - There are 2 additional commits in branch https://github.com/thesofproject/sof/tree/mtl-002-drop-stable

Please confirm, other than this there are no other changes, and would there be a release based on these changes?

mszleszy, 2023-01-03:
@RDharageswari I've check logs attached, we have IPC request& extension part but the payload part is missing, and this part is crucial for repro attempt.

[image]

Is there any chance for ipc messages with their payloads included?

abonislawski, 2023-01-03:
> @abonislawski - There are 2 additional commits in branch https://github.com/thesofproject/sof/tree/mtl-002-drop-stable
> 
> Please confirm, other than this there are no other changes, and would there be a release based on these changes?

mtl-002 is ready to use, no more changes

RDharageswari, 2023-01-05:
with the mtl-002 firmware.
capture is still noisy with the below format S24_LE, same is clear with s32_le
localhost ~ # arecord -Dhw:0,0 hp_cap.wav -c2 -r48000 -fs24_le -d15 -vv.

RDharageswari, 2023-01-10:
tried the same with mtl-003 firmware. Still the noise issue is observed.
localhost ~ # arecord -Dhw:0,0 hp_cap.wav -c2 -r48000 -fs24_le -d15 -vv
Recording WAVE 'hp_cap.wav' : Signed 24 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-rt5682' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S24_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 8192
  period_size  : 2048
  period_time  : 42666
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2048
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8192
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
  appl_ptr     : 0
  hw_ptr       : 0
[logs.zip](https://github.com/thesofproject/sof/files/10386852/logs.zip)
please find the kernel, firmware trace logs and the captured file in the attachment.
Please let me know if you need more info

kfrydryx, 2023-01-31:
@RDharageswari Is this issue still relevant?

RDharageswari, 2023-02-01:
we see the I/O error with mtl-003 release as well. I can raise a separate bug to track that.

closed 2023-03-10 by wszypelt