thesofproject/sof #6837: [BUG] distortion during playback on MTL-NOCODEC

keqiaozhang, 2022-12-19:
**Describe the bug**
we observed the waveform distortion on MTL-NOCODEC, we have a similar issue [SOF#6727](https://github.com/thesofproject/sof/issues/6727) on mtl-stable-002, but it has been fixed on 002 branch, so let's use this bug to track the distortion issue on main branch.

[image]

**To Reproduce**
~/sof-test/test-case/check-alsabat.sh -p hw:sofnocodec,0 -c hw:sofnocodec,0 -C 2

**Reproduction Rate**
100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[c41b1735d3da](https://github.com/thesofproject/linux/commit/c41b1735d3da)
SOF Branch: main
SOF Commit: 
[e6f72dee8289](https://github.com/thesofproject/sof/commit/e6f72dee8289)
Platform: MTL-NOCODEC

[bat.wav.3Vbz0k.zip](https://github.com/thesofproject/sof/files/10256182/bat.wav.3Vbz0k.zip)

juimonen, 2023-01-31:
This is because of missing link clock setting in mtl/ace platforms in ssp driver.

mtl 003 branch has a hard coded quick fix for "link clock", but that is not in sof main.

"non hard coded" fix is coming from:
https://github.com/zephyrproject-rtos/zephyr/pull/53297

after this, there also needs to be a topology change for mtl to configure/set the link clock.

wszypelt, 2023-02-06:
Reproduction test under implementation

keqiaozhang, 2023-02-22:
This issue is still exists in CI.
[image]

lgirdwood, 2023-02-22:
ok, looks like we could still be missing a MTL branch patch in main @mwasko fyi.

juimonen, 2023-03-07:
So we seem to have 2 overlapping bugs here:
1) link clock setting in mtl (this should not cause choppiness, but will amplify it)
- there is a fix pending for it https://github.com/thesofproject/sof/pull/7060
2) even without anything link clock related we have the choppiness:
- https://github.com/thesofproject/sof/issues/7184
- something is fishy in the feedback stream

fredoh9, 2023-03-10:
I tested with mtl-0004. alsabat test failed on both branch but waveforms are totally different. @abonislawski Do we miss any patch on main branch?

[image]

[alsabat_wav_capture.zip](https://github.com/thesofproject/sof/files/10946225/alsabat_wav_capture.zip)

fredoh9, 2023-03-11:
You mean this one, https://github.com/thesofproject/sof/pull/7060 ? I will take a look and check with it also.

kv2019i, 2023-03-13:
MTL only, moving to v2.6 milestone.

keqiaozhang, 2023-03-20:
> You mean this one, #7060 ? I will take a look and check with it also.

I tried PR#7060 on MTL-NOCODEC, but it doesn't help.
@fredoh9 , please help to double confirm it.

kv2019i, 2023-03-21:
#7060 is expected to help with the issue. @juimonen tested this with ok results, but the resulting audio file still has gaps related to smart-amp pipeline. Test with #7060 and without smart-amp, has not been done.

plbossart, 2023-03-21:
We really need the ability to take the SmartAmp out of the nocodec topology.
Adding more features needs to come with the ability to come back to a less complex part.

juimonen, 2023-03-21:
yeah, tested today with #7060 in "jf mtl nocodec 3" device - and ssp0 (smart amp pipe) has these very rare gaps in the right channel (loopback feedback) and ssp1 is clean. so this needs to be divided into 2 issues 1. link clock setting, that #7060 fixes 2. smart amp gaps

keqiaozhang, 2023-03-23:
Confirmed with https://github.com/thesofproject/sof/pull/7060.
Waveform is distorted at the beginning of both channels. 
[image]

There're also a lot of glitches in right channel.
[image]
I will file a new bug after https://github.com/thesofproject/sof/pull/7060 merged.

lrudyX, 2023-03-24:
Close as original  issue has been resolved. @keqiaozhang please create new ticket and provide IPC logs with payloads for FW reproduction.

keqiaozhang, 2023-04-03:
https://github.com/thesofproject/sof/pull/7060 is still under review, reopening this bug.

keqiaozhang, 2023-04-03:
Verified this issue again with PR https://github.com/thesofproject/sof/pull/7060, the result is same as before.
At beginning:
[image]
Some glitches in the right channel:
[image]

ranj063, 2023-04-04:
These are the replacements for #7060 

1. To set the link clock based on bit clock https://github.com/alsa-project/alsa-utils/pull/198
2. To set the blob version: https://github.com/thesofproject/sof/pull/7363
3. To ignore the link clock_source for TGL: https://github.com/zephyrproject-rtos/zephyr/pull/56497

closed 2023-04-05 by ranj063