thesofproject/sof #6843: [BUG] Dell XPS 9320 cannot find register sound card

IchikaZou, 2022-12-19:
Distrobution: Gentoo linux
Kernel version: 6.0.13
Sof-firmware version: 2.2

There is a kernel bug before 6.0.13 that won't load sof-firmware properly, but after update to 6.0.13 the sonund card would not be register, dmesg always telling me the "sof_sdw sof_sdw: snd_soc_register_card failed -517"

here is the dsemg log: https://dpaste.com/DMLF4V3HZ

plbossart, 2022-12-20:
We've already told you @IchikaZou that the relevant fixes were picked by Linux-stable. If it's not in 6.0.13 it'll be in the next release.

IchikaZou, 2022-12-20:
> We've already told you @IchikaZou that the relevant fixes were picked by Linux-stable. If it's not in 6.0.13 it'll be in the next release. 

So the patch could also fix `sof_sdw sof_sdw: snd_soc_register_card failed -517` right?

plbossart, 2022-12-21:
please enable dynamic debug by copying this file 
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/10279926/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.txt and provide the dmesg log. The log will be very large and verbose so please make sure the console buffer is large enough.

IchikaZou, 2022-12-26:
> please enable dynamic debug by copying this file [sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/10279926/sof-dyndbg.conf.txt) as /etc/modprobe.d/sof-dyndbg.txt and provide the dmesg log. The log will be very large and verbose so please make sure the console buffer is large enough.

here is the dsemg log file: https://dpaste.com/75BT8YS8A

IchikaZou, 2022-12-31:
Is there any other thing I need to provide?

aiChaoSONG, 2023-01-03:
@IchikaZou  looks the debug is not enabled, can you try to copy this file
[sof-dyndbg.conf.txt](https://github.com/thesofproject/sof/files/10279926/sof-dyndbg.conf.txt) as ***/etc/modprobe.d/sof-dyndbg.conf*** and provide the dmesg log. copy it to a txt file seems not work.

plbossart, 2023-01-03:
yeah, my bad, the .conf extension is required.

IchikaZou, 2023-01-03:
> yeah, my bad, the .conf extension is required.

Distrobution: Gentoo linux
Kernel version: 6.0.16
Sof-firmware version: 2.2.2

dmesg log:
https://dpaste.com/53KBWNJ9N

plbossart, 2023-01-03:
@plbossart the logs show that there are a couple of -517 errors, which are just deferred prone issues. The -517 is not an error proper, it's just telling that a dependency is missing, and the deferred probe mechanism will retry when a new driver is probed later. The card is created once all drivers probed, so not sure what your issue is.

IchikaZou, 2023-01-03:
> @IchikaZou the logs show that there are a couple of -517 errors, which are just deferred prone issues. The -517 is not an error proper, it's just telling that a dependency is missing, and the deferred probe mechanism will retry when a new driver is probed later. The card is created once all drivers probed, so not sure what your issue is.

could it be related to my kernel doesn't have proper module enable?

plbossart, 2023-01-03:
> could it be related to my kernel doesn't have proper module enable?

not sure what this means. The point is that the card IS registered eventually. Can you please check with and without the dynamic debug that you see a card with 'aplay -l' ?

IchikaZou, 2023-01-03:
aplay -l with dynamic debug
https://dpaste.com/CSWQFGQSL

aplay -L with dynamic debug
https://dpaste.com/EKT9SMRJK

aplay -l without dynamic debug
https://dpaste.com/FBQF3E2G5

aplay -l without dynamic debug
https://dpaste.com/GJ8ZXJ5YB

IchikaZou, 2023-01-03:
I do a hardware check and it doesn't report any malfunction, so might be rule out hardware malfunction.

plbossart, 2023-01-03:
@IchikaZou there is a card reported in all cases, so the -517 error is not a problem. I am not sure what the issue is here, you have a functional system in the end, don't you?

IchikaZou, 2023-01-03:
> @IchikaZou there is a card reported in all cases, so the -517 error is not a problem. I am not sure what the issue is here, you have a functional system in the end, don't you?

No, I check alsamixer and only Intel alderlake-p hdmi is presented. I tried both YouTube and local sound file, both have no sound output.

[image]

plbossart, 2023-01-03:
that's a naming red-herring @aiChaoSONG I can see rt1316 references that are related to the speaker functionality. You probably have a alsa-ucm-conf issue that prevents the mixers from being initialized. From a driver perspective there is no problem here, and the title of the issue is just not right.

IchikaZou, 2023-01-03:
> that's a naming red-herring @aiChaoSONG I can see rt1316 references that are related to the speaker functionality. You probably have a alsa-ucm-conf issue that prevents the mixers from being [...]

Okey, is there a way to solve this problem? And what title should I change the into.

plbossart, 2023-01-03:
well, first you need to test at the alsa hw: level, e.g.

speaker-test -Dhw:0,2 -c2 -r48000 

and see if you hear anything. then use alsa-ucm to set the endpoint as 'Speakers' and retry.

plbossart, 2023-01-03:
So if things appear to work at the ALSA hw: device level and don't when you go though PulseAudio or PipeWire, that's a clear indication that the alsa-ucm-conf is not up to date or correctly installed by your distribution.

IchikaZou, 2023-01-03:
I only use apulse but I assume this still related to alsa-umc-conf is not update.

IchikaZou, 2023-01-04:
I updated alsa-lib and alsa-ucm-conf to the newest version, unfortunately problem still persist.

aiChaoSONG, 2023-01-04:
@IchikaZou could you please share your ucm files, they are here: `/usr/share/alsa/ucm2/sof-soundwire`, do you have rt1316-1.conf and rt1316-2.conf in this folder?

IchikaZou, 2023-01-04:
> @IchikaZou could you please share your ucm files, they are here: `/usr/share/alsa/ucm2/sof-soundwire`, do you have rt1316-1.conf and rt1316-2.conf in this folder?

rt1316-1.conf
https://dpaste.com/7RP2A5ZCF

rt1316-2.conf
https://dpaste.com/FZ9LFQBRA

alsa-info
https://dpaste.com/CTNPTD3EE

aiChaoSONG, 2023-01-04:
@IchikaZou alsa-info says you don't have sound server installed, is pulseaudio or pipewiere running?
```
!!Sound Servers on this system
!!----------------------------

No sound servers found.
```

plbossart, 2023-01-04:
@IchikaZou if you only use ALSA then you need to manually use alsa-ucm to set-up the desired endpoint.

alsaucm -c sof-soundwire set _verb HiFi set _enadev Speaker
alsaucm -c sof-soundwire set _verb HiFi set _enadev Headphone
alsaucm -c sof-soundwire set _verb HiFi set _enadev Mic
alsaucm -c sof-soundwire set _verb HiFi set _enadev Headset

I am typing this off the top of my head so you'll have to double check from UCM files.

IchikaZou, 2023-01-04:
> @IchikaZou if you only use ALSA then you need to manually use alsa-ucm to set-up the desired endpoint.
>
>
>
> alsaucm -c sof-soundwire set _verb HiFi set _enadev Speaker
>
> alsaucm -c sof-soundwire set [...]

YES!! The command worked but I still can't adjust Speaker volume in alsamixer since there is no volume bar.
[image]

Is it because I don't have a sound server? Or it might related to other problem?

plbossart, 2023-01-04:
yes, if you don't have a sound server then the volume needs to be adjusted by hand. You are in a very specific configuration that not very many people use, so it's not surprising a lot of configuration is needed. A lot of UCM magic happens in PulseAudio/PipeWire.

IchikaZou, 2023-01-04:
I see thanks for answer my question, I will set up pipeqire later.
when I run  `aplay -Dhw:0,2 Downloads/StarWars3.wav` and it output this:

Playing WAVE 'Downloads/StarWars3.wav' : Signed 16 bit Little Endian, Rate 22050 Hz, Mono
aplay: set_params:1396: Channels count non available

this also related to I don't have a sound-server right?

plbossart, 2023-01-04:
The firmware expects 2ch 48kHz. if you play the wrong frequency and the wrong number of channels the driver will throw an error. You need userspace code to resample/upmix to the required format, either with plughw: if you use plain vanilla alsa, or a sound server.

IchikaZou, 2023-01-04:
> The firmware expects 2ch 48kHz. if you play the wrong frequency and the wrong number of channels the driver will throw an error. You need userspace code to resample/upmix to the required format, either with plughw: if you use plain vanilla alsa, or a sound server.

Understood, hope pipewire would adjust these problem automatically.