thesofproject/sof #6864: [BUG] ipc timed out for 0x13000004|0x1 GLB_SET_PIPELINE_STATE on IPC4 NOCODEC and SDW platforms

keqiaozhang, 2022-12-21:
**Describe the bug**
We observed this issue in daily test, it seems that this issue only happens on IPC4 NOCODEC and SDW platforms, we didn't observe this issue on HDA platform. DSP dead after ipc timed out, so there're no useful error logs in mtrace. 
The reproduction rate is about 20%. but I'm not sure if it's a regression or not, will do more tests to confirm it.

**dmesg**
```
[ 5182.381164] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 4 state: 4
[ 5182.381751] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x1: GLB_SET_PIPELINE_STATE
[ 5182.381849] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13000003|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 5182.381895] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000004|0x1: GLB_SET_PIPELINE_STATE [data size: 12]
[ 5182.382154] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
[ 5182.382214] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 4
[ 5182.382238] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x2014001e successful
[ 5182.382244] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 4 state: 4
[ 5182.887226] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13000004|0x1
[ 5182.887243] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 5182.887248] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 5182.887283] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000180 rirb 00
[ 5182.887291] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 5182.887330] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93000004|0x1|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 5182.887338] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 5182.887343] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 5182.887347] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 5182.887352] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 5182.887372] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 5182.887423] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x1
[ 5182.887429] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 5182.887435] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 5182.887449] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 5182.887458] kernel:  Deepbuffer Port0: ASoC: trigger FE cmd: 4 failed: -110
```

**To Reproduce**
20%

**Reproduction Rate**
~/sof-test/test-case/multiple-pause-resume.sh -r 50

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[c41b1735d3da](https://github.com/thesofproject/linux/commit/c41b1735d3da)
SOF Branch: main
SOF Commit: 
[80adcdf36a1d](https://github.com/thesofproject/sof/commit/80adcdf36a1d)
Zephyr Commit: 
[zephyr-v3.2.0-2606-gc5b270e7b003](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.2.0-2606-gc5b270e7b003)

bardliao, 2022-12-21:
It usually failed when I run `~/sof-test/test-case/multiple-pause-resume.sh -r 50` twice.  I can reproduce it even with `-r 20`. It is hard to reproduce if I run `~/sof-test/test-case/multiple-pause-resume.sh -r 50` only once. And increate `-r` number will not increase the reproduce rate.

juimonen, 2022-12-21:
Can you see anything in mtrace if the test is run manually and it fails... I mean starting mtrace before the test?

bardliao, 2022-12-21:
@juimonen mtrace log and kernel log are attached.
[out.txt](https://github.com/thesofproject/sof/files/10276165/out.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/10276184/dmesg.txt)

keqiaozhang, 2023-02-01:
Observed a similar error on TGLU_UP_HDA_IPC4ZPH.
https://sof-ci.sh.intel.com/#/result/planresultdetail/20379?model=TGLU_UP_HDA_IPC4ZPH&testcase=multiple-pipeline-all-25
```
[ 3429.677084] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 8 state 4
[ 3429.677087] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13080004|0x0: GLB_SET_PIPELINE_STATE
[ 3430.183535] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13080004|0x0
[ 3430.183610] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 3430.183615] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 3430.183678] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000f81 rirb 00
[ 3430.183738] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 3430.183799] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x93080004|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[ 3430.183866] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 3430.183926] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 3430.183965] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 3430.183995] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
[ 3430.184044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0xffffffff: unknown ROM status value
[ 3430.184100] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff
[ 3430.184185] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[ 3430.184234] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set final state 4 for all pipelines
[ 3430.184289] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 3430.184305] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 10 state 3
[ 3430.184364] kernel:  Deepbuffer HDA Analog: ASoC: trigger FE cmd: 1 failed: -110
[ 3430.184370] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x130a0003|0x0: GLB_SET_PIPELINE_STATE
[ 3430.184403] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 31 dir 0 cmd 0
[ 3430.184410] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
[ 3430.687534] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x130a0003|0x0
[ 3430.687590] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 3430.687622] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause all pipelines
[ 3430.687660] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 3430.687678] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x1e0]=0x540000 successful
[ 3430.687707] kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -110
[ 3430.687710] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 2
[ 3430.687733] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 8 state 2
[ 3430.687739] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13080002|0x0: GLB_SET_PIPELINE_STATE
[ 3430.687741] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[ 3430.687748] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-tgl 0000:00:1f.3: trigger cmd: 0 state: 3
```

lyakh, 2023-02-02:
Enabled `CONFIG_ASSERT` which immediately began triggering assertions in `mixin_process()`. After I've converted those to error prints (I'm really unsure they should be fatal), the test ran for a while, then triggered this:
```
ASSERTION FAIL [(intptr_t)ptr <= (intptr_t)buffer->end_addr] @ WEST_TOPDIR/modules/audio/sof/zephyr/../src/include/sof/audio/audio_stream.h:287
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 63 (zephyr exception)
 **  PC 0xbe017c6a VADDR (nil)
 **  PS 0x60020
 **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
 **  A0 0xbe015520  SP 0xbe0792f0  A2 0x4  A3 (nil)
 **  A4 0x9e08b990  A5 0x60020  A6 0xbe079300  A7 0x9e090d40
 **  A8 0x1  A9 0x1 A10 0x1 A11 0x1
 ** A12 (nil) A13 0x9e08cb38 A14 0x8f A15 0xbe04e104
 ** LBEG 0xbe04ba34 LEND 0xbe04ba3e LCOUNT (nil)
 ** SAR 0x4

Backtrace:0xbe017c67:0xbe0792f0 0xbe01551d:0xbe079300 0xbe0225e8:0xbe079320 0xbe0228eb:0xbe079340 0xbe039df5:0xbe079370 0xbe022f69:0xbe079460 0xbe024146:0xbe079480 0xbe0236aa:0xbe0794a0 0xbe02412c:0xbe0794d0 0xbe0236aa:0xbe0794f0 0xbe02412c:0xbe079520 0xbe0242ec:0xbe079540 0xbe023e37:0xbe0795a0 0xbe01f885:0xbe0795e0 0xbe024c6a:0xbe079600 0xbe01411b:0xbe079630 

>>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
Current thread: 0x9e08cb38 (unknown)
Halting system
```
The back-trace points to `module_adapter_copy()` -> `comp_update_buffer_produce()` -> `audio_stream_produce()` -> `audio_stream_wrap()`. Investigating

lyakh, 2023-02-06:
Here's what I've found out so far:
- Having mixin and mixout in different pipelines, they're scheduled independently, namely in case of playback mixout is scheduled before mixin.
- When testing pause-release respective when pause is pressed first the mixout pipeline task is stopped, then the mixin pipeline is scheduled, it gets the next data chunk in the buffer and stopped too.
- Then release is pressed, mixout is scheduled, it checks that its connected mixin is paused and does nothing
- Then mixin is scheduled, it checks that mixout already has data from before pause, checks if it has more space in its buffer (interestingly, sometimes it does and sometimes it doesn't) and possibly adds more data - if there's space. But often the mixout buffer is 48-frames large and the mixin is also trying to send the next 48 frames and there's no free space there.
- Then mixout is scheduled, it sends 48 frames from before the pause out
- Then mixin is scheduled again, now it already has 96 frames waiting, but can only send 48 of them.
- This continues until the next pause, then the backlog of frames increases, etc.

What's the right way to solve this?

keqiaozhang, 2023-02-13:
This issue is similar with https://github.com/thesofproject/sof/issues/6734 on MTL.
Please someone help to check if it's a duplicate.

lyakh, 2023-02-14:
After merging #7052 yesterday's daily test failed a "multiple pause-resume" test again (test result ID 20907) and the CI guessed that it was the same bug again, but this time the kernel error is "FW reported error: 3 - Not enough space in the IPC reply buffer to complete the request" and mtrace contains "pipe: pipeline_new(): Out of Memory" so that looks like a different problem to me

kv2019i, 2023-02-15:
@lyakh I got now one DSP oops in this test case:
http://sof-ci.sh.intel.com/storage/sofci/generic_test/20966/ADLP_RVP_NOCODEC_IPC4ZPH/multiple-pause-resume-50/mtrace.txt
... and this had #7052 included. On kernel side, first error was:
```
[ 4771.889728] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 4772.391758] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13000003|0x0
```

mengdonglin, 2023-02-16:
Fixed by https://github.com/thesofproject/sof/pull/7099

lyakh, 2023-02-16:
> Fixed by #7099

@mengdonglin I'm very skeptical about that. That PR only addresses cases when a pipeline has a component *inside* with more than one sink or source buffer, whereas that was only merged on Jan. 31 and this bug was opened on Dec. 21 2022. It's possible of course that this bug had several reasons and other reasons were fixed before, but it's not quite certain. I also still can easily reproduce assertion failures in my local tests which I thought were related to this bug. Anyway, we can keep it closed for now, but keep an eye on them...

lyakh, 2023-02-16:
I was wrong, smart amp was only enabled yesterday in #6996, i.e. on the same day as #7099, so #7099 cannot be a fix for this bug. I expect this bug to be reopened very soon.

miRoox, 2023-02-17:
This issue still happened today.

lyakh, 2023-02-17:
> This issue still happened today.

@miRoox I retain my opinion, that #7099 isn't a fix for this but, OTOH the failure today isn't this bug either: this bug is an IPC timeout, likely caused by a firmware crash, whereas today's failure is an out of memory allocation failure in the firmware

keqiaozhang, 2023-02-20:
> OTOH the failure today isn't this bug either: this bug is an IPC timeout, likely caused by a firmware crash, whereas today's failure is an out of memory allocation failure in the firmware

@lyakh thanks for the confirmation. I have filed https://github.com/thesofproject/sof/issues/7123 to track this new IPC issue.

This bug can be closed now.

closed 2023-02-20 by keqiaozhang