thesofproject/sof #6873: [BUG] [SRC] Playback and record not working with 8000hz with SRC component complied with XCC

iuliana-prodan, 2022-12-21:
**Describe the bug**
When running  a playback or record with 8000Hz the following error is met:
```
root@imx8mpevk:~# aplay -Dhw:3,0 -f S32_LE -c 2 -r 8000 -d 3 -t raw /dev/urandom
Playing raw data '/dev/urandom' : Signed 32 bit Little Endian, Rate 8000 Hz, Stereo
aplay: pcm_write:2127: write error: Input/output error
[ 925.526319] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 925.535485] sof-audio-of-imx8m 3b6e8000.dsp: ----------{}[ DSP dump start ]{}----------
[ 925.543567] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[ 925.548960] sof-audio-of-imx8m 3b6e8000.dsp: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 925.556340] sof-audio-of-imx8m 3b6e8000.dsp: unexpected fault 0xffffffff trace 0xffffffff
[ 925.564555] sof-audio-of-imx8m 3b6e8000.dsp: ----------{}[ DSP dump end ]{}----------
[ 925.572443] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at soc_component_trigger on 3b6e8000.dsp: -110
[ 925.581983] Port0: ASoC: trigger FE cmd: 0 failed: -110
[ 926.102312] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x60030000 (msg/reply size: 12/12)
[ 926.111477] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[ 926.116869] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at snd_soc_pcm_component_hw_free on 3b6e8000.dsp: -110
root@imx8mpevk:~# [ 929.142271] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x40010000 (msg/reply size: 76/12)
[ 929.151441] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[ 929.156778] sof-audio-of-imx8m 3b6e8000.dsp: ctx_save IPC error: -110, proceeding with suspend
```

**To Reproduce**
Use a SRC topology and run:
`aplay -Dhw:1,0 -f S16_LE -c 2 -r 8000 -d 3 -t raw /dev/urandom`
or
`arecord -Dhw:1,0 -d 3 -f S32_LE -c 2 -r 8000 -t raw /dev/null`

**Reproduction Rate**
With latest SOF - main branch, with **XCC**, the error is met 100%.
This was reproduced on all i.MX platforms: i.MX8MP, i.MX8QM, i.MX8QXP.

With GCC, all rates are working correctly - no error met.
 
**Expected behavior**
No error should appear.

**Impact**
Not working as expected.
 In case of XCC, the following rates should work:

                   - For playback : 8000 11025 16000 22050 32000 44100 48000 64000 88200 96000 176400 192000 Hz.

                   - For capture: 8000 11025 16000 22050 32000 44100 48000 Hz.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux version 6.1.0
    * SOF: SOF/main -(tested with Firmware info: version 2:0:0-06647)
2) Name of the topology file
    * Topology: sof-imx8-src-wm8960.tlg or sof-imx8mp-src-wm8960.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: i.MX8MP, i.MX8QM, i.MX8QXP.

**Screenshots or console output**
```
root@imx8qmmek:~# aplay -Dhw:1,0 -f S16_LE -c 2 -r 8000 -d 3 -t raw /dev/urandom
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
[   63.155674] sof-audio-of-imx8 556e8000.dsp: snd_sof_dsp_panic: dsp_oops_offset 8398848 differs from panic offset 233492486
[   63.169643] sof-audio-of-imx8 556e8000.dsp: ------------[ DSP dump start ]------------
[   63.177569] sof-audio-of-imx8 556e8000.dsp: DSP panic!
[   63.182716] sof-audio-of-imx8 556e8000.dsp: fw_state: SOF_FW_BOOT_COMPLETE (6)
[   63.189962] sof-audio-of-imx8 556e8000.dsp: reason: runtime exception (0x6)
[   63.196932] sof-audio-of-imx8 556e8000.dsp: trace point: 0x0dead006
[   63.203207] sof-audio-of-imx8 556e8000.dsp: panic at Jd:536923654
[   63.209306] sof-audio-of-imx8 556e8000.dsp: error: DSP Firmware Oops
[   63.215663] sof-audio-of-imx8 556e8000.dsp: error: Exception Cause: LoadStoreAlignmentCause, Load or store to an unaligned address
[   63.227408] sof-audio-of-imx8 556e8000.dsp: EXCCAUSE 0x00000009 EXCVADDR 0x92c26064 PS       0x00060725 SAR     0x00000662
[   63.238467] sof-audio-of-imx8 556e8000.dsp: EPC1     0x9241cffb EPC2     0x9241cab6 EPC3     0x00000000 EPC4    0x00000000
[   63.249530] sof-audio-of-imx8 556e8000.dsp: EPC5     0x00000000 EPC6     0x0000806a EPC7     0x0000602f DEPC    0x00000000
[   63.260585] sof-audio-of-imx8 556e8000.dsp: EPS2     0x00060120 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   63.271646] sof-audio-of-imx8 556e8000.dsp: EPS6     0x0500243e EPS7     0x0000e104 INTENABL 0x220000a8 INTERRU 0x0000000c
[   63.282697] sof-audio-of-imx8 556e8000.dsp: stack dump from 0x92491710
[   63.289235] sof-audio-of-imx8 556e8000.dsp: 0x92491710: 00000688 000006d8 92491730 92490da0
[   63.297597] sof-audio-of-imx8 556e8000.dsp: 0x92491720: d244c8c8 92491760 92491790 00000009
[   63.305954] sof-audio-of-imx8 556e8000.dsp: 0x92491730: 92491710 00000000 00000000 00000000
[   63.314312] sof-audio-of-imx8 556e8000.dsp: 0x92491740: 92491760 92c25c00 92c25004 92c26068
[   63.322669] sof-audio-of-imx8 556e8000.dsp: 0x92491750: 5244d89b 92491790 00000000 00000002
[   63.331026] sof-audio-of-imx8 556e8000.dsp: 0x92491760: 9241cffb 00000000 00000000 00000000
[   63.339384] sof-audio-of-imx8 556e8000.dsp: 0x92491770: 00000024 00000001 92491ab0 92491ab4
[   63.347741] sof-audio-of-imx8 556e8000.dsp: 0x92491780: 9244dc81 92491990 00000000 d244d89b
[   63.356098] sof-audio-of-imx8 556e8000.dsp: ------------[ DSP dump end ]------------
aplay: pcm_write:2127: [   65.158396] sof-audio-of-imx8 556e8000.dsp: ipc3_tx_msg_unlocked: ipc message send for 0x60050000 failed: -19
write error: Input/output error
[   65.170237] sof-audio-of-imx8 556e8000.dsp: ASoC: error at soc_component_trigger on 556e8000.dsp: -19
[   65.182303]  Port0: ASoC: trigger FE cmd: 0 failed: -19
[   65.187646] sof-audio-of-imx8 556e8000.dsp: ipc3_tx_msg_unlocked: ipc message send for 0x60030000 failed: -19
[   65.197603] sof-audio-of-imx8 556e8000.dsp: ASoC: error at snd_soc_pcm_component_hw_free on 556e8000.dsp: -19
root@imx8qmmek:~#
```

iuliana-prodan, 2022-12-21:
I did git-bisect and found the commit that introduce this bug: [16fbd5f08](https://github.com/thesofproject/sof/commit/16fbd5f0810d66afe1de41eb74fafdca1c0940d0) ("Audio: SRC: Increase stage buffer size and change copy() run condition")
The problem is the increased size of the stage buffer.

@singalsu can you, please, help me fix this?
Or should I revert the above patch?

iuliana-prodan, 2022-12-21:
Is this reproducible on Intel's platforms?

singalsu, 2022-12-21:
> Is this reproducible on Intel's platforms?

Thanks for reporting, I'll check if it happens. The commit fixed a pipeline freeze, so revering it is not straightforward.

lgirdwood, 2022-12-21:
@iuliana-prodan what was the SRC in/out rate combination with 8kHz that failed ?

iuliana-prodan, 2022-12-21:
> @iuliana-prodan I'm right now listening music with up-squared board with 8000 -> 48000 Hz rate (APL, HiFi3). No crash so far. Is the platform with issue using HiFi4?

Yes, is HiFi4.

iuliana-prodan, 2022-12-21:
> 

> @iuliana-prodan what was the SRC in/out rate combination with 8kHz that failed ?

Is 8000hZ to 48000Hz

singalsu, 2022-12-21:
Also, are you able to point the location of crash from addresses in stack dump, find the same addresses from build disassembly .lst. And could it be a data-align exception?

iuliana-prodan, 2022-12-21:
> > > @iuliana-prodan I'm right now listening music with up-squared board with 8000 -> 48000 Hz rate (APL, HiFi3). No crash so far. Is the platform with issue using HiFi4?
> > 
> > 
> > Yes, is HiFi4.
> 
> Are you able to try with a HiFi3 platform?
All i.MX platforms with DSP have a HiFi4. 
So, I don't have any with HiFi3.

> Or does the HiFi3 version source build for your HiFi4 configuration?
No, is using src_hifi4 source

singalsu, 2022-12-21:
> > Or does the HiFi3 version source build for your HiFi4 configuration?
> 
> No, is using src_hifi4 source

You could try to edit src_config.h after #if XCHAL_HAVE_HIFI4 == 1 to select HIFI3 instead of HIFI4 code version. In our case apparently we've lost source compatibility but it depends on used xtensa configuration.

lgirdwood, 2022-12-22:
@andrula-song possible something that can be checked/fixed on MTL

singalsu, 2022-12-29:
@iuliana-prodan Were you able to decode the stack dump vs. disassembly listing file from your build? It would reveal where the crash happened.

iuliana-prodan, 2023-01-11:
> @iuliana-prodan Were you able to decode the stack dump vs. disassembly listing file from your build? It would reveal where the crash happened.

Not yet :(
I was in vacation.
I tried again today, with latest SOF (main branch) and I couldn't get the `error: Exception Cause: LoadStoreAlignmentCause, Load or store to an unaligned address`, just

```
root@imx8mpevk:~# aplay -Dhw:3,0 -f S32_LE -c 2 -r 8000 -d 3 -t raw /dev/urandom
Playing raw data '/dev/urandom' : Signed 32 bit Little Endian, Rate 8000 Hz, Stereo
aplay: pcm_write:2127: write error: Input/output error
[  160.775140] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[  160.784363] sof-audio-of-imx8m 3b6e8000.dsp: preventing DSP entering D3 state to preserve context
[  160.793266] sof-audio-of-imx8m 3b6e8000.dsp: ------------[ DSP dump start ]------------
[  160.801319] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[  160.806674] sof-audio-of-imx8m 3b6e8000.dsp: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  160.814049] sof-audio-of-imx8m 3b6e8000.dsp: unexpected fault 0xffffffff trace 0xffffffff
[  160.822325] sof-audio-of-imx8m 3b6e8000.dsp: ------------[ DSP dump end ]------------
[  160.830183] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at soc_component_trigger on 3b6e8000.dsp: -110
[  160.839600]  Port0: ASoC: trigger FE cmd: 0 failed: -110
[  161.351097] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x60030000 (msg/reply size: 12/12)
[  161.360269] sof-audio-of-imx8m 3b6e8000.dsp: preventing DSP entering D3 state to preserve context
[  161.369165] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[  161.374527] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at snd_soc_pcm_component_hw_free on 3b6e8000.dsp: -110
root@imx8mpevk:~# 
```

I'll keep looking!

iuliana-prodan, 2023-01-16:
I've decoded the stack dump vs disassembly and here are my findings.

This is the stack dump on 8QM:
```
root@imx8qmmek:~# aplay -Dhw:1,0 -f S16_LE -c 2 -r 8000 -d 3 -t raw /dev/urandom
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
[ 4622.419530] sof-audio-of-imx8 556e8000.dsp: snd_sof_dsp_panic: dsp_oops_offset 8398848 differs from panic offset 233492486
[ 4622.433395] sof-audio-of-imx8 556e8000.dsp: ------------[ DSP dump start ]------------
[ 4622.441318] sof-audio-of-imx8 556e8000.dsp: DSP panic!
[ 4622.446461] sof-audio-of-imx8 556e8000.dsp: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 4622.453704] sof-audio-of-imx8 556e8000.dsp: reason: runtime exception (0x6)
[ 4622.460673] sof-audio-of-imx8 556e8000.dsp: trace point: 0x0dead006
[ 4622.466947] sof-audio-of-imx8 556e8000.dsp: panic at ":0
[ 4622.472265] sof-audio-of-imx8 556e8000.dsp: error: DSP Firmware Oops
[ 4622.478624] sof-audio-of-imx8 556e8000.dsp: error: Exception Cause: LoadStoreAlignmentCause, Load or store to an unaligned address
[ 4622.490367] sof-audio-of-imx8 556e8000.dsp: EXCCAUSE 0x00000009 EXCVADDR 0x92c27164 PS       0x00060725 SAR     0x00000200
[ 4622.501417] sof-audio-of-imx8 556e8000.dsp: EPC1     0x9241cefb EPC2     0x9241c9b6 EPC3     0x00000000 EPC4    0x00000000
[ 4622.512469] sof-audio-of-imx8 556e8000.dsp: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[ 4622.523518] sof-audio-of-imx8 556e8000.dsp: EPS2     0x00060120 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[ 4622.534570] sof-audio-of-imx8 556e8000.dsp: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000008
[ 4622.545621] sof-audio-of-imx8 556e8000.dsp: stack dump from 0x9248e270
[ 4622.552159] sof-audio-of-imx8 556e8000.dsp: 0x9248e270: 00000688 000006d8 9248e290 9248d900
[ 4622.560514] sof-audio-of-imx8 556e8000.dsp: 0x9248e280: d2449428 9248e2c0 9248e2f0 00000009
[ 4622.568869] sof-audio-of-imx8 556e8000.dsp: 0x9248e290: 9248e270 00000000 00000000 00000000
[ 4622.577227] sof-audio-of-imx8 556e8000.dsp: 0x9248e2a0: 9248e2c0 92c26d00 92c26104 92c27168
[ 4622.585584] sof-audio-of-imx8 556e8000.dsp: 0x9248e2b0: 5244a3db 9248e2f0 00000000 00000002
[ 4622.593941] sof-audio-of-imx8 556e8000.dsp: 0x9248e2c0: 9241cefb 00000000 00000000 00000000
[ 4622.602299] sof-audio-of-imx8 556e8000.dsp: 0x9248e2d0: 00000024 00000002 9248e610 9248e614
[ 4622.610656] sof-audio-of-imx8 556e8000.dsp: 0x9248e2e0: 9244a7c1 9248e4f0 00000000 d244a3db
[ 4622.619014] sof-audio-of-imx8 556e8000.dsp: ------------[ DSP dump end ]------------
aplay: pcm_write:2127: [ 4624.444981] sof-audio-of-imx8 556e8000.dsp: ipc3_tx_msg_unlocked: ipc message send for 0x60050000 failed: -19
write error: Input/output error
[ 4624.456868] sof-audio-of-imx8 556e8000.dsp: ASoC: error at soc_component_trigger on 556e8000.dsp: -19
[ 4624.468949]  Port0: ASoC: trigger FE cmd: 0 failed: -19
[ 4624.474253] sof-audio-of-imx8 556e8000.dsp: ipc3_tx_msg_unlocked: ipc message send for 0x60030000 failed: -19
[ 4624.484238] sof-audio-of-imx8 556e8000.dsp: ASoC: error at snd_soc_pcm_component_hw_free on 556e8000.dsp: -19
root@imx8qmmek:~#
```

From the above stack I only found the following addresses in `.lst` file (changed extension in order to attach it here
[sof-imx8.txt](https://github.com/thesofproject/sof/files/10425162/sof-imx8.txt) 

**EXCVADDR 0x92c27164** - this address is in `buffer_heap` , based on Memory Configuration from `.map` file:
```
Memory Configuration

Name             Origin             Length             Attributes
vector_reset_text 0x596f8000         0x000002e0
vector_reset_lit 0x596f82e0         0x00000120
vector_base_text 0x596f8400         0x00000178
vector_int2_lit  0x596f8578         0x00000004
...
system_heap      0x92c05000         0x0000e000
system_runtime_heap 0x92c13000         0x00006000
runtime_heap     0x92c19000         0x00007900
buffer_heap      0x92c20900         0x007de700  =>>> end = 0x933ff000
sof_stack        0x933ff000         0x00001000
sof_sdram1       0x92c00000         0x00800000
...
*default*        0x00000000         0xffffffff
```
**EPC1     0x9241cefb**
```
/**
 * \brief Called in the case of exception.
 */
static void exception(void)
{
9241cef8:	006136                      	entry	a1, 48
	uintptr_t epc1;
[... 28 lines cut ...]
9244a7aa:	e1e8                          	l32i.n	a14, a1, 56
9244a7ac:	fd160a0771de              	{ l32i	a13, a1, 28; mov.n	a10, a7 }
9244a7b2:	fd160b0281fe              	{ l32i	a15, a1, 32; mov.n	a11, a2 }
9244a7b8:	fd160c048eee              	{ l32i	a14, a14, 32; mov.n	a12, a4 }
9244a7be:	ffb525                      	call8	9244a310 <fir_filter>
			wp += nch_x_odm;
			cp = (uint8_t *)cp + subfilter_size;
9244a7c1:	d81609b1913e              	{ l32i	a3, a1, 36; l32i	a9, a1, 44 }
	struct src_stage *stage;
};
...
```
So, this `9244a7aa ` seems to be [here](https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi4.c#L514).
I've tried also with `src_hifi3.c`, but I get the same error, ~ at the same spot - [here](https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi3.c#L514)

Since I couldn't find the exact crash line, I did some code instrumentation and the fail seems to happen at the first iteration from the above [for loop](https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi4.c#L511), in the [fir_filter() function](https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi4.c#L171).
Next, in `fir_filter()` function, it stops at the first iteration at line https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi4.c#L221
```
			/* Load two data samples from two channels */
			AE_L32X2F24_XC(d0, dp, inc); /* r0, l0 */
9244a3db:	c4e244                      	ae_l32x2f24.xc	aed14, a2, a4
```
@singalsu do you have any hints?

Thanks!

iuliana-prodan, 2023-01-16:
I've also realized that on 8MP, which has a smaller heap_buffer I don't get the` Exception Cause: LoadStoreAlignmentCause, Load or store to an unaligned address`:

```
root@imx8mpevk:~# aplay -Dhw:3,0 -f S16_LE -c 2 -r 8000 -d 3 -t raw /dev/urandom
Playing raw data '/dev/urandom' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
aplay: pcm_write:2127: write error: Input/output error
[ 4391.154543] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 4391.163705] sof-audio-of-imx8m 3b6e8000.dsp: preventing DSP entering D3 state to preserve context
[ 4391.172662] sof-audio-of-imx8m 3b6e8000.dsp: ------------[ DSP dump start ]------------
[ 4391.180715] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[ 4391.186066] sof-audio-of-imx8m 3b6e8000.dsp: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 4391.193422] sof-audio-of-imx8m 3b6e8000.dsp: unexpected fault 0xffffffff trace 0xffffffff
[ 4391.201653] sof-audio-of-imx8m 3b6e8000.dsp: ------------[ DSP dump end ]------------
[ 4391.209592] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at soc_component_trigger on 3b6e8000.dsp: -110
[ 4391.219041]  Port0: ASoC: trigger FE cmd: 0 failed: -110
[ 4391.733888] sof-audio-of-imx8m 3b6e8000.dsp: ipc tx timed out for 0x60030000 (msg/reply size: 12/12)
[ 4391.743050] sof-audio-of-imx8m 3b6e8000.dsp: preventing DSP entering D3 state to preserve context
[ 4391.751934] sof-audio-of-imx8m 3b6e8000.dsp: IPC timeout
[ 4391.757273] sof-audio-of-imx8m 3b6e8000.dsp: ASoC: error at snd_soc_pcm_component_hw_free on 3b6e8000.dsp: -110
root@imx8mpevk:~# 
```

```
Memory Configuration

Name             Origin             Length             Attributes
vector_reset_text 0x3b6f8000         0x000002e0
vector_reset_lit 0x3b6f82e0         0x00000120
vector_base_text 0x3b6f8400         0x00000178
vector_int2_lit  0x3b6f8578         0x00000004
vector_int2_text 0x3b6f857c         0x0000001c
...
runtime_heap     0x92c19000         0x0000b900
buffer_heap      0x92c24900         0x007da700
sof_stack        0x933ff000         0x00001000
sof_sdram1       0x92c00000         0x00800000
...
```

iuliana-prodan, 2023-01-16:
Also, increasing the buffer between stage 1 and stage 2 by half, _all_ rates are working on _all_ i.MX boards:
```
---
 src/audio/src/src.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/audio/src/src.c b/src/audio/src/src.c
index 168e2de02..ddadb99d2 100644
--- a/src/audio/src/src.c
+++ b/src/audio/src/src.c
@@ -206,7 +206,7 @@ static int src_buffer_lengths(struct comp_dev *dev, struct comp_data *cd,
 		 * there is no equation known for minimum size.
 		 */
 		a->sbuf_length = 2 * nch * stage1->blk_out * r1;
-		a->sbuf_length += a->sbuf_length >> 3;
+		a->sbuf_length += a->sbuf_length >> 1;
 	}
 
 	a->src_multich = a->fir_s1 + a->fir_s2 + a->out_s1 + a->out_s2;
-- 
```
But, since I don't understand all the mathematics behind SRC, I don't know if it's correct.

lgirdwood, 2023-01-18:
@andrula-song @singalsu any comments here, buffer overflow ?

dbaluta, 2023-01-20:
As you said above, the crash likely to happen at:

https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi4.c#L221
```
			AE_L32X2F24_XC(d0, dp, inc); /* r0, l0 */
```

Running with some debug enabled found out that dp is `92C26064`. And this will cause the exception because the RM says that AE_L32X2F24_XC need to be 8-byte aligned.

dbaluta, 2023-01-25:
@singalsu I figured out the root cause here:

https://github.com/thesofproject/sof/blob/main/src/audio/src/src_hifi4.c#L221
```
AE_L32X2F24_XC(d0, dp, inc); /* r0, l0 */
```

expects `dp` to be 8-byte aligned but on i.MX we get into a scenario where for 8000Hz is not aligned.

Any idea how to force this alignment? The code that sets `dp` value is pretty complicated and I couldn't understand it.

lgirdwood, 2023-01-27:
@andrula-song any comment ?

dbaluta, 2023-01-27:
@andrula-song @singalsu proposed solution for this: https://github.com/thesofproject/sof/pull/6995

andrula-song, 2023-01-30:
what about using align load/store(AE_LA32X2F24_IC) to fix this bug?

dbaluta, 2023-01-30:
@andrula-song let me try this, I'm not familiar with Xtensa instruction set.

dbaluta, 2023-01-30:
@singalsu any idea how to use `AE_LA32X2F24_IC`. 

With current instruction we use ` AE_L32X2F24_XC(d0, dp, inc); ` but I'm don't know how to use  the increment with `AE_LA32X2F24_IC`.

Care to have a look? Otherwise, for the moment we can go with the fix in https://github.com/thesofproject/sof/pull/6995

singalsu, 2023-01-30:
I'd prefer to keep it simple and go with #6995, there's extra effort it with align register. The memory need for 8 bytes size align is minimal. Likely more RAM is needed for the extra code for aligned load/store instruction.

singalsu, 2023-02-10:
#6995 is now merged. Is it OK to close this bug @iuliana-prodan ?

dbaluta, 2023-02-10:
Merged. Fixed with https://github.com/thesofproject/sof/pull/6995

closed 2023-02-10 by dbaluta