thesofproject/sof #6874: [BUG] can't build sof firmware with docker

LongWang100, 2022-12-21:
**Describe the bug**
When I run the command "./scripts/docker-run.sh ./scripts/xtensa-build-all.sh tgl" to build firmware with docker, I got this error.
error: cannot parse firmware version major.minor.micro

**To Reproduce**
./scripts/docker-run.sh ./scripts/xtensa-build-all.sh tgl

**Reproduction Rate**
100%

**Expected behavior**
build successfully with docker.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF:  adl-004-drop-stable 
    * Commit hash: 
             commit f80e58ea11f88a2397a7bc5cc914dfd9e30be063 (HEAD, origin/adl-004-drop-stable, adl-004-drop-stable)

build log:

long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$
long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$
long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$ ./scripts/docker-run.sh ./scripts/xtensa-build-all.sh tgl
REPOSITORY   TAG       DIGEST    IMAGE ID       CREATED       SIZE
sof          latest    <none>    bc0abdcc45e0   5 weeks ago   3.5GB
REPOSITORY          TAG       DIGEST                                                                    IMAGE ID       CREATED       SIZE
thesofproject/sof   latest    sha256:bc16321c7d7cdf229f5625f30e96d0bbfce10afa45b9f1b79614d91b6b4d3c7c   bc0abdcc45e0   5 weeks ago   3.5GB
id: cannot print only names or real IDs in default format
Warning: this script should be run as user ID 1001 to match the container
+ id -u
+ docker run -i -v /home/long/external/long/work/sof/sof:/home/sof/work/sof.git -v /home/long/external/long/work/sof/sof:/home/sof/work/sof-bind-mount-DO-NOT-DELETE --env CMAKE_BUILD_TYPE --env PRIVATE_KEY_OPTION --env http_proxy= --env https_proxy= --user 1003 --tty sof ./scripts/xtensa-build-all.sh tgl
Build in build_tgl_gcc
PATH=/home/sof/work/sof.git/../xtensa-cnl-elf/bin:/home/sof/work/xtensa-mt8195-elf/bin:/home/sof/work/xtensa-mt8186-elf/bin:/home/sof/work/xtensa-imx8ulp-elf/bin:/home/sof/work/xtensa-imx8m-elf/bin:/home/sof/work/xtensa-imx-elf/bin:/home/sof/work/xtensa-rmb-elf/bin:/home/sof/work/xtensa-rn-elf/bin:/home/sof/work/xtensa-cnl-elf/bin:/home/sof/work/xtensa-apl-elf/bin:/home/sof/work/xtensa-hsw-elf/bin:/home/sof/work/xtensa-byt-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ cmake -DTOOLCHAIN=xtensa-cnl-elf -DROOT_DIR=/home/sof/work/sof.git/../xtensa-root/xtensa-cnl-elf -DMEU_OPENSSL= '' -DRIMAGE_PRIVATE_KEY=/home/sof/work/sof.git/keys/otc_private_key_3k.pem -DINIT_CONFIG=tgplp_defconfig -DEXTRA_CFLAGS= /home/sof/work/sof.git
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter
-- Preparing Xtensa toolchain
-- The C compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sof/work/xtensa-cnl-elf/bin/xtensa-cnl-elf-gcc
-- Found Git: /usr/bin/git (found version "2.25.1")
CMake Warning at scripts/cmake/git-submodules.cmake:31 (message):
  There are submodule changes, check git status and git diff

  +65f345a52e06a084192124364c563d8133d834c2 rimage (v1.6-106-g65f345a)

   e3a03f5ec7d8d33be705c5ce8a632d998ce9b4d1 rimage/tomlc99 (heads/master)

Call Stack (most recent call first):
  CMakeLists.txt:50 (include)

-- version.cmake starting SOF build at 2022-12-21T14:20:54Z UTC
-- Building git commit with parent(s):
f80e58ea1 ef56f4717 (HEAD, origin/adl-004-drop-stable, adl-004-drop-stable) module_adapter:dts: required code update for DTS library v1.0.7
-- GIT_TAG / GIT_LOG_HASH : v2.0-84-gf80e58ea11f8-dirty / f80e58ea1
-- Source content hash: fe9d56bb. Note: by design, source hash is broken by config changes. See #3890.
-- Generating /home/sof/work/sof.git/build_tgl_gcc/generated/include/version.h
-- (Re-)generating /home/sof/work/sof.git/build_tgl_gcc/generated/.config
   and /home/sof/work/sof.git/build_tgl_gcc/generated/include/autoconfig.h
   from /home/sof/work/sof.git/src/arch/xtensa/configs/tgplp_defconfig
warning: the int symbol CORE_COUNT (defined at src/platform/Kconfig:272) has a non-int default MP_NUM_CPUS (undefined)
-- Done, future changes to /home/sof/work/sof.git/src/arch/xtensa/configs/tgplp_defconfig
   will be IGNORED by this build directory! The primary .config
   file is now 'generated/.config' in the build directory.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sof/work/sof.git/build_tgl_gcc
Scanning dependencies of target genconfig
Scanning dependencies of target check_version_h
Scanning dependencies of target bin_extras
[... 329 lines cut ...]
        22      0xbe05c550      0xbe05c590      0x40    DATA    .module_init
        23      0xbe05c5c0      0xbe087500      0x2af40 DATA    .shared_data
        24      0xbe087500      0xbe087e20      0x920   DATA    .data
        25      0xbe087e20      0xbe087e8c      0x6c    DATA    .fw_ready
        26      0xbe087e90      0xbe087eb8      0x28    DATA    .AltBootManifest
        27      0xbe088000      0xbe2e0000      0x258000        BSS     .bss
        39      0x1fffa000      0x1fffaa80      0xa80   NOTE    .static_uuid_entries
        40      0x20000000      0x20016a4c      0x16a4c NOTE    .static_log_entries
        41      0x22000000      0x220002f0      0x2f0   NOTE    .fw_metadata

 module: input size 372488 (0x5af08) bytes 27 sections
 module: text 155780 (0x26084) bytes
    data 216708 (0x34e84) bytes
    bss  2621440 (0x280000) bytes

fw abi main version:    3.20.0
fw abi dbg version:     5.3.0
logs dictionary size:   92824
including fw version of size:   60
uids dictionary size:   2704
[ 98%] Built target run_smex
[ 98%] Built target sof_dump
make[1]: *** [CMakeFiles/Makefile2:2456: src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
make: *** [Makefile:814: bin] Error 2
long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$

lgirdwood, 2022-12-21:
@marc-hb @fredoh9 it looks like the container needs an update for rimage. 
@LongWang100 can you locally run a shell in your container and update the rimage git submodule. Fwiw, we have recently moved git repository alignment to use west, but this could be missing from the container. Issue update may also be slow due to holidays.

LongWang100, 2022-12-21:
@lgirdwood  Thank you for fast feedback. I am not familiar with it. And I don't know how to update the rimage git submodule. It is appreciated that you share detailed command.

lgirdwood, 2022-12-21:
> @lgirdwood Thank you for fast feedback. I am not familiar with it. And I don't know how to update the rimage git submodule. It is appreciated that you share detailed command.

Pls try 
```
./scripts/docker-run.sh bash
git submodule update --recursive
exit
```
and then rerun your container build.

marc-hb, 2022-12-21:
Hi,

The rimage clone is not inside the container. You can (and should) update it like this: `git submodule update --recursive`

I don't think it will help _for this particular problem_ but you must also update the docker image regularly:
```
docker images
docker pull thesofproject/sof
docker images
```
Let us know what difference this makes.

marc-hb, 2022-12-21:
The `git status` command or (better) the equivalent feature in your editor / IDE shows when submodules are out of date.

When something fails `git status` is always the first place to look at.

LongWang100, 2022-12-22:
> > @lgirdwood Thank you for fast feedback. I am not familiar with it. And I don't know how to update the rimage git submodule. It is appreciated that you share detailed command.
> 
> Pls try
> 
> ```
> ./scripts/docker-run.sh bash
> git submodule update --recursive
> exit
> ```
> 
> and then rerun your container build.

Thanks. I follow your command and it works for me.

LongWang100, 2022-12-22:
> Hi,
>
> The rimage clone is not inside the container. You can (and should) update it like this: `git submodule update --recursive`
>
> I don't think it will help _for this particular problem_ but you must [...]

Thanks for your support.
The docker images is same as before after I run this command "docker pull thesofproject/sof".

long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$ docker images
REPOSITORY          TAG       IMAGE ID       CREATED         SIZE
sof                 latest    bc0abdcc45e0   6 weeks ago     3.5GB
thesofproject/sof   latest    bc0abdcc45e0   6 weeks ago     3.5GB
hello-world         latest    feb5d9fea6a5   15 months ago   13.3kB
long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$ docker pull thesofproject/sof
Using default tag: latest
latest: Pulling from thesofproject/sof
Digest: sha256:bc16321c7d7cdf229f5625f30e96d0bbfce10afa45b9f1b79614d91b6b4d3c7c
Status: Image is up to date for thesofproject/sof:latest
docker.io/thesofproject/sof:latest
long@ryan-ThinkPad-T430:~/external/long/work/sof/sof$ docker images
REPOSITORY          TAG       IMAGE ID       CREATED         SIZE
sof                 latest    bc0abdcc45e0   6 weeks ago     3.5GB
thesofproject/sof   latest    bc0abdcc45e0   6 weeks ago     3.5GB
hello-world         latest    feb5d9fea6a5   15 months ago   13.3kB

closed 2022-12-22 by LongWang100