thesofproject/sof #6895: How to build the sof firmware for Alder Lake target

waterphotography, 2022-12-29:
Hi,

I'm currently working on a Chromebook target with Alder Lake platform. But it looks adl is not in the list of supported platforms in the sof firmware build script. I tried to rename and use the tdl firmware, but it didn't work. Could anyone help point out how to build the sof firmware for adl platform? Thank you very much!

aiChaoSONG, 2023-01-03:
For platforms use community key (chromebook should be one of such), adl and tgl share the same build target, which is tgl. as you can see here, the adl firmware is symbol link to tgl firmware: https://github.com/thesofproject/sof-bin/tree/main/v2.3.x/sof-v2.3/community

So theoretically, it should work to build the tgl target, and rename it. could you please provide more information? what is the build command you run? what is the error in dmesg?

plbossart, 2023-01-05:
also there are two variants for TGL and TGL-H, you need to make sure you use the right version for your device.

waterphotography, 2023-01-06:
@aiChaoSONG @plbossart it works on my side. Thanks for your kind help!

closed 2023-01-06 by waterphotography