thesofproject/sof #6903: [BUG] [SDW] Only SDW link0 can work

bardliao, 2023-01-03:
**Describe the bug**
PCM IO error when we play audio on any sdw link other than link 0.
The ALH dai_index definition in zephyr native driver seems to be different from what SOF expects.
The dai_index of sdw link 1 port 2 is 0x100, but 0x100 seems not a valid dai_index in zephyr native driver.

**To Reproduce**
Play an audio on SDW link 1 or 2
For example, Dell laptop SKU0A3E's audio configuration is that rt715(DMIC) on sdw link 0, rt711 (headset) on sdw link 1 and rt1308 (speaker) on sdw link 2.
Run `aplay -Dhw:0,0 test.wav -vv` on Dell laptop SKU0A3E

**Reproduction Rate**
100%

**Expected behavior**
Playback works.

**Impact**
Only SoundWire link 0 works

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {e3707e4dd275bc731f6ed4067a2694534675ca52}
    * SOF: {c06669e1e5ebacfefc2cf6b0486048101930db10}
2) Name of the topology file
    * Topology: sof-tgl-rt715-rt711-rt1308-mono.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL}

**Screenshots or console output**
The error trace is
[out_io_err.txt](https://github.com/thesofproject/sof/files/10335275/out_io_err.txt)

And the working trace is
[out_work.txt](https://github.com/thesofproject/sof/files/10335277/out_work.txt)
which is got after reverting https://github.com/thesofproject/sof/commit/24c0aad8328bd3f21bedb141cd63838af26de9c2

bardliao, 2023-01-03:
The ALH dai_index definition in zephyr native driver is fine. The issue is that we only specify 2  alh instances in `dts/xtensa/intel/intel_adsp_cavs25.dtsi`. The issue was fixed after adding all 16 alh instances in `dts/xtensa/intel/intel_adsp_cavs25.dtsi`
[intel_adsp_cavs25.diff.txt](https://github.com/thesofproject/sof/files/10336476/intel_adsp_cavs25.diff.txt)
Can someone file a PR to fix the issue?

lgirdwood, 2023-01-04:
@juimonen are you able to fix ?

kv2019i, 2023-01-10:
FYI  to @serhiy-katsyuba-intel

serhiy-katsyuba-intel, 2023-01-10:
Below are some issues related to ALH I observed while implementing "multi-gateway" feature:

Currently on MTL only 4 ALH streams are working. These are streams with v_index (aka stream ID) 7, 8, 9 and 10. On TGL it's probably (I did not check) 7 and 8 as TGL only has alh0 and alh1 in device tree (MTL now has alh0, alh1, alh2 and alh3).

(1)
alh_handshake_map which maps ALH stream ID to DMA handshake value starts from stream ID 7. That means streams with IDs below 7 do not work. However, we have a test on Windows platform that uses stream ID 3. Close source ACE FW works fine with that test. I had a look into the close source ACE FW sources: map there starts from stream ID 1. Stream ID 3 is BULK_OUT stream. We seems need to fix our alh_handshake_map. Moreover, in close source ACE FW there are multiple stream ID to DMA handshake mapping tables depending on hardware version. We only have one map in Zephyr driver.

(2)
Macro that creates ALH device from device tree definition looks like this (truncated to only interesting parts):
```
#define DAI_INTEL_ALH_DEVICE_INIT(n)						\
...
	static struct dai_intel_alh dai_intel_alh_data_##n = {			\
		.index = (n / DAI_NUM_ALH_BI_DIR_LINKS_GROUP) << 8 |		\
			(n % DAI_NUM_ALH_BI_DIR_LINKS_GROUP),			\
...
	};									\
```
Where DAI_NUM_ALH_BI_DIR_LINKS_GROUP defined as 4. So for alh0 to alh3, indices would be from 0 to 3, but for alh4 (if someone will add it to device tree) index would be 256. Why is that? Probably that is a bug that might be fixed, not sure.

(3)
Such macro is used in code to obtain DAI index from v_index (stream ID). Here IPC4_ALH_DAI_INDEX_OFFSET is 7 and DAI_NUM_ALH_BI_DIR_LINKS_GROUP is 4.
```
#define IPC4_ALH_DAI_INDEX(x) ((((x) & 0xF0) << DAI_NUM_ALH_BI_DIR_LINKS_GROUP) + \
				(((x) & 0xF) - IPC4_ALH_DAI_INDEX_OFFSET))
```
So seems this will not work from streams with ID less then 7. Once (1) is fixed this macro should also be fixed.
However, probably, the idea of obtaining DAI index from stream ID is wrong. For example, streams with ID 3, 7 and 32 are all valid ALH streams. Subtracting 7 (or anything else) will not work -- supported range of streams IDs is not continues, but multiple separated groups of stream IDs. Not sure how this can be fixed, maybe we can eliminate usage of DAI indices is Zephyr ALH driver and only use ALH stream IDs?

bardliao, 2023-01-11:
> (2) Macro that creates ALH device from device tree definition looks like this (truncated to only interesting parts):
>
> ```
> #define DAI_INTEL_ALH_DEVICE_INIT(n)						\
> ...
> 	static struct dai_intel_alh [...]

I am not sure if it is a bug or not. We use the formula since IPC3

lgirdwood, 2023-01-11:
> @serhiy-katsyuba-intel we don't support BULK in Linux for now, that's probably why the alh_handshake_map starts with ID 7. We only care about PCM streams for now, that should be the priority.

@plbossart are you planning BULK support after the PCM streams (or any time after) ? Will need to track this.

juimonen, 2023-01-12:
@serhiy-katsyuba-intel did we add the 4 alh nodes in mtl zephyr dt? @bardliao how many alh endpoints do you really need? all 16 or is 4 enough for now. If enough let's just replicate the mtl dt to tgl?

serhiy-katsyuba-intel, 2023-01-12:
> @serhiy-katsyuba-intel did we add the 4 alh nodes in mtl zephyr dt?

@juimonen,  yes, recently alh2 and alh3 were added to MTL device tree: https://github.com/zephyrproject-rtos/zephyr/pull/53066
This is to pass two basic "multi-gateway" tests: aggregated capture and aggregated playback.

serhiy-katsyuba-intel, 2023-01-12:
FYI @mwasko: don't we have in a roadmap for a near future things like EVAD gateway and UAOL? This issue might be blocking for those two features implementation.

bardliao, 2023-01-13:
> @serhiy-katsyuba-intel did we add the 4 alh nodes in mtl zephyr dt? @bardliao how many alh endpoints do you really need? all 16 or is 4 enough for now. If enough let's just replicate the mtl dt to tgl?

I need all 16 alh endpoints. Because alh 0 to alh 3 are on sdw link0, and we do use sdw link 1, 2, and 3. For example, rt711 (headset) on sdw link0, rt1316 (amplifier) on sdw link 1 and 2, rt715 (dmic) on sdw link 3.

juimonen, 2023-01-13:
@serhiy-katsyuba-intel could you update both cavs25 (tgl) and ace15 (mtl) DT in zephyr to have all 16 endpoints? These are needed for both platforms. We can let zephyr folks do the possible "sanitazation" later....

abonislawski, 2023-01-16:
@juimonen @bardliao we need to have all 16 alh entries in dtsi for this to work with linux just like without zephyr driver (out_work.txt)

Later we can align indexes to closed source but this is another step.

serhiy-katsyuba-intel, 2023-01-16:
> @serhiy-katsyuba-intel could you update both cavs25 (tgl) and ace15 (mtl) DT in zephyr to have all 16 endpoints? These are needed for both platforms. We can let zephyr folks do the possible "sanitazation" later....

@juimonen , well, I'm in the middle of developing something else. Feel free to update DT yourself. Adding 16 endpoints unlikely break something. I will not be able to test Bard's test case anyway. Such fix will NOT fix BULK_OUT stream ID 3 test case, but, that seems not a high priority for now and could be addressed later.

juimonen, 2023-01-17:
@bardliao @serhiy-katsyuba-intel https://github.com/zephyrproject-rtos/zephyr/pull/53853

lgirdwood, 2023-01-18:
> @bardliao @serhiy-katsyuba-intel [zephyrproject-rtos/zephyr#53853](https://github.com/zephyrproject-rtos/zephyr/pull/53853)

@juimonen Zephyr PR now merged. Can we close after a west update ?

kv2019i, 2023-02-17:
I'm guessing this is pending on Zephyr baseline update (like so many other things). This should happen soon though, and then we can close. Right? @juimonen @bardliao @lgirdwood

bardliao, 2023-02-20:
> I'm guessing this is pending on Zephyr baseline update (like so many other things). This should happen soon though, and then we can close. Right? @juimonen @bardliao @lgirdwood

Yes, I think we can close.

lgirdwood, 2023-02-20:
@mengdonglin any reason to re-open ? or closed in error as still waiting for results.

kv2019i, 2023-03-06:
@mengdonglin ?

kv2019i, 2023-03-09:
@bardliao @juimonen @mengdonglin Can this be closed or not? This is one of the bugs on the v2.5 list.

bardliao, 2023-03-10:
> @bardliao @juimonen @mengdonglin Can this be closed or not? This is one of the bugs on the v2.5 list.

Yes, we can close the issue now.

closed 2023-03-10 by kv2019i