thesofproject/sof #6917: [FEATURE] pass vendor specific FW signing options to west sign

kv2019i, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
On Intel platforms, we need to sign production releases with west sign options "-i" and  "-p". Currently SOF's xtensa-build-zephyr.py does not allow passing these options.

**Describe the solution you'd like**
Option to pass a list of options that suits multivendor usage with "west sign".

kv2019i, 2023-01-05:
FYI @abonislawski @marc-hb @lgirdwood @juimonen

lgirdwood, 2023-01-06:
@nashif fyi

marc-hb, 2023-01-09:
So as I suspected, https://github.com/zephyrproject-rtos/zephyr/pull/52942 is trying to solve this problem (among others). I just finished a (detailed) review of it.

If upstream Zephyr solves this problem with some new configuration file or something and we add some hack in `xtensa-build-zephyr.py` at _the same time_ then we're very likely to end-up in some forward/backward compatibility nightmare with upstream.

So temporarily for this 2.4 release would it be possible to (re-)run `west sign` manually with the modified `rimage` parameters? AFAIK production releases are NOT automated anyway. This will overwrite the `.ri` file in the `build-tgl` directory, then it only needs to be copied to `build-sof-staging/`.

After running `xtensa-build-zephyr.py`, scrolling up 1 or 2 screens shows the exact `west sign` command it has run.

kv2019i, 2023-01-09:
@marc-hb Let's leave the 2.4 release aside and consider just the specific question of ability to pass signing options. I think a generic ability to pass FW signing options seems like a generic thing we will need now and future. I don't see Zephyr PR52942 changing this.

If I look up Zephyr docs on west sign:
https://docs.zephyrproject.org/latest/develop/west/sign.html
```
west sign -t your_tool -- ARGS_FOR_YOUR_TOOL
```

... so generic ability to pass extra  options from xtensa-build-zephyr.py to "ARGS_FOR_YOUR_TOOL" seems like a good capability to have as it's something "west sign" tool has.

If signing is merged to build step on Zephyr side, then it would seem the extra options would need to be passed to build step, but otherwise still applicable.

Whether changing the signing options via the script parameters is a good approach or not, is another topic, but it's something SOF has in the past offered and at least the Intel binary release process has dependent upon. So moving to e.g. some in-tree configuration file to describe different signing targets, would be a very large change to the release model and something that should be discussed separately.

marc-hb, 2023-01-09:
> I think a generic ability to pass FW signing options seems like a generic thing we will need now and future.

Yes.

> I don't see Zephyr PR52942 changing this.

It does - and more.

> So moving to e.g. some in-tree configuration file

The configuration file I suggested in Zephyr 52942 does not have to be in git. It's just a suggestion, work in progress.

In the meantime:
- #6923

marc-hb, 2023-02-02:
Proof of concept available in draft PR:
-  zephyrproject-rtos/zephyr#54189

kv2019i, 2023-03-06:
@marc-hb  @abonislawski  Still on track for V2.5 -- we aim to branch this week?

marc-hb, 2023-03-06:
I have something fairly well tested and ready for review in zephyr PR 54700. I also got an approval on the concept and even one happy beta-tester but no actual code review yet.

If 54700 does not make it on time then we can cherry-pick again the small, 1 commit hack we used for 2.4, this one: 190a0ab7b874 / #6923

kv2019i, 2023-03-13:
This didn't make the cut, moving to v2.6.

lgirdwood, 2023-05-10:
@marc-hb still trending for v2.6 or can we push later ?

marc-hb, 2023-05-11:
This should make it to v2.6 depending on maintainer and code review bandwidth

aborisovich, 2023-05-20:
Ok, I'm looking at all of this and I do not agree with what you are trying to do.

> ... so generic ability to pass extra options from xtensa-build-zephyr.py to "ARGS_FOR_YOUR_TOOL" seems like a good capability to have as it's something "west sign" tool has.

It is **not** a good capability. Please note, that SOF and Rimage and two different cmake projects.
By passing user input in the `xtensa-build-zephyr.py` to project 1 (SOF) then project 2 (rimage), you are expanding functionalities of `xtensa-build-system.py` script, which should not exist at all and is a legacy burden from bash project configuration. Scripts are not a part of a build system. What we should do, is pass information to SOF CMake, which will pass information to Rimage CMake during configuration and build. This way `xtensa-build-zephyr.py` still acts like a wrapper only and passes arguments to a single cmake project (SOF).
One of bad ideas is treating `west` as build system. It is not one. It is a wrapper Zephyr tool. Neither is xtensa-build-zephyr.py script. Importing python west packages and doing stuff around Cmake is not a right/quality solution and will bring many problems in future integration tasks (like loadable modules for example). I'm refering to @marc-hb did in the https://github.com/thesofproject/sof/pull/7595 . This is not right approach in my opinion. We should build the cmake architecture not scripts.
To start doing this, we need following
1. Add new [Custom CMake target](https://cmake.org/cmake/help/latest/command/add_custom_target.html) which will run rimage with proper input arguments when SOF FW build completes.
2. New target will call `rimage` using given arguments (either using `wrapper west` sign or directly)
This way we do not need to parse any data in xtensa-build-zephyr.py as it would be passed in side CMake (this incudes regex reading of version files!).

marc-hb, 2023-05-22:
> Add new [Custom CMake target](https://cmake.org/cmake/help/latest/command/add_custom_target.html) which will run rimage with proper input arguments when SOF FW build completes.

I think this is basically what I did in https://github.com/zephyrproject-rtos/zephyr/commit/5ae7bd84d339 , copying what others did for intel_adsp before more. It does not do everything you suggested yet but I think it's a step in the direction you want.

This custom target does invoke `rimage` directly, it calls `west sign` instead because `west sign` is smarter, more flexible and it is the standard Zephyr interface for this but from a build flow perspective I think it's the same logic.

> By passing user input in the xtensa-build-zephyr.py to project 1 (SOF) then project 2 (rimage), you are expanding functionalities of xtensa-build-system.py script, [...]
> This way we do not need to parse any data in xtensa-build-zephyr.py

I agree parsing version numbers in `xtensa-build-zephyr.py` is bad but I'm not adding that; "It wasn't me!", I'm only moving this parsing to a different place in the script because it's blocking me from doing something else. 

Back to this #6917 request, my (almost completed) plan is NOT to make the "vendor specific FW signing options" go through `xtensa-build-zephyr.py`, quite the opposite. These options are configured either with `west config / west sign` or in the CMake files that invoke them.

marc-hb, 2023-05-23:
> Add new [Custom CMake target](https://cmake.org/cmake/help/latest/command/add_custom_target.html) which will run rimage with proper input arguments when SOF FW build completes.

One of the complexities is: Zephyr depends on rimage for hardware testing but Zephyr does not depend SOF. So the Zephyr build cannot pass SOF versions to rimage because there may not be any SOF version when Zephyr gets built. I bet this problem can be solved but that's out of scope here and if anything I think the changes have been making go in the right direction. In Zephyr, `west sign` used to be part of `west flash` ONLY, now it can be part of the actual build.

aborisovich, 2023-05-24:
> > Add new [Custom CMake target](https://cmake.org/cmake/help/latest/command/add_custom_target.html) which will run rimage with proper input arguments when SOF FW build completes.
> 
> I think this is basically what I did in [zephyrproject-rtos/zephyr@5ae7bd84d339](https://github.com/zephyrproject-rtos/zephyr/commit/5ae7bd84d339) , copying what others did for intel_adsp before me. It does not do everything you suggested yet but I think it's a significant step in the direction you want.
> 
> This custom target does invoke `rimage` directly, it calls `west sign` instead because `west sign` is smarter, more flexible and it is the standard Zephyr interface for this but from a build flow perspective I think it's the same logic as you described.
> 
> > By passing user input in the xtensa-build-zephyr.py to project 1 (SOF) then project 2 (rimage), you are expanding functionalities of xtensa-build-system.py script, [...]
> > This way we do not need to parse any data in xtensa-build-zephyr.py
> 
> I agree parsing version numbers in `xtensa-build-zephyr.py` is bad but I'm not adding that; "It wasn't me!", I'm only moving this parsing to a different place in the script because it's blocking me from doing something else.
> 
> Back to this #6917 request, my (almost completed) plan is NOT to make the "vendor specific FW signing options" go through `xtensa-build-zephyr.py`, quite the opposite. I've EDITed the description at the top. These options are configured either with `west config / west sign` or in the CMake files that invoke them.

Ok I get it, I'm just unhappy that we can not move straight to the point with this.

marc-hb, 2023-05-30:
Everything's merged except some documentation update and "bonus" feature https://github.com/zephyrproject-rtos/zephyr/pull/58215

Passing extra rimage arguments is performed like this:

```
rm -rf build-*
west config rimage.extra-args '-i 42 -p 43'
```

Let's close this issue after it's been used at least once for a real release candidate.

kv2019i, 2023-06-22:
We can close this once the 2.6 release is signed utilizing this feature (validating the feature).

marc-hb, 2023-06-28:
What a surprise (not!): a last-minute discovery that a Quickbuild change is required.

Shouldn't be a big deal but a good reminder that if it's not tested then it does not work.

marc-hb, 2023-07-12:
After one last minute `rimage` regression and panic (https://github.com/thesofproject/rimage/pull/168) we got confirmation from @abonislawski that this feature works as intended. Closing.

closed 2023-07-12 by marc-hb